Koenkk/zigbee2mqtt #4529: Error while trying to pair Zigbee2MQTT dev branch 1.15.0-dev - BW-SHP13

pibyth, 2020-10-02:
I'm running the Zigbee2MQTT dev branch ("version":"1.15.0-dev")'

Trying to pair BlitzWolf Smart Socket Model BW-SHP13

Anybody seen this kind of messages?

Oct 02 11:15:53 pi npm[13868]: (node:13884) UnhandledPromiseRejectionWarning: Error: SREQ '--> AF - dataRequestExt - {"dstaddrmode":2,"dstaddr":"0x000000000000fffd","destendpoint":242,"dstpanid":0,"srcendpoint":242,"clusterid":33,"transid":63,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[25,23,2,11,254,0]}}' failed with status '(0x10: MEM_ERROR)' (expected '(0x00: SUCCESS)')
Oct 02 11:15:53 pi npm[13868]:     at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
Oct 02 11:15:53 pi npm[13868]:     at Generator.next (<anonymous>)
Oct 02 11:15:53 pi npm[13868]:     at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
Oct 02 11:15:53 pi npm[13868]: (node:13884) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
Oct 02 11:15:53 pi npm[13868]: (node:13884) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
Oct 02 11:16:27 pi npm[13868]: Zigbee2MQTT:info  2020-10-02 11:16:27: MQTT publish: topic 'zigbee2mqtt/Bewegingsmelder 2', payload '{"battery":100,"device":{"applicationVersion":11,"friendlyName":"Bewegingsmelder 2","ieeeAddr":"0x00158d00030a1a5e","manufacturerID":4151,"manufacturerName":"LUMI","model":"RTCGQ01LM","networkAddress":17979,"powerSource":"Battery","type":"EndDevice"},"linkquality":70,"occupancy":true,"voltage":3015}'
O

Koenkk, 2020-10-02:
Try increasing the `measurement_poll_interval` to `99999` (https://www.zigbee2mqtt.io/devices/TS0121_plug.html#device-type-specific-configuration)

matteopiccioni, 2020-10-09:
I have just update from my previous 1.14.4 dev branch to the master 1.15.0 version...and my three BW-SHP13 disappear from my Devices.....maybe the problems are related?
Thanks

closed 2020-11-16 by github-actions[bot]