Koenkk/zigbee2mqtt #4532: TS0121 Switch add Power Measurment Support

Diddlik, 2020-10-02:
I recently bought a TS0121 switch. It has also been recognized, but only as on/off switch without measuring function. In the log you will find the following line:

**debug 2020-10-02 13:28:42: Received Zigbee message from 'Switch.ChinaTest', type 'readResponse', cluster 'haElectricalMeasurement', data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":0}' from endpoint 1 with groupID 0
debug 2020-10-02 13:28:42: No converter available for 'TS0121_switch' with cluster 'haElectricalMeasurement' and type 'readResponse' and data '{"activePower":0,"rmsCurrent":0,"rmsVoltage":0}'**

Koenkk, 2020-10-02:
- Can you provide the `data/database.db` entry of this device?
- When attaching a load (e.g. 50w bulb), do the values change? (and to what?)

Diddlik, 2020-10-02:
Output from database.db ist:
{"id":46,"type":"Router","ieeeAddr":"0x14b457fffe364093","nwkAddr":57470,"manufId":4098,"manufName":"_TZ3000_m0vaazab","powerSource":"Mains (single phase)","modelId":"TS0207","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":8,"inClusterList":[0,10,3],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"\u0000\u0000\u0000\u0000\u0005","65506":22,"modelId":"TS0207","manufacturerName":"_TZ3000_m0vaazab","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[],"meta":{}}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1601664145408}

Attaching a load I will provide you by tomorrow.

Diddlik, 2020-10-03:
So, I tested with 60w bulb and got the same values: '{"activePower":0,"rmsCurrent":0,"rmsVoltage":0}'.

It seems to be the same internal implementation for both switch and plug..

Koenkk, 2020-10-03:
@Diddlik that looks not very promising, the switch reports no power even when a load is connected. Are you sure the device is capable of this?

Diddlik, 2020-10-05:
I was not sure. I was mistaken that it was in the logs

closed 2020-11-13 by github-actions[bot]