Koenkk/zigbee2mqtt #4533: Moes BHT-002-GCLZB OFF\ON not working

Arkadi85, 2020-10-02:
## What happened
Moes BHT-002-GCLZB no convertor available for ON/OFF

## What did you expect to happen

Able to turn ON/OFF the thermostat.

## How to reproduce it (minimal and precise)
From the frontend try to turn ON/OFF the thermostat.

ERROR: No converter available for 'running' (OFF)

insipiens, 2020-10-02:
@Arkadi85 I've just submitted the PRs for all of its functions which have been merged so should be available completely in the DEV branch in a few hours

ftp27, 2020-10-25:
@insipiens It seems like it still doesn't work 
```
Zigbee2MQTT:error 2020-10-25 21:56:58: No converter available for 'system_mode' (off)
Zigbee2MQTT:error 2020-10-25 21:56:58: No converter available for 'system_mode' (heat)
Zigbee2MQTT:error 2020-10-25 21:56:58: No converter available for 'preset' (program)
Zigbee2MQTT:error 2020-10-25 21:56:59: No converter available for 'preset' (hold)
Zigbee2MQTT:error 2020-10-25 21:56:59: No converter available for 'preset' (off)
```

I just cherry picked commit [983340f44cf9ef0241327ddcaadd45495720d1c4](https://github.com/Koenkk/zigbee2mqtt/commit/983340f44cf9ef0241327ddcaadd45495720d1c4) to the master

Arkadi85, 2020-10-25:
It works with Home assistant but from the front end there is a convertor error.

insipiens, 2020-10-26:
I'm not sure what you mean. It works fine for everyone as far as I'm aware. Have you modified the Dev code locally?

insipiens, 2020-10-26:
Perhaps something else has happened to the code

closed 2020-12-03 by github-actions[bot]