Koenkk/zigbee2mqtt #4536: Hive bulb TWBulb02UK

P3nkiln, 2020-10-02:
Hi, I'm trying to add Hive HV-GSCXZB229B 9W WHITE TUNABLE.

I can get it paired but it shows as UNSUPPORTED.

Zigbee2MQTT:info  2020-10-02 16:41:55: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x00158d0002d62389"},"type":"pairing"}'
Zigbee2MQTT:info  2020-10-02 16:41:56: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x00158d0002d62389","ieee_address":"0x00158d0002d62389"},"type":"device_announce"}'
Zigbee2MQTT:info  2020-10-02 16:41:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0x00158d0002d62389"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-10-02 16:41:58: Successfully interviewed '0x00158d0002d62389', device has successfully been paired
Zigbee2MQTT:warn  2020-10-02 16:41:58: Device '0x00158d0002d62389' with Zigbee model 'TWBulb02UK' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

I've added this to devices.js

    {
        zigbeeModel: ['TWBulb02UK'],
        model: 'HV-GSCXZB229B',
        vendor: 'Hive',
        description: 'Active light, warm to cool white (E27 & B22)',
        extend: generic.light_onoff_brightness_colortemp,
    }, 

This is based on 

    {
        zigbeeModel: ['TWBulb01UK'],
        model: 'HV-GSCXZB279_HV-GSCXZB229',
        vendor: 'Hive',
        description: 'Active light, warm to cool white (E27 & B22)',
        extend: generic.light_onoff_brightness_colortemp,
    },

which I believe to be an earlier model.

Any help would be greatly appreciated.

Koenkk, 2020-10-02:
Added.

Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-02 by Koenkk

P3nkiln, 2020-10-02:
Top man, cheers 👍