Koenkk/zigbee2mqtt #4540: Support Hive SLR2

aliijm, 2020-10-03:
Hi, I am trying to support the Hive SLR2 Thermostat Receiver, but I don't know enough to be able to.

## Information about the device + link
The SLR2 uses endpoint 5 for heating like the SLR1b which is already supported, but it also has endpoint 6 for hot water. This seems to have the same functions as endpoint 5, except it doesn't report local_temperature. 

I have created an entry in my devices.js file by adapting the SLR1b entry:

        zigbeeModel: ['SLR2'],
        model: 'SLR2',
        vendor: 'Hive',
        description: 'Dual channel heating thermostat',
        supports: 'thermostat, occupied heating, weekly schedule',
        fromZigbee: [fz.thermostat_att_report, fz.thermostat_weekly_schedule_rsp],
        toZigbee: [
            tz.thermostat_local_temperature, tz.thermostat_system_mode, tz.thermostat_running_state,
            tz.thermostat_occupied_heating_setpoint, tz.thermostat_control_sequence_of_operation,
            tz.thermostat_weekly_schedule, tz.thermostat_clear_weekly_schedule,
            tz.thermostat_temperature_setpoint_hold, tz.thermostat_temperature_setpoint_hold_duration,
        ],
        meta: {configureKey: 4, disableDefaultResponse: true, multiEndpoint: true},
        endpoint: (device) => {return {'heat': 5, 'water': 6};},
        configure: async (device, coordinatorEndpoint) => {
            const heatEndpoint = device.getEndpoint(5);
            const waterEndpoint = device.getEndpoint(6);
            const binds = [
                'genBasic', 'genIdentify', 'genAlarms', 'genTime', 'hvacThermostat',
            ];
            await bind(heatEndpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatTemperature(heatEndpoint, 0, repInterval.HOUR, 1);
            await configureReporting.thermostatRunningState(heatEndpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(heatEndpoint);
            await configureReporting.thermostatTemperatureSetpointHold(heatEndpoint);
            await configureReporting.thermostatTemperatureSetpointHoldDuration(heatEndpoint);

            await bind(waterEndpoint, coordinatorEndpoint, binds);
            await configureReporting.thermostatRunningState(waterEndpoint);
            await configureReporting.thermostatOccupiedHeatingSetpoint(waterEndpoint);
            await configureReporting.thermostatTemperatureSetpointHold(waterEndpoint);
            await configureReporting.thermostatTemperatureSetpointHoldDuration(waterEndpoint);
        },

The topic receives updates from both endpoints, so e.g. the temperature setpoint changes between the two values at random. I have been able to restrict it to the heating endpoint only by adding `if (msg.endpoint.ID == 5)` into the thermostat_att_report section of `fromZigbee.js`, but I'm not able to filter the two endpoints into different topics to access both the heating and hot water.

Can anyone help with setting this up? Thanks!

## data/database.db entry of the device

`{"id":5,"type":"Router","ieeeAddr":"0x001e5e09020f779c","nwkAddr":34121,"manufId":4153,"manufName":"Computime",
"powerSource":"Mains (single phase)","modelId":"SLR2","epList":[5,6,7,8],"endpoints":{"5":{"profId":260,"epId":5,
"devId":769,"inClusterList":[0,3,9,10,513,64768],"outClusterList":[10,1026,25],"clusters":{"genBasic":{"attributes"
:{"modelId":"SLR2","manufacturerName":"Computime","powerSource":1,"zclVersion":1,"appVersion":138,"stackVersion":4,
"hwVersion":1,"dateCode":"20170622"}},"hvacThermostat":{"attributes":{"occupiedHeatingSetpoint":100,"systemMode":4,
"localTemp":1780,"tempSetpointHoldDuration":0,"runningState":0,"tempSetpointHold":0}}},"binds":[{"cluster":0,
"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},{"cluster":513,"type":"endpoint",
"deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},{"cluster":3,"type":"endpoint",
"deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},{"cluster":9,"type":"endpoint",
"deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},{"cluster":10,"type":"endpoint",
"deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1}]},"6":{"profId":260,"epId":6,"devId":769,
"inClusterList":[513],"outClusterList":[],"clusters":{"hvacThermostat":{"attributes":{
"occupiedHeatingSetpoint":3200,"systemMode":4,"tempSetpointHoldDuration":0,"tempSetpointHold":0,
"runningState":0}}},"binds":[{"cluster":0,"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a",
"endpointID":1},{"cluster":513,"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},
{"cluster":3,"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},
{"cluster":9,"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1},
{"cluster":10,"type":"endpoint","deviceIeeeAddress":"0x00124b0019365c0a","endpointID":1}]},
"7":{"profId":260,"epId":7,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},
"binds":[]},"8":{"profId":260,"epId":8,"devId":769,"inClusterList":[1026],"outClusterList":[],"clusters":{},
"binds":[]}},"appVersion":138,"stackVersion":4,"hwVersion":1,"dateCode":"20170622",
"zclVersion":1,"interviewCompleted":true,"meta":{"reporting":1,"configured":4},"lastSeen":1601725087626}`

Koenkk, 2020-10-03:
Please make a pr to https://github.com/koenkk/zigbee-herdsman-converters so I can assist.

gemignani, 2020-10-22:
Hello, many thanks for this patch, I always wanted to get rid of Hive bridge, just recently homeassistant hive cloud integration was broken too! I've written some notes that could be used for the device page.
I have a SLT3 thermostat remote control + SLR2 thermostat receiver and I've tried this change (docker dev which has latest zigbee-herdsman-coverters merged). For those who also want to do this: 
1. Remove old bridge and battery from SLT3.
2. Enable zigbee2mqtt for accept new devices and watch the logs.
3. Hold Central Heating button on SLR2 until it flashes pink then wait a few secs and hold it again, now it will be flashing amber. SLR2 will join the network.
4. Put the batteries back on SLT3, hold back and menu to perform a factory reset. It will reboot and join the network.

Now you can configure your schedule on SLT3 thermostat and it will turn on/off the boiler via SLR2!
Zigbee2mqtt will auto discovery to homeassistant two climates (cool and heat) for SLR2, but only link quality for SLT3. This is not working correctly for me.
The manual configuration works, but currently homeassistant can't parse or set the zigbee2mqtt exposed attribute "weekly_schedule_heat", so to change the schedule you have to use the thermostat controller or maybe send a json payload.
This is what I'm using:
`climate:
  - platform: "mqtt"
    name: "Hive Thermostat"
    modes: 
      - "off"
      - "auto"
      - "heat"
     
    availability_topic: "zigbee2mqtt/bridge/state"
    temperature_unit: "C"
    min_temp: "7"
    max_temp: "30"

    mode_state_topic: "zigbee2mqtt/hive_thermostat"
    mode_state_template: "{{ value_json.system_mode }}"
    mode_command_topic: "zigbee2mqtt/hive_thermostat/set/system_mode"

    current_temperature_topic: "zigbee2mqtt/hive_thermostat"
    current_temperature_template: "{{ value_json.local_temperature_heat }}"
    
    temp_step: "0.5"
    action_topic : "zigbee2mqtt/hive_thermostat"
    action_template : "{% set values = {'idle':'off','heat':'heating','cool':'cooling','fan only':'fan'} %}{{ values[value_json.running_state_heat] }}"

    temperature_state_topic : "zigbee2mqtt/hive_thermostat"
    temperature_state_template : "{{ value_json.occupied_heating_setpoint_heat }}"
    temperature_command_topic : "zigbee2mqtt/hive_thermostat/set/occupied_heating_setpoint_heat"`

closed 2020-11-29 by github-actions[bot]