Koenkk/zigbee2mqtt #4541: Device with modelID 'undefined' is not supported Tradifi walloutlet afther changing to CC2652R

MORD0R, 2020-10-03:


Koenkk, 2020-10-03:
Have you tried repairing? (remove through https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove and pair)

MORD0R, 2020-10-04:
When i try to remove it with zigbee2mqtt/bridge/config/remove

Device '0xccccccfffeae80e9' left the network
Oct 04 09:25:17  npm[8633]: Zigbee2MQTT:info  2020-10-04 09:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_removed"}'
Oct 04 09:25:17  npm[8633]: Zigbee2MQTT:warn  2020-10-04 09:25:17: Device '0xccccccfffeae80e9' left the network
Oct 04 09:25:17 npm[8633]: Zigbee2MQTT:info  2020-10-04 09:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_removed"}'
Oct 04 09:25:17  npm[8633]: Zigbee2MQTT:warn  2020-10-04 09:25:17: Device '0xccccccfffeae80e9' left the network
Oct 04 09:25:17  npm[8633]: Zigbee2MQTT:info  2020-10-04 09:25:17: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_removed"}'
Oct 04 09:25:22  npm[8633]: Zigbee2MQTT:error 2020-10-04 09:25:22: Failed to remove LampSmallCouch (Error: SRSP - ZDO - mgmtLeaveReq after 6000ms)
Oct 04 09:25:22 npm[8633]: Zigbee2MQTT:error 2020-10-04 09:25:22: See https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove for more info
Oct 04 09:25:22 npm[8633]: Zigbee2MQTT:info  2020-10-04 09:25:22: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"LampSmallCouch","type":"device_removed_failed"}'

MORD0R, 2020-10-05:
I reflashed the frimware for the stick again.

I had to do this on a windows pc because i could not unzip it on my pi.

I downloaded a few times  https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200925.zip

But i  got this error 
End-of-central-directory signature not found.  Either this file is not
  a zipfile, or it constitutes one disk of a multi-part archive.  In the
  latter case the central directory and zipfile comment will be found on
  the last disk(s) of this archive.

Maybe the file corrupt?

MORD0R, 2020-10-06:
This one unzips https://github.com/Koenkk/Z-Stack-firmware/raw/master/coordinator/Z-Stack_3.x.0/bin/CC26X2R1_20200925.zip

I reflashed it.

And then i did a force remove 
 npm[31600]: Zigbee2MQTT:info  2020-10-06 07:31:00: Force removing 'LampSmallCouch'
Oct 06 07:31:00  npm[31600]: Zigbee2MQTT:info  2020-10-06 07:31:00: Successfully force_removed LampSmallCouch
Oct 06 07:31:00 npm[31600]: Zigbee2MQTT:info  2020-10-06 07:31:00: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"LampSmallCouch","type":"device_force_removed"}'

Then i pressed the button on the Ikea tradifi

(node:31612) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Oct 06 07:34:16 npm[31600]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Oct 06 07:34:16  npm[31600]:     at listOnTimeout (internal/timers.js:549:17)
Oct 06 07:34:16  npm[31600]:     at processTimers (internal/timers.js:492:7)
Oct 06 07:34:16  npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)

Then i pressed the button on the Ikea tradifi again a few times. Because i did not get an message device_connected

Oct 06 07:34:16   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Oct 06 07:34:16   npm[31600]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Oct 06 07:34:16   npm[31600]:     at listOnTimeout (internal/timers.js:549:17)
Oct 06 07:34:16   npm[31600]:     at processTimers (internal/timers.js:492:7)
Oct 06 07:34:16   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 2)
Oct 06 07:37:36   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Oct 06 07:37:36   npm[31600]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Oct 06 07:37:36   npm[31600]:     at listOnTimeout (internal/timers.js:549:17)
Oct 06 07:37:36   npm[31600]:     at processTimers (internal/timers.js:492:7)
Oct 06 07:37:36   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 3)
Oct 06 07:40:56   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms
Oct 06 07:40:56   npm[31600]:     at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
Oct 06 07:40:56   npm[31600]:     at listOnTimeout (internal/timers.js:549:17)
Oct 06 07:40:56   npm[31600]:     at processTimers (internal/timers.js:492:7)
Oct 06 07:40:57   npm[31600]: (node:31612) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 4)
Oct 06 07:45:51   npm[31600]: Zigbee2MQTT:warn  2020-10-06 07:45:51: Device '0xccccccfffeae80e9' left the network
Oct 06 07:45:51   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:45:51: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"left_network","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_removed"}'
Oct 06 07:46:04   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:46:04: Device '0xccccccfffeae80e9' joined
Oct 06 07:46:04   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:46:04: Starting interview of '0xccccccfffeae80e9'
Oct 06 07:46:04   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_connected"}'
Oct 06 07:46:04   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:46:04: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"pairing"}'
Oct 06 07:46:07   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:46:07: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_announced"}'
[... cut ...] he CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 5)
Oct 06 07:47:52   npm[31600]: Zigbee2MQTT:error 2020-10-06 07:47:52: Failed to interview '0xccccccfffeae80e9', device has not successfully been paired
Oct 06 07:47:52   npm[31600]: Zigbee2MQTT:info  2020-10-06 07:47:52: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"pairing"}'

Koenkk, 2020-10-06:
- What system/hardware are you running z2m on?
- Can you provide the herdsman debug logging of `UnhandledPromiseRejectionWarning: Error: SRSP - ZDO - mgmtPermitJoinReq after 6000ms`?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

MORD0R, 2020-10-07:
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 1
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 2
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

processor       : 3
model name      : ARMv7 Processor rev 3 (v7l)
BogoMIPS        : 108.00
Features        : half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xd08
CPU revision    : 3

Hardware        : BCM2711
Revision        : d03114
Serial          : 10000000eb38e7cf
Model           : Raspberry Pi 4 Model B Rev 1.4

`[15:27:32] openhabian@SmartHome:~$ ^C
[15:27:37] openhabian@SmartHome:~$ cd /opt/zigbee2mqtt
[15:28:35] openhabian@SmartHome:/opt/zigbee2mqtt$ dir
data  data-backup  docker  images  index.js  lib  LICENSE  node_modules  npm-shrinkwrap.json  package.json  README.md  scripts  test  update.sh
[15:28:37] openhabian@SmartHome:/opt/zigbee2mqtt$ cd data
[15:28:41] openhabian@SmartHome:/opt/zigbee2mqtt/data$ dir
configuration.yaml       configuration.yaml.save.1  configuration.yaml.save.3  configuration.yaml.save.5  data         database.db.backup  log        state.json
configuration.yaml.save  configuration.yaml.save.2  configuration.yaml.save.4  coordinator_backup.json    database.db  database.db-old     nano.save
[15:28:47] openhabian@SmartHome:/opt/zigbee2mqtt/data$ sudo nano configuration.yaml
sudo: unable to resolve host SmartHome: Name or service not known
[sudo] password for openhabian:
[15:29:51] openhabian@SmartHome:/opt/zigbee2mqtt/data$ sudo service zigbee2mqtt stop
sudo: unable to resolve host SmartHome: Name or service not known
[15:30:45] openhabian@SmartHome:/opt/zigbee2mqtt/data$ cd ..
[15:30:55] openhabian@SmartHome:/opt/zigbee2mqtt$ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> node index.js
[... 770 lines cut ...]
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":1,"offset":0,"osal":true,"product":-1} +15ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,1,0,0,43] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,8,0,8,55,6,96,33,0,75,18,0,66] +13ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,8,0,8,55,6,96,33,0,75,18,0,66] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 8 - [0,8,55,6,96,33,0,75,18,0] - 66 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":8,"value":{"type":"Buffer","data":[55,6,96,33,0,75,18,0]}} +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":33,"offset":0,"osal":true,"product":-1} +15ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,33,0,0,11] +15ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [0,0,0,0,0,0,0,0,0,0,100,26,8,0,0,128,0,0,15,15,4,0,1,0,0,0,1,0,0,0,0,221] +16ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0,0,0,0,0,0,0,0,0,100,26,8,0,0,128,0,0,15,15,4,0,1,0,0,0,1,0,0,0,0,221] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,145,125,0,110] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [0,0,0,0,0,0,0,0,0,0,100,26,8,0,0,128,0,0,15,15,4,0,1,0,0,0,1,0,0,0,0,221,254,3,69,196,145,125,0,110] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [145,125,0] - 110 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":32145,"relaycount":0,"relaylist":[]} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,145,125,0,110] +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,145,125,0,110] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [145,125,0] - 110 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":32145,"relaycount":0,"relaylist":[]} +2s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,34,68,129,0,0,25,0,145,125,1,13,0,93,0,61,229,116,0,0,14,1,0,1,1,124,1,35,38,2] +49ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,34,68,129,0,0,25,0,145,125,1,13,0,93,0,61,229,116,0,0,14,1,0,1,1,124,1,35,38,2] +1ms
Zigbee2MQTT:error 2020-10-07 15:34:46: Failed to stop zigbee
`

Koenkk, 2020-10-07:
For openhab, make sure to disable the zwave integration: https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245

MORD0R, 2020-10-07:
> For openhab, make sure to disable the zwave integration: https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245

Ik updated the z wave binding to the latest version manauly. And i Also disabled It and Then tested IT. Same problem. I can disable it again and make a new log.

Koenkk, 2020-10-08:
In your config I see:

```yaml
advanced:
  channel: 15
  pan_id: 6756
  rtscts: false
  network_key:
    - 134
    - 30
    - 108
```

A 3 number network key is invalid, maybe this causes the problem, can you try changing it to:

```yaml
advanced:
  channel: 15
  pan_id: 6756
  rtscts: false
  network_key: GENERATE
```

(you have to repair devices after this)

MORD0R, 2020-10-08:
It is more then 3 sry.

mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
  user: openhabian
  password: Gmmarkgm15!
serial:
  port: /dev/ttyUSB0
experimental:
  output: attribute
advanced:
  log_level: debug
  channel: 15
  pan_id: 6756
  rtscts: false
  network_key:
    - 134
    - 30
    - 108
    - 229
    - 86
    - 202
    - 127
    - 10
    - 1
    - 170
    - 150
    - 228
    - 233
    - 98
    - 106
    - 173
devices:
.... And all my devices.

I even removed the zwave razberry https://z-wave.me/products/razberry/

Oct 08 16:17:56 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:17:56: Device '0xccccccfffeae80e9' joined
Oct 08 16:17:56 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:17:56: Starting interview of '0xccccccfffeae80e9'
Oct 08 16:17:56 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:17:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_connected"}'
Oct 08 16:17:56 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:17:56: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"pairing"}'
Oct 08 16:17:59 SmartHome npm[767]: Zigbee2MQTT:debug 2020-10-08 16:17:59: Device '0xccccccfffeae80e9' announced itself
Oct 08 16:17:59 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:17:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_announced"}'
Oct 08 16:19:36 SmartHome npm[767]: Zigbee2MQTT:error 2020-10-08 16:19:36: Failed to interview '0xccccccfffeae80e9', device has not successfully been paired
Oct 08 16:19:36 SmartHome npm[767]: Zigbee2MQTT:info  2020-10-08 16:19:36: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"pairing"}'
Oct 08 16:21:43 SmartHome npm[767]: Zigbee2MQTT:debug 2020-10-08 16:21:43: Saving state to file /opt/zigbee2mqtt/data/state.json

I'm going to test this on another pi tomorrow with a fresh image.

MORD0R, 2020-10-19:
I removed zigbee2mqtt and did a fresh install of zigbee2mqtt switched to dev and same problem. i stopped openhab2 and same problem.

MORD0R, 2020-10-27:
I updated to latest dev and now i do get  Failed to call 'BridgeLegacy' 'onMQTTConnected' (Error: SRSP - ZDO - extNwkInfo after 6000ms

Koenkk, 2020-10-28:
Can you provide the herdsman debug logging of this? (don't post it here but on pastebin.com and link it here).

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

MORD0R, 2020-10-29:
Here you go.
https://pastebin.com/aTiwN3Z4

Koenkk, 2020-10-29:
Something goes wrong with the serial communication, the following 2 things often cause problems:
- https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed
- https://www.zigbee2mqtt.io/information/FAQ.html#for-openhab-users-disable-zwave-binding

MORD0R, 2020-10-29:
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'modemmanager' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.

And no Z wave installed.
[image]

Koenkk, 2020-10-29:
Are there any other USB devices connected to your machine? And what machine are you running on? (raspberry pi?) It could also be that the USB cable is the problem.

MORD0R, 2020-10-30:
No other usb devices conected. Im using a PI 4. I tested it with USB extension cable and without. How can i find out if something is blocking it?

Koenkk, 2020-10-30:
Finding out what is causing this might not be easy, what applications are you running on your Pi?

MORD0R, 2020-10-30:
Openhabain https://www.openhab.org/docs/installation/openhabian.html

With the follwoing addons
binding = miio , yeelight , network , chromecast , mqtt , solaredge, spotify, unifi, feed,
ui = habmin , basic , paper , habpanel , habot ,
persistence = influxdb , mapdb,
transformation = bin2json , jsonpath , javascript , map ,
misc =  ruleengine, openhabcloud, 

dsmr-reader (full software version not connected to p1 port) https://dsmr-reader.readthedocs.io/nl/v4/installation.html
Grafana https://grafana.com/

Koenkk, 2020-10-31:
Nothing strange there. To debug further what causes the issue I recommend to setup a separate clean system with only Zibee2MQTT installed and see if the issue also happens there, if not, try to add thing to identify the point of failure.

MORD0R, 2020-11-01:
i made separate clean system and that one works. Now i only have to figure out what is blocking it.

`Zigbee2MQTT:info  2020-11-01 12:40:51: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-11-01.12-40-50' filename: log.txt
Zigbee2MQTT:info  2020-11-01 12:40:51: Starting Zigbee2MQTT version 1.16.0 (commit #9ed84e1)
Zigbee2MQTT:info  2020-11-01 12:40:51: Starting zigbee-herdsman...
Zigbee2MQTT:info  2020-11-01 12:40:56: zigbee-herdsman started
Zigbee2MQTT:info  2020-11-01 12:40:56: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"}'
Zigbee2MQTT:info  2020-11-01 12:40:56: Currently 0 devices are joined:
Zigbee2MQTT:warn  2020-11-01 12:40:56: `permit_join` set to  `true` in configuration.yaml.
Zigbee2MQTT:warn  2020-11-01 12:40:56: Allowing new devices to join.
Zigbee2MQTT:warn  2020-11-01 12:40:56: Set `permit_join` to `false` once you joined all devices.
Zigbee2MQTT:info  2020-11-01 12:40:56: Zigbee: allowing new devices to join.
Zigbee2MQTT:info  2020-11-01 12:40:56: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:info  2020-11-01 12:40:56: Connected to MQTT server
Zigbee2MQTT:info  2020-11-01 12:40:56: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2020-11-01 12:40:56: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"9ed84e1","coordinator":{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200925,"transportrev":2},"type":"zStack3x0"},"log_level":"info","network":{"channel":11,"extendedPanID":"0xdddddddddddddddd","panID":6754},"permit_join":true,"version":"1.16.0"}'
Zigbee2MQTT:info  2020-11-01 12:41:43: Device '0xccccccfffeae80e9' joined
Zigbee2MQTT:info  2020-11-01 12:41:43: Starting interview of '0xccccccfffeae80e9'
Zigbee2MQTT:info  2020-11-01 12:41:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_connected"}'
Zigbee2MQTT:info  2020-11-01 12:41:43: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"pairing"}'
Zigbee2MQTT:info  2020-11-01 12:41:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"0xccccccfffeae80e9"},"type":"device_announced"}'
Zigbee2MQTT:info  2020-11-01 12:41:57: Successfully interviewed '0xccccccfffeae80e9', device has successfully been paired
Zigbee2MQTT:info  2020-11-01 12:41:57: Device '0xccccccfffeae80e9' is supported, identified as: IKEA TRADFRI control outlet (E1603/E1702)
Zigbee2MQTT:info  2020-11-01 12:41:57: Configuring '0xccccccfffeae80e9'
Zigbee2MQTT:info  2020-11-01 12:41:57: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_successful","meta":{"description":"TRADFRI control outlet","friendly_name":"0xccccccfffeae80e9","model":"E1603/E1702","supported":true,"vendor":"IKEA"},"type":"pairing"}'
Zigbee2MQTT:info  2020-11-01 12:41:57: Successfully configured '0xccccccfffeae80e9'
`

MORD0R, 2020-11-04:
I think i fixed it.

`sudo systemctl disable hciuart`

MORD0R, 2020-11-05:
@Koenkk maby you can add this this to the docmentation?

sudo systemctl disable hciuart

It is stil working for me.

closed 2020-11-05 by MORD0R

Koenkk, 2020-11-06:
Added to the FAQ, thanks