Koenkk/zigbee2mqtt #4542: Add the "acquisition_time" to all messages

davekr8, 2020-10-03:
I think a "acquisition time" (the earliest timestamp, when the zigbee message is received, before any time consuming processing/convertion is done) is a (the only) relevant time information that is relevant for data aquisition and correct time-based home-automation task.

I don't want the second or third service in chain of data processing instances to handle time-stamps, this will add further delay and is a source of additional jitter as well.

I don't know if the sensors provide their own time source, which is of course not time-synched, but also those independent time-domains might interesting data for further reseach or monitoring features.

Anyway, I think integrating the acquisition time on the coordinator peer might be easy to implement and will increase accurary of subsequent data processing on top of zigbee2mqtt.

What do you guys think?

Remark: I know, there were similar feature requests in the past, but not with the special focus of the one here - and non of those made it to the master - as far as I know.

Koenkk, 2020-10-03:
I think this can already be achieved with the `last_seen` option: https://www.zigbee2mqtt.io/information/configuration.html#configuration

davekr8, 2020-10-04:
If this the meaning of the last_seen timestamp is equivalent to the one of acquision time, then it's fine. But I though this is the time, when the message was send via mqtt (so speaking acq.time + processingtime)

What do you know about a time-field set by the sensors? If one know the range and the overflow, this time-stamp might be interesting as well. But maybe this time is likely to drift ...

Koenkk, 2020-10-05:
Last seen is set as soon as Zigbee2MQTT receives the message (so before any processing takes place). Messages send by devices don't contain a timestamp when they were send.

closed 2020-11-13 by github-actions[bot]