Koenkk/zigbee2mqtt #4543: touchlink fails with "'BUFFER' not available" error

falcke, 2020-10-03:
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC1352P-2
Adapter firmware version: Z-Stack 3.x.0 20200925 firmware (#212)

**reproducible:** always
**how I've started touchlink:** [as described here](https://www.zigbee2mqtt.io/information/touchlink)
**expected:** touchlink should start
**what happend:** I'm trying to get touchlink running but it fails with a **Error: Write for 'BUFFER' not available** [Here's the full message](https://gist.github.com/falcke/41d0222a38a41796cc06134fc39cf525)

Koenkk, 2020-10-03:
could you provide the herdsman debug logging of this issue?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

falcke, 2020-10-04:
> could you provide the herdsman debug logging of this issue?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

[Here's what you've asked for](https://gist.github.com/falcke/3831e88faad4c5ad3204fe442a93e568). Thank you very much and please let me know if I can provide any further information.

Koenkk, 2020-10-05:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-05 by Koenkk

IntegersOfK, 2020-10-17:
I had the same problem and Googled for this issue. Dev branch solved it for me, thank you. Just thought it might be nice to have another confirmation.

wilcomir, 2020-10-17:
I can confirm I had the same issue, switching to the dev branch fixed it for me. Thanks Koen! Enjoy your evening.

falcke, 2020-11-02:
Sorry for the delay. I made it to use the dev branch now and I'm getting [these messages](https://gist.github.com/falcke/e695cd4ac9644dcc73c4af9d1f19e9d9) now. Is it related or something different? Thank you so far!

ashwinvandijk, 2020-11-12:
I'm not sure if it's related, but when I try to start Touchlink I get a time-out: `error":"SRSP - AF - interPanCtl after 6000ms`. When I try to restart zigbee2mqtt I get another timeout that it can't find my adapter anymore. The only way to fix it is to reprogram my zig-a-zig-ah. I'm using z2m 1.16, and the CC26X2R1_20201026 coordinator firmware.