Koenkk/zigbee2mqtt #4544: Sonoff ZigBee Bridge

akumamg, 2020-10-03:
Will there be support for Sonoff ZigBee Bridge in the future ?

Koenkk, 2020-10-03:
Currently there are no plans, but if anybody is interested in implementing this I can give some support.

LudoOne, 2020-10-28:
I have one! Also a TRV (TS0601) like in issue#3821.
What do I have to do to replace the serial connection for a IP socket (socket://[IP]:8888)?

Johndolk, 2020-10-30:
> Currently there are no plans, but if anybody is interested in implementing this I can give some support.

I am really interessted in implementing this - But how to? :)

akumamg, 2020-10-30:
Its simple, Write code

Westcott1, 2020-10-30:
I hope that Koenkk doesn't delete me for posting this, but I flashed my ZBbridge with T*smo*a, and wrote some Node.js to process the resulting MQTT.

LudoOne, 2020-11-08:
I tried both:
`socat:
  slave:
    tcp:192.168.1.107:8888,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5`
as https://www.zigbee2mqtt.io/how_tos/how_to_connect_to_a_remote_adapter.html#3-configure:
`serial:
  port: 'tcp://192.168.1.107:8888'`
But resulted in:
`Zigbee2MQTT:error 2020-11-08 01:42:28: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-11-08 01:42:28: Failed to start zigbee
Zigbee2MQTT:error 2020-11-08 01:42:28: Exiting...
Zigbee2MQTT:error 2020-11-08 01:42:28: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.16.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.16.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-11-08T00_42_28_346Z-debug.log
2020-11-08T01:42:29: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-11-08T01:42:29: PM2 log: App [npm:0] starting in -fork mode-
2020-11-08T01:42:29: PM2 log: App [npm:0] online
> zigbee2mqtt@1.16.1 start /zigbee2mqtt-1.16.1
> node index.js`

CJMarais, 2020-11-09:
> I hope that Koenkk doesn't delete me for posting this, but I flashed my ZBbridge with T_smo_a, and wrote some Node.js to process the resulting MQTT.

Can you share more of what you did?

Westcott1, 2020-11-09:
Hi CJMarias,
You are welcome to what I've done so far.
It's an add-on to more Node.js that I wrote to interface between Domoticz and Z2M
A config file holds the idx of any Domoticz devices that can be updated from incoming MQTT.
The ZBbridge is a coordinator for a separate ZigBee network.
It also reads from a Sonoff 433 bridge, also flashed with T-sm-ta.

tyjtyj, 2020-11-22:
after my cc2531 keep dropping device, i got sonoff zigbee. Its disappointed to learn it is not supported by z2m. anyone start working on this? The problem is i dont know where to start.

MattWestb, 2020-11-22:
@tyjtyj I 110% agree with you but for the moment its looks like 3 or 4 devs have starting and doing little but not so much for getting it released in public.
What i have understanding is EZSP mutch "low level" command that is making mutch work for implanting it but is very powerful and flexible.
The sonoff is working good with tasmota zigbee bridge but is little limited in functionality but OK for small system with sensors (Was implanted of one dev in some months from scratch !!).
ZHA devs have updating the code in the summer with around 20K lines for implanting the new EZSP protocol and is working well and is stabile and is have good support for real zigbee 3 devices. Its with Rasp/CornBee the most powerful coordinator on the market and much better then first and second gen TI:s.
I hope that some devs is going together and getting it working on the Z2M (java)train.

lotohov, 2020-11-27:
check out , there is another firmware for esp. https://www.zigbee2mqtt.io/information/connecting_cc2530.html it migt work ?

akumamg, 2020-12-29:
Any news about this ?

LudoOne, 2020-12-29:
Did you write code?

svrooij, 2021-01-06:
If it just has a tcp socket and you want to connect zigbee2mqtt without changing code you can checkout [this](https://stackoverflow.com/a/22689163/639153) answer. You have socat it’s an app that lets you configure a local virtual serial port that just connects to a tcp socket.

then you can connect zigbee2mqtt to /dev/virtualcom0 which it actually socat that connects to a tcp socket.
Changing this inside zigbee2mqtt will require refactoring actual communication to the serial port, but should be doable.

tyjtyj, 2021-01-07:
> If it just has a tcp socket and you want to connect zigbee2mqtt without changing code you can checkout [this](https://stackoverflow.com/a/22689163/639153) answer. You have socat it’s an app that [...]

EZSP not supported by z2m yet. So connecting via tcp has not use as the z2m does not speak EZSP language.

closed 2021-02-15 by github-actions[bot]

anubisg1, 2021-02-25:
This is extremely important for me too, has there been any update? This hub is very powerful and using zigbee2mqtt would really simplify a lot of things... Please make it happen

svrooij, 2021-02-25:
@anubisg1 this is an opensource project. A lot of work needs to be done before this will happen. Since this issue is already closed (because it was getting old) I would not get your hopes up. Sponsoring Coen €1000 might draw some attention to this topic.

anubisg1, 2021-02-25:
> @anubisg1 this is an opensource project. A lot of work needs to be done before this will happen. Since this issue is already closed (because it was getting old) I would not get your hopes up. Sponsoring Coen €1000 might draw some attention to this topic.

Thanks for the reply... I understand what you say but I think you added an extra 0 without noticing... Probably a typo

svrooij, 2021-02-25:
I was just send a on first look high amount, but people have no idea how much it would cost to get this build.

The device you're talking about probably needs custom firmware to enable some sort of tcp socket where zigbee2mqtt could talk to.

I cannot check the wallet of Koen or the other contributors, so I don't know what it would take to get attention to this issue.

Maybe ask on discord or start investing the device yourself.

Personally I really like the zig-a-zag (zzh) usb stick, it works perfectly with a raspberry (that is needed anyway), and would thus not invest any time in building support for a new coordinator

shyney7, 2021-02-28:
+1 for implementing support for the sonoff zb bridge

gabeale, 2021-03-01:
Hi guys, it would be nice to get Z2M working also on Silicon Labs based chips (e.g. Sonoff Zigbee bridge) that are as powerful as Texas Instruments ones (e.g. CC253x) or even more. I know that this is costing a big effort for developers and that a great job has already been done till date. I'm available to give my contribution for sponsoring this.
Thx to all! ;-)

kirovilya, 2021-03-01:
Guys i'm trying to do this... https://github.com/Koenkk/zigbee-herdsman/pull/317

gasecki, 2021-03-16:
I would love to see it working one day! So of bridges works fine. Only missing the way to get more out of it for automation.

Hedda, 2021-03-29:
FYI; a tip is to first follow these step-by-step instructions on how to flash Zigbee chip with ncp-uart-sw_6.7.8_115200.ota firmware:

https://www.digiblur.com/2020/07/how-to-use-sonoff-zigbee-bridge-with.html

Only after once managed to successfully flashed the ncp-uart-sw_6.7.8_115200.ota to the Zigbee chip should you attemt to follow:

https://www.digiblur.com/2021/03/zigbee2mqtt-with-sonoff-zigbee-bridge.html

Seams that many miss that they also need to flash the Zigbee chip or read other docs that refer to older or withdrawn versions:

https://github.com/arendst/Tasmota/tree/development/tools/fw_SonoffZigbeeBridge_ezsp

Toiday it is only 6.7.8 that is the recommended version however 6.7.9 version should also work (but not any older or newer).

Note! Warning is that Zigbee to WiFi bridges for EZSP is not recommended for production as it doesn't work well with packed loss:

https://github.com/zigpy/bellows#warning-about-zigbee-to-wifi-bridges

https://www.home-assistant.io/integrations/zha/#warning-about-wi-fi-based-zigbee-to-serial-bridgesgateways

PS: As noted elsewhere, if want to help as a pre-alpha/dev tester then currently post EZSP issues to https://github.com/Koenkk/zigbee-herdsman/pull/317