Koenkk/zigbee2mqtt #4545: Unable to unbind Xiaomi Aqara Wall Switch from zzh! CC2652R coordinator

Martin004, 2020-10-03:
It is simaler to Issue "Unable to unbind Philips Hue dimmer from deCONZ coordinator #4171"

But don't work with the actual koenkk/zigbee2mqtt:latest-dev version.

Please inspect the log:
info  2020-10-03 18:22:15: Logging to console and directory: '/app/data/log/2020-10-03.18-22-15' filename: log.txt
debug 2020-10-03 18:22:15: Removing old log directory '/app/data/log/2020-10-03.16-39-11'
debug 2020-10-03 18:22:15: Loaded state from file /app/data/state.json
info  2020-10-03 18:22:15: Starting Zigbee2MQTT version 1.15.0-dev (commit #43f085b)
info  2020-10-03 18:22:15: Starting zigbee-herdsman...
debug 2020-10-03 18:22:15: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"baudRate":115200,"path":"/dev/ttyUSB0","rtscts":false}}'
info  2020-10-03 18:22:18: zigbee-herdsman started
info  2020-10-03 18:22:18: Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200805,"transportrev":2},"type":"zStack3x0"}'
debug 2020-10-03 18:22:18: Zigbee network parameters: 
...
info  2020-10-03 18:22:18: Currently 43 devices are joined:

error 2020-10-03 18:27:06: Failed to unbind cluster 'lightingColorCtrl' from 'TrippelSchalter2' to 'Coordinator' (Error: Unbind 0x04cf8cdf3c75b431/1 lightingColorCtrl from '0x00124b00215ff711/1' failed (AREQ - ZDO - unbindRsp after 10000ms))
info  2020-10-03 18:27:06: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"cluster":"lightingColorCtrl","from":"TrippelSchalter2","to":"Coordinator"},"type":"device_unbind_failed"}'
error 2020-10-03 18:27:06: Failed to unbind

Greatings
Martin

Koenkk, 2020-10-03:
- Right before executing this command, did you wake up the Xiaomi wall switch by pressing a button on it?
- What is the exact model of this device?

Martin004, 2020-10-04:
Thanks for your replay!

I use Xiaomi WXCJKG13LM #
Yes, I waked it up. But a very good hint.

Now, I tried to set the command mode.

I will send you the log as attachment.

[log_unbind_problem.txt](https://github.com/Koenkk/zigbee2mqtt/files/5322829/log_unbind_problem.txt)

greatings 
Martin

Koenkk, 2020-10-05:
Can you provide me your networkmap?https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap

closed 2020-11-13 by github-actions[bot]

dennis-bell, 2020-12-13:
@Martin004 initially I had the same problem with 2 of those Opple switches and 2 ikea light bulbs.
I got it working by these steps:
- re-pair
- keep it awake by pressing a button
- and sent out the instruction for binding.
[bind.txt](https://github.com/Koenkk/zigbee2mqtt/files/5684706/bind.txt)

jogleasonjr, 2021-12-08:
Thanks @dennis-bell, I struggled to get an Aqara Double Rocker wireless to pair with CC2652R / ZHA. Your 2nd bullet was the trick -- pressing buttons repeatedly to keep the device awake long enough for a full initialization and configuration. Thank you!