Koenkk/zigbee2mqtt #4546: zigbee2mqtt@1.15.0 is crashing several seconds after start on 64bit raspberry linux kernel

antonakv, 2020-10-03:
When enabling arm_64bit=1 in /boot/config.txt on Raspberry Pi linux zigbee2mqtt@1.15.0 is crashing several seconds after start with 64bit raspberry linux kernel: Linux raspberrypi 5.4.51-v8+ #1333 SMP PREEMPT Mon Aug 10 16:58:35 BST 2020 aarch64 GNU/Linux

Latest firmware and latest version installed, connected with extension cable

Logs: https://pastebin.com/bfquvE1e

Koenkk, 2020-10-03:
This is the npm logging (not useful for debugging Zigbee2MQTT issues). Please provide the herdsman debug logging.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

antonakv, 2020-10-04:
@Koenkk It stopped to crash for a while after enabling log_level: debug in configuration.yaml and reboot with 64bit kernel. I will monitor the status and post logs when it crashes again.

closed 2020-11-12 by github-actions[bot]