Koenkk/zigbee2mqtt #4547: Disable MQTT logging?

David-Lor, 2020-10-03:
## What happened

It seems that latest (or one of last) zigbee2mqtt update/s started publishing new messages on MQTT topics: `zigbee2mqtt/bridge/info`, `zigbee2mqtt/bridge/devices` and `zigbee2mqtt/bridge/logging`.

## What did you expect to happen

I'd like to disable messages on these topics. I could not find how to disable these messages on [Configuration docs](https://www.zigbee2mqtt.io/information/configuration.html), neither found any references about including these topics on [Releases changelogs](https://github.com/Koenkk/zigbee2mqtt/releases).

## How to reproduce it (minimal and precise)

Updated zigbee2mqtt (docker) to latest version available ATM, and started running it. I have logs of all the MQTT messages of my broker, and these 3 topics had no messages prior to updating the container image.

## Debug info

Zigbee2MQTT version: 1.15.0

Koenkk, 2020-10-03:
This is part of the new api (https://github.com/Koenkk/zigbee2mqtt/issues/3281) and cannot be disabled for now. Disabling this would also break the frontend (https://github.com/Koenkk/zigbee2mqtt/issues/4266)

bartplessers, 2020-10-15:
Hi @Koenkk 

But is it possible to have the MQTT device messages under, let's say
zigbee2mqtt/bridge/devices?
subscribing now to zigbee2mqtt/# gives overwelming feedback

Koenkk, 2020-10-15:
@bartplessers it already is for the new api: https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#zigbee2mqttbridgedevices

bartplessers, 2020-10-15:
@Koenkk probably I didn't express myself correctly, sorry for that.
I would like to have something like
`zigbee2mqtt/devices/[FRIENDLY_NAME]`
were 'devices" is a common fixed path for all the "normal" messages of the devices.

so that I can subscribe again with
`mosquitto_sub -h mqtt.bearforce.org -p 1883 -d -t 'zigbee2mqtt/devices/#'`

If I subscribe now to
`mosquitto_sub -h mqtt.bearforce.org -p 1883 -d -t 'zigbee2mqtt/#'`
I get a lot of overhead.
i.e. a single click on an aqara switch gives (home assistant support is on):

```
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (168 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/bewegingssensor (O05)', payload '{\"battery\":97,\"linkquality\":36,\"occupancy\":false,\"voltage\":2995}'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bewegingssensor (O05)', ... (64 bytes))
{"battery":97,"linkquality":36,"occupancy":false,"voltage":2995}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (190 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/schakelaar (S08)', payload '{\"action\":\"single_left\",\"battery\":100,\"click\":\"left\",\"linkquality\":57,\"voltage\":3345}'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/schakelaar (S08)', ... (85 bytes))
{"action":"single_left","battery":100,"click":"left","linkquality":57,"voltage":3345}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (160 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/schakelaar (S08)', payload '{\"action\":\"\",\"battery\":100,\"linkquality\":57,\"voltage\":3345}'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/schakelaar (S08)', ... (59 bytes))
{"action":"","battery":100,"linkquality":57,"voltage":3345}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (159 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/schakelaar (S08)', payload '{\"battery\":100,\"click\":\"\",\"linkquality\":57,\"voltage\":3345}'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/schakelaar (S08)', ... (58 bytes))
{"battery":100,"click":"","linkquality":57,"voltage":3345}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (109 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/schakelaar (S08)/action', payload 'single_left'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/schakelaar (S08)/action', ... (11 bytes))
single_left
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (101 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/schakelaar (S08)/click', payload 'left'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/schakelaar (S08)/click', ... (4 bytes))
left
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/licht (L08)/set', ... (15 bytes))
{"state": "ON"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/bridge/logging', ... (201 bytes))
{"level":"info","message":"MQTT publish: topic 'zigbee2mqtt/licht (L08)', payload '{\"brightness\":254,\"linkquality\":36,\"state\":\"ON\",\"update\":{\"state\":\"idle\"},\"update_available\":false}'"}
Client mosqsub|7093-RPI-86 received PUBLISH (d0, q0, r0, m0, 'zigbee2mqtt/licht (L08)', ... (99 bytes))
{"brightness":254,"linkquality":36,"state":"ON","update":{"state":"idle"},"update_available":false}
```

A lot of messages, especially because the logging messages are also displayed.

Of course, I can subscribe to 
`mosquitto_sub -h mqtt.bearforce.org -p 1883 -d -t 'zigbee2mqtt/schakelaar (S08)'`
but then I do not see the messages of my other devices.

Probably this will brake a lot of things.
Another solution could be to log to a different main topic, i.e.:
'zigbee2mqttlogging'
So this wil not interfear with device messages

kind regards
Bart

Koenkk, 2020-10-16:
Why not subscribe to `zigbee2mqtt/+`? This will work unless you use the attribute output or have slashes in your friendly names.

bartplessers, 2020-10-16:
Zigbee2mqtt/+ --> Because I've forgotten that one? 😄
It's a good workaround, but then you miss some information.
Just consider: in my humble opinion, it's not a good idea to mix 'data' and 'logging' in the same branche. But probably you have good reason to do so. Maybe elaborating this and explaining the new behaviour would be of some help.

Grtz
B

closed 2020-11-23 by github-actions[bot]

David-Lor, 2021-01-09:
Hi @Koenkk 

I saw that the issue #3281 was closed and is part of the latest zigbee2mqtt version (1.17.0). I configured the `"legacy_api": false`, which I though would stop these logging messages from being sent, but they keep coming. Would we expect them to stop when legacy api is disabled, or is not possible yet?

Koenkk, 2021-01-10:
@David-Lor zigbee2mqtt/bridge/info, zigbee2mqtt/bridge/devices and zigbee2mqtt/bridge/logging are actually the topics used by the new api.

frogale, 2021-01-10:
This should be fixed somehow.. I have ale problems with this "new" topics..

dmak, 2021-10-06:
How is it possible to disable sending messages to `zigbee2mqtt/bridge/log` for version 1.18.x? In most cases log file is more than enough.

Semmu, 2022-09-22:
I would also like to know if it is possible to disable these.

Eistee82, 2023-10-05:
my iobroker mqtt client spams 18GB of zigbee2mqtt bridge logging to my /var/log/syslog so I'm out of diskspace

slush0, 2023-12-29:
I second this, using the same space for actual sensor readings and application logging is very annoying. I have two locations connected via mosquitto bridges and currently there's no other way how to stop bridging this spam traffic except of explicit listing of all sensors in mosquitto bridge config.

Internal application logging and frontend functionality should use separate prefix (ideally zigbee2mqtt/bridge/* for logging and zigbee2mqtt/device/*).

Please reconsider this issue. The fix is rather trivial.