Koenkk/zigbee2mqtt #4549: zigbe2mqtt crashes on start with option zigbee_shepherd_devices: true

skumka, 2020-10-03:
zigbe2mqtt crashes on start

Issue 1: is this procedure still working? https://github.com/danielwelch/hassio-zigbee2mqtt#adding-support-for-new-devices
Looks it is not since:
Issue 2: zigbe2mqtt crashes on start when adding **zigbee_shepherd_devices: true** on top-level of configuration and after copying devices.js in /share/zigbee2mqtt folder (devices.js copiet from master branch.

removing **zigbee_shepherd_devices: true** make z2m starting well.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Searching for custom devices file in zigbee2mqtt data path...
[Info] Socat is DISABLED and not started
2020-10-03T22:07:46: PM2 log: Launching in no daemon mode
2020-10-03T22:07:46: PM2 log: App [npm:0] starting in -fork mode-
2020-10-03T22:07:47: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-03 22:07:51: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-03.22-07-51' filename: log.txt
Zigbee2MQTT:debug 2020-10-03 22:07:51: Removing old log directory '/share/zigbee2mqtt/log/2020-10-03.21-56-18'
internal/modules/cjs/loader.js:968
  throw err;
  ^
Error: Cannot find module './lib/exposes'
Require stack:
- /zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/devices.js
- /zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/index.js
- /zigbee2mqtt-1.15.0/lib/zigbee.js
- /zigbee2mqtt-1.15.0/lib/controller.js
- /zigbee2mqtt-1.15.0/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:965:15)
    at Function.Module._load (internal/modules/cjs/loader.js:841:27)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/devices.js:29:17)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Module.require (internal/modules/cjs/loader.js:1025:19) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/devices.js',
    '/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/index.js',
    '/zigbee2mqtt-1.15.0/lib/zigbee.js',
    '/zigbee2mqtt-1.15.0/lib/controller.js',
    '/zigbee2mqtt-1.15.0/index.js'
  ]
}
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

Koenkk, 2020-10-03:
Try switching to the edge version of the addon. The devices.js from master is not compatible with 1.15.0

skumka, 2020-10-03:
I am on the last version of add-on  and corresponding devices.js

Koenkk, 2020-10-03:
Last version != edge, see https://github.com/danielwelch/hassio-zigbee2mqtt#installation

skumka, 2020-10-04:
Thank you @Koenkk . I figured out my mistake. I took .95 version of devices.js instead .94 which is a proper one for z2m in version115.

closed 2020-10-04 by skumka