Koenkk/zigbee2mqtt #4552: Iris iL07_1 Missing Temperature/Humidity

kbickar, 2020-10-04:
## What happened
The Iris iL07_1 device has a temperature/humidity sensor and reports those values, but they are not published by zigbee2mqtt

## What did you expect to happen
temperature/humidity values should be visible
## How to reproduce it (minimal and precise)
Pair device, view fields published
## Debug info
Zigbee2MQTT version:  1.15.0
Adapter hardware: CC2531
Adapter firmware version: sonoff version

TilmanK, 2020-10-05:
I've submitted a PR, once Koen merges it and updates the dev branch you'll need to test if it works for you.

Koenkk, 2020-10-05:
@kbickar can you test the changes done by @TilmanK, might be that we also need to setup reproting/binding (like https://github.com/Koenkk/zigbee-herdsman-converters/blob/a8e5c760f0b1d3e8a0d8445c94e8d2b2b5da36fd/devices.js#L3383). Thanks for helping @TilmanK 

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

kbickar, 2020-10-05:
It's not actually my device, but I got my friend to test it out and after adding to the homeassistant.js config: 
`'iL07_1': [cfg.binary_sensor_occupancy, cfg.binary_sensor_tamper, cfg.binary_sensor_battery_low, cfg.sensor_temperature, cfg.sensor_humidity],`

The sensors are showing up with "unknown" values so it sounds like it might need something else

TilmanK, 2020-10-07:
> It's not actually my device, but I got my friend to test it out and after adding to the homeassistant.js config:
> `'iL07_1': [cfg.binary_sensor_occupancy, cfg.binary_sensor_tamper, [...]

Can you confirm you tried with my suggested changes?

kbickar, 2020-10-07:
Yes, that was with the updated devices.js

Koenkk, 2020-10-16:
So to confirm, with these changes: https://github.com/Koenkk/zigbee-herdsman-converters/pull/1630/files you get temperature and humidity values for this device?

kbickar, 2020-10-16:
No values were reported

Koenkk, 2020-10-16:
Can you try with:

```js
{
    zigbeeModel: ['1117-S'],
    model: 'iL07_1',
    vendor: 'Iris',
    description: 'Motion Sensor',
    supports: 'motion, tamper, temperature, humidity and battery',
    fromZigbee: [fz.ias_occupancy_alarm_2, fz.temperature, fz.humidity],
    toZigbee: [],
    exposes: [exposes.boolean('occupancy'), exposes.boolean('battery_low'), exposes.boolean('tamper')],
    meta: {configureKey: 1},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'msRelativeHumidity']);
        await configureReporting.temperature(endpoint);
        await configureReporting.humidity(endpoint);
    },
},
```

After this manually configure via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure (make sure to wakeup the device right before executing it by clicking a button on it) and check the log if OK?

kbickar, 2020-11-11:
I commented on the PR the config that was working.   

I added the line 
```js
exposes: [e.occupancy(), e.battery_low(), e.temperature(), e.humidity()],
```

After that was added it worked and the HA discovery messages were correct.

I also checked and the device physically has no tamper sensor so that can be removed

closed 2020-11-11 by Koenkk

Tediore, 2021-02-20:
Running 1.17.1. I have three of these motion sensors and none of them would report temp or humidity without adding what Koenkk mentioned above to my `devices.js` (https://github.com/Koenkk/zigbee2mqtt/issues/4552#issuecomment-710308593)