Koenkk/zigbee2mqtt #4555: Error: Cannot request when znp has not been initialized yet

Baggerfahrer, 2020-10-04:
## What happened
I did not change anything.

## What did you expect to happen
Normal start of zigbee2mqtt.

## How to reproduce it (minimal and precise)
After the start of zigbee2mqtt i get the following error message:

Zigbee2MQTT:error 2020-10-04 07:47:11: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-04 07:47:11: Failed to start zigbee
Zigbee2MQTT:error 2020-10-04 07:47:11: Exiting...
Zigbee2MQTT:error 2020-10-04 07:47:11: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:87:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.3 start: `node index.js`
npm ERR! Exit status 1

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531
Adapter firmware version: unknown

Koenkk, 2020-10-05:
Please provide the herdsman debug logging of this issue.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Baggerfahrer, 2020-10-05:
Hey! Thanks fornthe help.
See attached the requested logfile...

`
Zigbee2MQTT:info  2020-10-05 21:12:23: Starting Zigbee2MQTT version 1.15.0 (commit #1bccc5d)

Zigbee2MQTT:info  2020-10-05 21:12:23: Starting zigbee-herdsman...

Zigbee2MQTT:debug 2020-10-05 21:12:23: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[13],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyUSB-zigbee2mqtt"}}'

2020-10-05T19:12:24.600Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[13]},"serialPort":{"path":"/dev/ttyUSB-zigbee2mqtt"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null}}'

2020-10-05T19:12:24.610Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB-zigbee2mqtt and {"baudRate":115200,"rtscts":false,"autoOpen":false}

2020-10-05T19:12:24.629Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened

2020-10-05T19:12:25.103Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload

2020-10-05T19:12:25.103Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]

2020-10-05T19:12:26.112Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}

2020-10-05T19:12:26.121Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]

2020-10-05T19:12:26.907Z zigbee-herdsman:adapter:zStack:znp:log Port closed

Zigbee2MQTT:error 2020-10-05 21:12:32: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2020-10-05 21:12:32: Failed to start zigbee

Zigbee2MQTT:error 2020-10-05 21:12:32: Exiting...

Zigbee2MQTT:error 2020-10-05 21:12:32: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)

    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)

    at Generator.throw (<anonymous>)

    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

npm ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`

npm ERR! Exit status 1

 

npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.

npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

npm ERR!     /root/.npm/_logs/2020-10-05T19_12_32_231Z-debug.log

Using '/app/data' as data directory

> zigbee2mqtt@1.15.0 start /app

> node index.js

Zigbee2MQTT:info  2020-10-05 21:12:38: Logging to console and directory: '/app/data/log/2020-10-05.21-12-38' filename: log.txt

Zigbee2MQTT:debug 2020-10-05 21:12:39: Removing old log directory '/app/data/log/2020-10-05.21-08-33'

Zigbee2MQTT:debug 2020-10-05 21:12:40: Loaded state from file /app/data/state.json`

Koenkk, 2020-10-06:
- Can you share your output of `dmesg` after this error?
- What is your setup (computer/os)?

skumka, 2020-10-06:
@Koenkk I have the same issue.
Here is dmesg
[  428.317946] usb 1-1.4: SerialNumber: __0X00124B0014D4802A
[  428.323067] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  477.546157] usb 1-1.4: USB disconnect, device number 13
[  477.546710] cdc_acm 1-1.4:1.0: failed to set dtr/rts
[  477.859524] usb 1-1.4: new full-speed USB device number 14 using xhci_hcd
[  478.008492] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[  478.008512] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  478.008528] usb 1-1.4: Product: TI CC2531 USB CDC
[  478.008543] usb 1-1.4: Manufacturer: Texas Instruments
[  478.008558] usb 1-1.4: SerialNumber: __0X00124B0014D4802A
[  478.013667] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[  494.188637] usb 1-1.4: USB disconnect, device number 14
[  494.189087] cdc_acm 1-1.4:1.0: failed to set dtr/rts
[  494.489714] usb 1-1.4: new full-speed USB device number 15 using xhci_hcd
[  494.648263] usb 1-1.4: New USB device found, idVendor=0451, idProduct=16a8, bcdDevice= 0.09
[  494.648271] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  494.648278] usb 1-1.4: Product: TI CC2531 USB CDC
[  494.648284] usb 1-1.4: Manufacturer: Texas Instruments
[  494.648290] usb 1-1.4: SerialNumber: __0X00124B0014D4802A
[  494.653269] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

Looks like when ZigBee during start  try to connect adapter it disconects

petermolnar, 2020-10-06:
I'm facing the same problem in FreeBSD. 

- device inserted or machine booted
- first start of zigbee2mqtt fails with the error message above
- device seem to reset and move tty ports (ttyU port in my case)
- once repointed to the "new" device, zigbee2mqtt starts

This is not a new problem at all; it manifested for me months ago, and so far I haven't found a solution, only workarounds (such as expecting the failure starting zigbee2mqtt twice instead of once in the init script)

skumka, 2020-10-06:
@Koenkk 
I have a fresh install on Raspberry pi 4.
OS - Raspbian
Mosquitto and Zigbee NOT in docker.
Yesterday everything worked normally, but after 7 hours I noticed I have a growing number of devices not linked to any router or controller. I run the network map a few times. 
Devices hung in the "air" without any link.
Over the night something worse has happened. ZigBee hung itself. After restart Zigbee is not starting at all.
I have 5 routers and 54 devices discovered.

Koenkk, 2020-10-06:
@skumka looks that your CC2531 crashes which is not strange with so much devices. To improve stability a bit you can try switching the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing. If that doesn't help I recommend a more powerful adapter (https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter)

Baggerfahrer, 2020-10-06:
Maybe as well interesting for Findung the root cause:

I use a rpi 8gb with the actual Beta OS. See:

hostnamectl                                                     Static hostname: raspberrypi
         Icon name: computer
        Machine ID: de8542661a924694b577f09f87ccdbd7
           Boot ID: b4b828fbaddd424290214c6d9a84e407
  Operating System: Debian GNU/Linux 10 (buster)
            Kernel: Linux 5.4.51-v8+
      Architecture: arm64

petermolnar, 2020-10-06:
> @skumka looks that your CC2531 crashes which is not strange with so much devices. 

In my case, it crashes exactly one time; when the service is first started either after plugging it in or after the machine is rebooted - so when the dongle is power cycled.

It doesn't look like a hardware error, it would be crashing randomly then.

Koenkk, 2020-10-06:
@Baggerfahrer can you check if modemmanager is installed (https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed) and do you maybe run openhab?

@petermolnar can you try creating an udev rule to assign a static path? (https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port)

skumka, 2020-10-07:
@Koenkk 
> @skumka looks that your CC2531 crashes which is not strange with so much devices. To improve stability a bit you can try switching the source routing firmware: [...]
Koen, Is CC2531 an issue for 55 devices if it works with 5 routers? I did read an evalutaion telling that for coordinator I should consider 13 devices and for each router 21. With 5 routers it gives 100+ devices...
Per your answer I understand the issue for such a configuration is the bandwidth of processing messages by CC2531. Is this a real problem?
2nd. If I migrate to more powerful coordinator. What should I do to preserve data of already discovered device and have HA working without the need for re-discovering and automating all from the scratch? I see comment about preserving padding... What does it really means?

petermolnar, 2020-10-07:
> @petermolnar can you try creating an udev rule to assign a static path? 

FreeBSD. There's no udev. There's devfs, which should serve the same function, but no matter what I tried, the reset keeps happening. I'll try to dig deeper into the rulesets.

Baggerfahrer, 2020-10-07:
> @Baggerfahrer can you check if modemmanager is installed (https://www.zigbee2mqtt.io/information/FAQ.html#modemmanager-is-installed) and do you maybe run openhab?

Package 'modemmanager' is not installed, so not removed

And yes, openhab ist installed via docker... But thus Setup did run perfectly for some months already...

Koenkk, 2020-10-07:
@Baggerfahrer make sure to disable the zwave binding in openhab (https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245)

Baggerfahrer, 2020-10-09:
> @Baggerfahrer make sure to disable the zwave binding in openhab (https://community.openhab.org/t/apparently-the-zwave-binding-blocks-the-dev-ttyusb0-port-in-combination-with-a-cc2652rb-zigbee2mqtt-dongle/103245)

Thanks for this answer! Unfortunately, this does not help.
I either deactivated the zwave binding in openhab or even shutdown the docker container of openhab. but both actions did not solve the issue, the error message is still the same.

Any more tipps?

Baggerfahrer, 2020-10-09:
> I'm facing the same problem in FreeBSD.
> 
>     * device inserted or machine booted
> 
>     * first start of zigbee2mqtt fails with the error message above
> 
>     * device seem to reset and move tty ports (ttyU port in my case)
> 
>     * once repointed to the "new" device, zigbee2mqtt starts
> 
> 
> This is not a new problem at all; it manifested for me months ago, and so far I haven't found a solution, only workarounds (such as expecting the failure starting zigbee2mqtt twice instead of once in the init script)
> 
> dmesg example:
> 
> ```
> umodem2 on uhub1
> umodem2: <Texas Instruments TI CC2531 USB CDC, class 2/0, rev 2.00/0.09, addr 7> on usbus0
> umodem2: data interface 1, has no CM over data, has no break
> [...]
> umodem2: at uhub1, port 4, addr 7 (disconnected)
> umodem2: detached
> umodem2 on uhub1
> umodem2: <Texas Instruments TI CC2531 USB CDC, class 2/0, rev 2.00/0.09, addr 7> on usbus0
> umodem2: data interface 1, has no CM over data, has no break
> ```
> 
> before start the device was on `/dev/ttyU2`, after the reset, on `ttyU3`.
> 
> First run:
> 
> ```
> [domoticz@localhost /opt/zigbee2mqtt]$ npm start
> 
> > zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> > node index.js
> 
> Zigbee2MQTT:info  2020-10-06 16:00:34: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-06.16-00-34' filename: log.txt
> Zigbee2MQTT:info  2020-10-06 16:00:35: Starting Zigbee2MQTT version 1.15.0 (commit #e30b257)
> Zigbee2MQTT:info  2020-10-06 16:00:35: Starting zigbee-herdsman...
> Zigbee2MQTT:error 2020-10-06 16:00:42: Error while starting zigbee-herdsman
> Zigbee2MQTT:error 2020-10-06 16:00:42: Failed to start zigbee
> Zigbee2MQTT:error 2020-10-06 16:00:42: Exiting...
> Zigbee2MQTT:error 2020-10-06 16:00:42: Error: Failed to connect to the adapter (Error: Cannot request when znp has not been initialized yet)
>     at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
>     at Generator.throw (<anonymous>)
>     at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
> npm ERR! code ELIFECYCLE
> npm ERR! errno 1
> npm ERR! zigbee2mqtt@1.15.0 start: `node index.js`
> npm ERR! Exit status 1
> npm ERR! 
> npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
> npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
> 
> npm ERR! A complete log of this run can be found in:
> npm ERR!     /opt/domoticz/.npm/_logs/2020-10-06T15_00_42_881Z-debug.log
> ```
> 
> second run, after repointing the symlink to the new tty:
> 
> ```
> [domoticz@alia /opt/zigbee2mqtt]$ npm start
> 
> > zigbee2mqtt@1.15.0 start /opt/zigbee2mqtt
> > node index.js
> 
> Zigbee2MQTT:info  2020-10-06 16:01:56: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-10-06.16-01-56' filename: log.txt
> Zigbee2MQTT:info  2020-10-06 16:01:57: Starting Zigbee2MQTT version 1.15.0 (commit #e30b257)
> Zigbee2MQTT:info  2020-10-06 16:01:57: Starting zigbee-herdsman...
> Zigbee2MQTT:info  2020-10-06 16:02:07: zigbee-herdsman started
> Zigbee2MQTT:info  2020-10-06 16:02:07: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
> Zigbee2MQTT:info  2020-10-06 16:02:07: Currently 0 devices are joined:
> Zigbee2MQTT:warn  2020-10-06 16:02:07: `permit_join` set to  `true` in configuration.yaml.
> Zigbee2MQTT:warn  2020-10-06 16:02:07: Allowing new devices to join.
> Zigbee2MQTT:warn  2020-10-06 16:02:07: Set `permit_join` to `false` once you joined all devices.
> Zigbee2MQTT:info  2020-10-06 16:02:07: Zigbee: allowing new devices to join.
> Zigbee2MQTT:info  2020-10-06 16:02:07: Connecting to MQTT server at mqtt://localhost
> Zigbee2MQTT:info  2020-10-06 16:02:08: Connected to MQTT server
> Zigbee2MQTT:info  2020-10-06 16:02:08: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
> Zigbee2MQTT:info  2020-10-06 16:02:08: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"commit":"[...]"}'
> ```

as i use symlinks for my usb sticks, i originally have had a different target. after having read this quoted message, i also tried to change the config of zigbee2mqtt to the the originally tty, but this as well did not solve the issue. same error message again

Koenkk, 2020-10-09:
@Baggerfahrer 
- Are there any other USB devices attached? 
- Have you also tried reflashing the adapter (for CC2531 try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

Baggerfahrer, 2020-10-10:
> @Baggerfahrer
> 
>     * Are there any other USB devices attached?
> 
>     * Have you also tried reflashing the adapter (for CC2531 try the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing)

Hey @Koenkk 

many thanks for your tipps and comments.
I reflashed the adapter with your mentioned firmware and, apparently, it works again!
`
Zigbee2MQTT:info  2020-10-10 12:12:53: Starting Zigbee2MQTT version 1.15.0 (commit #1bccc5d)
Zigbee2MQTT:info  2020-10-10 12:12:53: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-10 12:12:53: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[13],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM2"}}'
Zigbee2MQTT:info  2020-10-10 12:12:55: zigbee-herdsman started
Zigbee2MQTT:info  2020-10-10 12:12:55: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190619,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2020-10-10 12:12:55: Zigbee network parameters: {"channel":13,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2020-10-10 12:12:55: Currently 23 devices are joined:`

Great help, many thanks!

closed 2020-10-10 by Baggerfahrer