Koenkk/zigbee2mqtt #4558: Sonoff Basic zbr3 does not work in groups

olialb, 2020-10-04:
First thank you for this great project!!

## What happened
I added a Sonoff Basic zbr3 to a working group of older osram switches. I added the Sonoff successfully but it does not react on any command to the group. It simply keeps the state. On direct commands to the sonoff it works fine.

I assume this is an issue of the Sonoff SW, or can be something wrong with the group implementation of?
Did anybody else had the same problem?

## Reproduced like this:
I created also a test group with the sonoff only. Added successfully:
`info  2020-10-04 17:31:58: Adding 'SonoffBasic01' to 'Testgroup'
info  2020-10-04 17:31:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":{"friendly_name":"SonoffBasic01","group":"Testgroup"},"type":"device_group_add"}'
`

Than I send send a comand to the group (Set to Off):
`info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/state', payload 'OFF'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-friendlyName', payload 'SonoffBasic01'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-model', payload 'BASICZBR3'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-ieeeAddr', payload '0x00124b001b78ba70'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-networkAddress', payload '4891'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-type', payload 'Router'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-manufacturerID', payload '0'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-manufacturerName', payload 'SONOFF'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-powerSource', payload 'Mains (single phase)'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-applicationVersion', payload '2'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-stackVersion', payload ''
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-zclVersion', payload '1'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-hardwareVersion', payload '1'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-dateCode', payload '20191113'
info  2020-10-04 17:44:24: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-softwareBuildID', payload ''`

**No reaction of the sonoff switch!**

## Debug info
Zigbee2MQTT version: 15.0
Adapter hardware: not sure: CC2531
Adapter firmware version: 2019-06-08

olialb, 2020-10-10:
hey @Koenkk,

thanks a lot for locking in this issiue. I just tried it with the dev branch. 
Put the SOnoff back in my Testgroup and send an ON command to the group. **No reaction of the switch**.

Here the requested log:

`  zigbee-herdsman:controller:group Command 18 genLevelCtrl.moveToLevelWithOnOff({"level":254,"transtime":10}) +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000012","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":8,"transid":2,"options":0,"radius":30,"len":6,"data":{"type":"Buffer","data":[17,3,4,254,10,0]}} +17s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,26,36,2,1,18,0,0,0,0,0,0,0,255,0,0,1,8,0,2,0,30,6,0,17,3,4,254,10,0,33] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +17s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,2,196] +4ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,2,196] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,2] - 196 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":2} +18s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/brightness', payload '254'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/Testgroup/brightness', payload '254'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/linkquality', payload '39'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/Testgroup/state', payload 'ON'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/state', payload 'ON'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-friendlyName', payload 'SonoffBasic01'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-model', payload 'BASICZBR3'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-ieeeAddr', payload '0x00124b001b78ba70'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-networkAddress', payload '4891'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-type', payload 'Router'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-manufacturerID', payload '0'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-manufacturerName', payload 'SONOFF'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-powerSource', payload 'Mains (single phase)'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-applicationVersion', payload '2'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-stackVersion', payload ''
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-zclVersion', payload '1'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-hardwareVersion', payload '1'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-dateCode', payload '20191113'
Zigbee2MQTT:info  2020-10-10 08:39:50: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-softwareBuildID', payload ''
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,5,69,196,27,19,1,115,69,187] +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,5,69,196,27,19,1,115,69,187] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 5 - 2 - 5 - 196 - [27,19,1,115,69] - 187 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":4891,"relaycount":1,"relaylist":[17779]} +3s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,27,19,1,47,0,81,0,105,157,19,0,0,7,8,3,10,0,0,16,0,115,69,28,116] +12ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,27,19,1,47,0,81,0,105,157,19,0,0,7,8,3,10,0,0,16,0,115,69,28,116] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,27,19,1,47,0,81,0,105,157,19,0,0,7,8,3,10,0,0,16,0,115,69,28] - 116 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":4891,"srcendpoint":1,"dstendpoint":47,"wasbroadcast":0,"linkquality":81,"securityuse":0,"timestamp":1285481,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,3,10,0,0,16,0]}} +14ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":0}]},"address":4891,"endpoint":1,"linkquality":81,"groupID":0}' +22s
[... cut ...]
Zigbee2MQTT:info  2020-10-10 08:39:54: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-hardwareVersion', payload '1'
Zigbee2MQTT:info  2020-10-10 08:39:54: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-dateCode', payload '20191113'
Zigbee2MQTT:info  2020-10-10 08:39:54: MQTT publish: topic 'zigbee2mqtt/SonoffBasic01/device-softwareBuildID', payload ''
`

I hope I did everything  right. First time that I did this.
I also want to note that the nice zigbee2mgtt frontend did not work in the development branch...

Have a nice weekend!

Koenkk, 2020-10-10:
- Can you share a bit more of your log? I'm interested in what MQTT command is received, so current log + a bit what happend before.
- Can you share your `configuration.yaml`? (leave out personal details like urls/passwords).

olialb, 2020-10-10:
Hi @Koenkk 

my configuration.yaml:

`homeassistant: false
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: ...
  user: ...
  password: ...
  client_id: Zigbee
  reject_unauthorized: true
  include_device_information: true
advanced:
  network_key:
....
  log_level: info
  log_directory: data/log/%TIMESTAMP%
  log_output:
    - console
    - file
serial:
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___...
  disable_led: true
experimental:
  output: attribute
  new_api: true
frontend:
  port: 3001
groups: groups.yaml
devices: devices.yaml`

Part of my davices.yaml:

`
...
'0x00124b001b78ba70':
  friendly_name: SonoffBasic01
...
`
Part of my groups.yaml:

`'18':
  friendly_name: Testgroup
  retain: false
  transition: 1
  optimistic: false
  devices:
    - 0x00124b001b78ba70/1
`

Log file attached.

[SonoffBasicErrorLog.txt](https://github.com/Koenkk/zigbee2mqtt/files/5359457/SonoffBasicErrorLog.txt)

Koenkk, 2020-10-11:
Does it work when removing `transition` for that group?

olialb, 2020-10-11:
Just tried it with the V15 master **and it worked**!!
So better not using transition in a group with ON/OFF switches inside... I thought it will just be ignored.

Anyway. Thanks a lot!

olialb, 2020-10-12:
Should I close the issue or is it unexpected behaviour? My Osram switches seams to ignore the `transition` attribute.

Josov, 2020-10-12:
I think as long as the transition command issues an error or the zbr3 not to work it will cause trouble as it will happen. Of course these devices will be used in groups together with bulbs and the transition command will be used. Better get it fixed is my opinion.

Josov, 2020-10-13:
I tested the zbr3 in a group with bulbs. The zbr3 switches ON with no errors even with brightness, color and color_temp specified. This is good. However it does not switch OFF anymore. No errors are generated. Unfortunately no action as well.

Koenkk, 2020-10-13:
To understand why this happens, let me first explain how this works. Zigbee devices implement certain clusters having certain functionalities. For this problem the two relevant are:
- onOff cluster: implements commands to turn on/off the device
- level control cluster: implements commands to change the level (= brightness) of a device.

The level control cluster is a somewhat special cluster as it interacts with the on/off cluster. When a command is send (let's call it `setBrightness` in this example) it will influence the on/off cluster. If brightness > 0 it will turn it on, if brightness = 0 it will turn it off. This is very convenient, when a bulb is off and you want to turn it on at a certain brightness, you can simply call `setBrightness` and the bulb will turn on at the given brightness.

Back to Zigbee2MQTT, we map the MQTT payloads to Zigbee commands as follows:
- `{"state": "ON"}` -> `onOff(on)`
- `{"state": "OFF"}` -> `onOff(off)`
- `{"state": "ON", "brightness": 20}` -> `setBrightness(20)`
- `{"state": "ON", "brightness": 20}` -> `onOff(off)`

The problem here is that the Sonoff Basic zbr3 does not understand the `setBrightness` command. Your OSRAM switch seems to support this (that's why it works).

As can be seen in the table above, the only case were `setBrightness` is used is for `{"state": "ON", "brightness": 20}`. In order to make the Sonoff Basic zbr3  respond to this command, when sending `{"state": "ON", "brightness": 20}` we now send 2 Zigbee commands: `onOff(on)` followed by `setBrightness(20)`.

Adding transitions however brings this problem to another level. Instead of the `onOff` commands a `setBrightness` command is now always used, since only that command supports transitions. If we apply the same solution from above, we will get into very tricky situations. e.g. when sending `{"state": "ON", "brightness": 20,"transition": 10}` and we send: `onOff(on)` followed by `setBrightness(20)`, instead of having a transition of 20 seconds, the bulb will immediately turn on (because of the `onOff(on)`). 

I think the best solution here would be to keep the implementation as is and not do all kinds of hackery because only a limited amount of switches and users seem to be affected by this. Instead we should generate a warning in the Zigbee2MQTT log, mentioning that if the switch does not turn on it's because the `transition`.  

@Josov I'm still interested to see why it doesn't turn OFF in your case. If you could provide the herdsman debug logging of this I can investigate. To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

olialb, 2020-10-13:
@Koenkk Thanks for the detailed explaniation. I learned a lot about Zigbee with this project! 
I totally agree that the software should be kept generic and not implement work arrounds for purely implemented devices.

I also think most of the users will have another layer like OpenHab (in my case) or Home Assistant to bild groups over Zigbee groups. With this you can solve such issues easily.

closed 2020-10-17 by olialb

Josov, 2020-10-27:
@Koenkk I made a few more tests. Here is the result:

- {"state": "ON"} works, does switch on the zbr3
- {"state": "ON", "transition": t} does not work
- {"state": "ON", "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}} works
- {"state": "ON", "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}, "transition": t} **works as well!!**
- {"state": "OFF"} works
- {"state": "OFF", "transition": t} does not work
- {"state": "OFF", "brightness": 10, "transition": t} does not work
- {"state": "OFF", "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}} works
- {"state": "OFF",  "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}, "transition": t} **works!**

It looks like when "color" is added there is no problem with "transition". If color is omitted you can't add transition.
If you still need a debug logging Koen, let me know.

Koenkk, 2020-10-28:
Strange that `{"state": "OFF", "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}, "transition": t} works!` works, could you provide the herdsman debug logging of that?

Josov, 2020-11-02:
@Koenkk, 

Find below the logging as requested for {"state": "OFF", "brightness": 10, "color": {"x": 0.3855, "y": 0.1546}, "transition": 0} of group 'Test ZBR3' This group contains HUE Power Plug, IKEA Power Plug, the Basic ZBR3 and an IKEA CWS bulb.
 
`Zigbee2MQTT:debug 02-11-2020 10:07:31: Received MQTT message on 'zigbee2mqtt/Test ZBR3/set' with data '{"state":"OFF","brightness":10,"color":{"x":0.3855,"y":0.1546},"transition":0}'
Zigbee2MQTT:debug 02-11-2020 10:07:31: Publishing 'set' 'transition' to 'Test ZBR3'
Zigbee2MQTT:debug 02-11-2020 10:07:31: Publishing 'set' 'color' to 'Test ZBR3'
  zigbee-herdsman:controller:group Command 22 lightingColorCtrl.moveToColor({"transtime":0,"colorx":25264,"colory":10132}) +33s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000016","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":768,"transid":34,"options":0,"radius":30,"len":9,"data":{"type":"Buffer","data":[17,12,7,176,98,148,39,0,0]}} +31s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,29,36,2,1,22,0,0,0,0,0,0,0,255,0,0,1,0,3,34,0,30,9,0,17,12,7,176,98,148,39,0,0,159] +31s
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,34,228] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,34,228] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,34] - 228 +1ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":34} +31s
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
Zigbee2MQTT:info  02-11-2020 10:07:32: MQTT publish: topic 'zigbee2mqtt/HUE PP 1', payload '{"brightness":80,"color":{"x":0.3855,"y":0.1546},"color_temp":44,"device":{"applicationVersion":2,"dateCode":"20200124","friendlyName":"HUE PP 1","hardwareVersion":1,"ieeeAddr":"0x0017880108619bc6","manufacturerID":4107,"manufacturerName":"Philips","model":"929002240401","networkAddress":30179,"powerSource":"Mains (single phase)","softwareBuildID":"1.65.9_hB3217DF4","stackVersion":1,"type":"Router","zclVersion":2},"last_seen":"2020-11-02T10:06:58+01:00","linkquality":30,"state":"ON","update_available":false}'
Zigbee2MQTT:info  02-11-2020 10:07:32: MQTT publish: topic 'zigbee2mqtt/Test ZBR3', payload '{"brightness":80,"color":{"x":0.3855,"y":0.1546},"color_temp":44,"state":"ON"}'
Zigbee2MQTT:debug 02-11-2020 10:07:32: Publishing 'set' 'state' to 'Test ZBR3'
  zigbee-herdsman:controller:group Command 22 genOnOff.off({}) +226ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequestExt - {"dstaddrmode":1,"dstaddr":"0x0000000000000016","destendpoint":255,"dstpanid":0,"srcendpoint":1,"clusterid":6,"transid":35,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[17,13,0]}} +225ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,23,36,2,1,22,0,0,0,0,0,0,0,255,0,0,1,6,0,35,0,30,3,0,17,13,0,252] +225ms
Zigbee2MQTT:info  02-11-2020 10:07:32: MQTT publish: topic 'zigbee2mqtt/IKEA PP 1', payload '{"brightness":80,"color":{"x":0.3855,"y":0.1546},"color_temp":44,"device":{"applicationVersion":32,"dateCode":"20190308","friendlyName":"IKEA PP 1","hardwareVersion":1,"ieeeAddr":"0x086bd7fffe1cb729","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"E1603/E1702","networkAddress":44455,"powerSource":"Mains (single phase)","softwareBuildID":"2.0.022","stackVersion":98,"type":"Router","zclVersion":3},"last_seen":"2020-11-02T10:06:58+01:00","linkquality":33,"state":"ON","update_available":true}'
Zigbee2MQTT:info  02-11-2020 10:07:32: MQTT publish: topic 'zigbee2mqtt/L2', payload '{"brightness":80,"color":{"x":0.3855,"y":0.1546},"color_temp":44,"device":{"applicationVersion":19,"dateCode":"20200220","friendlyName":"L2","hardwareVersion":1,"ieeeAddr":"0x14b457fffe3bdbb8","manufacturerID":4476,"manufacturerName":"IKEA of Sweden","model":"LED1624G9","networkAddress":42184,"powerSource":"Mains (single phase)","softwareBuildID":"1.3.013","stackVersion":87,"type":"Router","zclVersion":1},"last_seen":"2020-11-02T08:36:29+01:00","linkquality":30,"state":"ON","update_available":false}'
Zigbee2MQTT:info  02-11-2020 10:07:32: MQTT publish: topic 'zigbee2mqtt/Sonoff Zigbee device', payload '{"brightness":80,"color":{"x":0.3855,"y":0.1546},"color_temp":44,"device":{"dateCode":"20190907-V102","friendlyName":"Sonoff Zigbee device","hardwareVersion":1,"ieeeAddr":"0x00124b001e7330d6","manufacturerID":0,"manufacturerName":"SONOFF","model":"BASICZBR3","networkAddress":48301,"powerSource":"Mains (single phase)","type":"Router","zclVersion":1},"last_seen":"2020-11-02T10:06:58+01:00","linkquality":33,"state":"ON"}'
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,2,0,103] +220ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,2,0,103] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 2 - [0] - 103 +1ms
  zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequestExt - {"status":0} +225ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
[... cut ...]
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +0ms
  zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,43,237] +3ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,43,237] +1ms
  zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,43] - 237 +0ms
  zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":43} +17ms
  zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [] +1ms
`

Koenkk, 2020-11-02:
@Josov thanks for the log, but this is not complete log of the command. It has to start at `Received MQTT message`