Koenkk/zigbee2mqtt #4561: Occasional flickering of iCasa lamps

vdkeybus, 2020-10-04:
## What happened

For the last 2 weeks, I am trying a CC2652R instead of the Conbee II. There is one highly annoying problem that was never there with the CC2531 and Conbee II: the 4 iCasa lights ICZB-B2FC125 (together in a group) flash occasionally, like once in 5 mins.

They are the only devices that seem to behave like that, and it never happened with the other coordinators. Not sure if the Conbee used Zigbee 3.0 (the slaesh CC2652 does), otherwise that would be a change too.

Also, when turning the lights are turned on on, they briefly light up at full power, rather than at the preset level. This does not always happen, but pretty always when they have been off for a longer time (> 5 mins).

Oct 05 01:39:16 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:16: Received Zigbee message from '0x00158d00013d3ce5', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":420025876,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Oct 05 01:39:31 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:31: Received Zigbee message from '0x00158d00013d320c', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":420025876,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Oct 05 01:39:34 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:34: Received Zigbee message from '0x00158d00013d3cfc', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":420025876,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Oct 05 01:39:35 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:35: Received Zigbee message from '0x00158d00013d3202', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":420025876,"imageType":4364,"manufacturerCode":4137}' from endpoint 1 with groupID 0
Oct 05 01:39:42 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:42: Received Zigbee message from '0x086bd7fffe03db78', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Oct 05 01:39:42 zacate npm[3125805]: Zigbee2MQTT:info  2020-10-05 01:39:42: MQTT publish: topic 'zigbee2mqtt/0x086bd7fffe03db78', payload '{"brightness":254,"linkquality":126,"state":"OFF","update_available":false}'
Oct 05 01:39:42 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:42: Received Zigbee message from '0x14b457fffe40fded', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Oct 05 01:39:42 zacate npm[3125805]: Zigbee2MQTT:info  2020-10-05 01:39:42: MQTT publish: topic 'zigbee2mqtt/0x14b457fffe40fded', payload '{"brightness":254,"linkquality":126,"state":"OFF","update_available":false}'
Oct 05 01:39:53 zacate npm[3125805]: Zigbee2MQTT:debug 2020-10-05 01:39:53: Received Zigbee message from '0x00158d00013d320c', type 'attributeReport', cluster 'lightingColorCtrl', data '{"currentY":24701}' from endpoint 1 with groupID 0
Oct 05 01:39:53 zacate npm[3125805]: Zigbee2MQTT:info  2020-10-05 01:39:53: MQTT publish: topic 'zigbee2mqtt/licht_polyvalente_ruimte_2', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":440,"state":"OFF"}'
Oct 05 01:39:53 zacate npm[3125805]: Zigbee2MQTT:info  2020-10-05 01:39:53: MQTT publish: topic 'zigbee2mqtt/0x00158d00013d320c', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":440,"linkquality":126,"state":"OFF"}'

No other light has this problem.

## What did you expect to happen

Light stays off when told to...

## Debug info

Coordinator firmware version: '{"meta":{"maintrel":1,"majorrel":2,"minorrel":7,"product":1,"revision":20200830,"transportrev":2},"type":"zStack3x0"}'

TilmanK, 2020-10-05:
Is this the light affected?

MQTT publish: topic 'zigbee2mqtt/licht_polyvalente_ruimte_2', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":440,"state":"OFF"}'

This is the only message that is sent in the logs provided. I tried to send the same payload to one of my lights, without any problem. What I observed is, that the color settings are sent before the brightness. Maybe your light switches on, when getting a color value.

Could you check what happens, when you send {"color":{"x":0.3805,"y":0.3769},"color_temp":440} as payload to the light while it's switched off?

redahb, 2020-10-09:
I can add to this, it is **not** a problem with the CC2652R stick. I have a group of 6 bulbs from the same manufacturer, and they did the exact same thing on my CC2531: Every so often the group of bulbs flashes. I just had the bulbs turned off from mains to prevent it, because I thought I had some script/automation error in Home Assistant. However on first glance I could not find anything in Home Assistant and I actually even removed every single trigger/script/automation I had on these bulbs. Yet the error persists.

vdkeybus, 2020-10-09:
This is curious. The problem does not occur anymore. The only things I can remember that have changed in the meantime are:

- updated HA to 0.115.6 (and subsequent restart);
- renewed Nabu Casa account login (deleted all in .cloud) due to expired keys;
- removed kodi automation.

With @redahb, my setup shares the use of groups for this light.

I took some logs again:

Oct 09 17:47:41 zacate npm[3201603]: Zigbee2MQTT:info  2020-10-09 17:47:41: MQTT publish: topic 'zigbee2mqtt/licht_polyvalente_ruimte_2', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":440,"state":"OFF"}'
Oct 09 17:47:41 zacate npm[3201603]: Zigbee2MQTT:info  2020-10-09 17:47:41: MQTT publish: topic 'zigbee2mqtt/0x00158d00013d3202', payload '{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_temp":440,"linkquality":30,"state":"OFF"}'

paulgbrichardson, 2021-06-09:
Did you ever get to the bottom of the issue.  I have four of these bulbs in my dining room and office and its driving me nuts!

vdkeybus, 2021-06-09:
No. All seemed fine on the z2m side.

We have somewhat gotten used to the flashes (about 3 in an evening), but it
is weird for the neighbours to see the lights flashing. Looks like we're
taking pictures at 4am.

It is also the only lamp type I have that exhibits this behaviour.

I simply did not find the time to engage with iCasa. Are you interested ?

paulgbrichardson, 2021-06-09:
thanks for the response.  I've raised a ticket with iCasa.  I couldn't see anything coming from my z2m instance either.  It feels like a heartbeat waking the lights up but honestly not a clue!

tungmeister, 2021-08-02:
@paulgbrichardson Did you get to the bottom of this? I'm also seeing the same issue.

paulgbrichardson, 2021-08-02:
> @paulgbrichardson Did you get to the bottom of this? I'm also seeing the same issue.

Afraid not.   iCasa responded and blamed the co-ordinator but as I can't see anything in the logs I haven't done any more troubleshooting.  The lights are in my office which I don't really use in the dark during the summer so I haven't had the lights on recently anyway.  

If I get chance soon I'll turn the lights on and turn off z2m and see if they still flicker on their own.

tungmeister, 2021-08-02:
@paulgbrichardson interestingly mine seem to have stopped. I'm currently running a cc2351 whilst waiting for my slaesh stick. I had an issue with it randomly crashing (mainly due to too many devices on the network) so I flashed it with the source routing firmware. Since I did that this morning I've been in the room all day (they're bedside lights and I'm renovating the room atm)  and they've not flickered once.

vdkeybus, 2021-08-04:
I also have observed that when you reset the entire system, the flashes go
away for a couple of days, but then they return.

tungmeister, 2021-08-04:
they're still fine (been in the room all day again) so that's at least 60 hours without any issues - I'll have to see if this changes over the coming week.

vdkeybus, 2021-08-04:
It will return. Rest assured.

I just did some tests, there are a few curious things going on.

In *test 1*, I turn on and off all 4 lights in a group:

zigbee2mqtt/0x00158d00013d3cfc
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"ON"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"ON"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":48,"state":"ON"}
zigbee2mqtt/0x00158d00013d3ce5
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"ON"}

zigbee2mqtt/0x00158d00013d3cfc
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":48,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3ce5
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}

After this, they regularly report their state:

zigbee2mqtt/0x00158d00013d3cfc
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":48,"state":"ON"}
zigbee2mqtt/0x00158d00013d3cfc
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":254,"color":{"x":0.3805,"y":0.4185},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":254,"color":{"x":0.4806,"y":0.4185},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":254,"color":{"x":0.4806,"y":0.4185},"color_mode":"color_temp","color_temp":404,"linkquality":48,"state":"OFF"}
...and a couple of mins. later...
zigbee2mqtt/0x00158d00013d320c
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":48,"state":"OFF"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3cfc
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3202
{"brightness":0,"color":{"x":0.4806,"y":0.4185},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d320c
{"brightness":0,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}

Observe:

1. Light 0x00158d00013d3ce5 is always missing in the reports. It does
respond to commands, though;
2. After a while, all lights report brightness 0 while 254 was set. But
they do have a state with brightness==254 and state==OFF;
3. Lights report different color values while they were never set by z2m.
E.g. light 0x00158d00013d3202 has an x color value that went from 0.3805 to
0.4806.

In *test 2*, I manually turn off the lights

zigbee2mqtt/0x00158d00013d3cfc/set {"brightness":"0","state":"OFF"}
zigbee2mqtt/0x00158d00013d3cfc
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}
zigbee2mqtt/0x00158d00013d3ce5/set {"brightness":"0","state":"OFF"}
zigbee2mqtt/0x00158d00013d3ce5
{"brightness":254,"color":{"x":0.3805,"y":0.3769},"color_mode":"color_temp","color_temp":404,"linkquality":51,"state":"OFF"}

Look how brightness is now 254 immediately after that. And lo and behold,
0x00158d00013d3ce5 suddenly does report again (only now - won't report
later anymore).

I see 2 possible options:
a) problem = light. But I can't understand at all why any of them (the
non-reporting 3e5 flashes just as much as the rest) would turn on and off.
It also looks more like a turn-on command quickly followed by a turn-off
command, and not like an electric disturbance. Also, iCasa would have been
out of business by now.
b) problem = coordinator. But it would be sending 'rogue' commands (maybe
remnants in a buffer) that don't end up in the logs.

One remaining item is the presence of this every minute or so from every
light:

zigbee2mqtt/bridge/logging {"level":"debug","message":"Received Zigbee
message from '0x00158d00013d3ce5', type 'commandQueryNextImageRequest',
cluster 'genOta', data
'{\"fieldControl\":0,\"fileVersion\":420025876,\"imageType\":4364,\"manufacturerCode\":4137}'
from endpoint 1 with groupID 0"}

Does this mean I need to update ?

tungmeister, 2021-08-04:
I don't think it's the lights  - when I first experienced the issue I thought it was only happening with one of the bulbs so I complained to Amazon and got another 2 pack sent out. The issue then continued with the new bulb I installed and I noticed it on the other one from the first pack - so I've had the issue with all 3 bulbs I've used now and how I ended up on this issue.

vdkeybus, 2021-08-05:
> I don't think it's the lights - when I first experienced the issue I
> thought it was only happening with one of the bulbs so I complained to
> Amazon and got another 2 pack sent out. The issue then [...]
The lamps could still have a software bug that affects them all, wouldn't
you think ? Or no bug at all for that matter. Apparently these lights are
running fine with other coordinators.

I think we need a sniffer for this.

vdkeybus, 2021-08-05:
Ok. I did some Zigbee network sniffing.

Additional things I noted:
1. the flashes occur approx. every 6..7 minutes;
2. they only affected 3 of the 4 lights (the seemingly unaffected one is
the 3202 light, not the 3ce5 that's not reporting its state)
3. a 'flash' is both brief turn-on when the light is off, as well as brief
turn-off when the light is on

Attached are a couple of logs that I terminated with 5 secs. after a
'flash'.

There's also logs of a turn-on as well as a turn-on and -off where in both
cases, all the lights briefly go to full brightness, to immediately go back
to nearly zero and then ramp up in 2s as requested.

tungmeister, 2021-08-05:
@Koenkk any thoughts?

vdkeybus, 2021-08-05:
This may need to be reopened. I didn't close, so I cannot reopen.

Not even sure if @Koenkk/zigbee2mqtt ***@***.***> is
subscribed to this.

paulgbrichardson, 2021-08-16:
@Koenkk any thoughts?

Koenkk, 2021-08-16:
If someone can share a Zigbee sniff of the moment it flashes I can look further into this (please provide the wireshark packet number the moment it flashes). https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

vdkeybus, 2021-08-16:
A couple of e-mails ago I actually did all that. Are there no attachments ?

Koenkk, 2021-08-16:
> […](#)

I don't see any, you can drag the sniff in the comment box.

vdkeybus, 2021-08-16:
Here you go.

[icasa_logs_1.zip](https://github.com/Koenkk/zigbee2mqtt/files/6994622/icasa_logs_1.zip)

Koenkk, 2021-08-17:
@vdkeybus thanks, but the sniffs are quite long, could you indicate exactly when the flickering happend in the situation where you send no commands and wait till it flickers? Please provide the wireshark packet no. of this.

[image]

sammyke007, 2021-08-18:
I have 5 iCasa lights in my house and had the blinking in the past. Yesterday, after a power outage in my house, they started blinking again. Log isn't showing anything. It does seems to happen once every 7 minutes like someone mentioned above.
I forgot what I did a couple of months ago to make it stop :-(.
Restarting Z2M or uplugging the bulbs doesn't seem to help.
Crazy thing is, that it blinks once every 6-7 minutes, but then stops blinking for an hour or longer and returns again later :-D

vdkeybus, 2021-08-18:
> @vdkeybus thanks, but the sniffs are quite long, could you indicate exactly when the flickering happend in the situation where you send no commands and wait till it flickers? Please provide the wireshark packet no. of this.

There's quite a lot of traffic coming in, and I cannot exactly pinpoint the transmission that caused the flicker.

I immediately stopped (within 5 to 10 secs.) any of my logs after observing a flicker event. **So the flicker will have happened in the last 5..10 secs of the logs I provided.**

After stopping a log, I started a new one. Hence the 7-minute intervals.

I will need to connect light detectors to the host running the sniffer to exactly determine the occurrence of a flicker, but that would obviously be a significantly more elaborate experiment. I can do it if really needed, but I hope the multiple logs of the last 10 secs will be sufficient.

Koenkk, 2021-08-19:
@vdkeybus what is the `networkAddress` of the problematic bulbs?

vdkeybus, 2021-08-19:
0x00158d00013d3cfc
0x00158d00013d320c
0x00158d00013d3202
0x00158d00013d3ce5

BTW Maybe we should re-open this issue ?

Koenkk, 2021-08-19:
That is the `ieeeAddr`, not the `networkAddress`. Anyway I've noticed that most of your sniffs end with a OTA request. I've added some extra debug logging for this in the dev branch. See https://www.zigbee2mqtt.io/information/debug.html on how to enable debug logging. Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Can you check that everytime the bulb flickers, you see a `Device '[MY_ICASA_BULB]' requested OTA` for this device?

sammyke007, 2021-08-19:
My bulbs have been flickering all day now, so I searched for my old CC2531, flashed it, configured ZBOSS, sniffing is working fine... and the flickering stopped for 30 minutes already...

sammyke007, 2021-08-19:
While sniffing (with filter enabled for only one of the problematic iCasa bulbs), I see that besides link statusses it also sniffs: 

- 9930	1969.901677	0x8bea	0x0000	ZigBee HA	57	ZCL OTA: Query Next Image Request, Seq: 24
- 10068	2018.473810	0x0000	Broadcast	ZigBee	51	Many-to-One Route Request, Dst: 0xfffc, Src: 0x0000

But these happen without the flickering for now.

vdkeybus, 2021-08-20:
The OTA requests and the flickers do not seem to be related.

All 4 iCasa lamps repeat the OTA requests every minute plus a little bit:

Aug 20 19:34:22 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:34:22: Device '0x00158d00013d3cfc' requested OTA
Aug 20 19:34:30 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:34:30: Device '0x00158d00013d3202' requested OTA
Aug 20 19:34:46 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:34:46: Device '0x00158d00013d320c' requested OTA
Aug 20 19:35:06 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:35:06: Device '0x00158d00013d3ce5' requested OTA
Aug 20 19:35:23 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:35:23: Device '0x00158d00013d3cfc' requested OTA
Aug 20 19:35:31 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:35:31: Device '0x00158d00013d3202' requested OTA
Aug 20 19:35:47 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:35:47: Device '0x00158d00013d320c' requested OTA
Aug 20 19:36:07 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:36:07: Device '0x00158d00013d3ce5' requested OTA
Aug 20 19:36:24 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:36:24: Device '0x00158d00013d3cfc' requested OTA
Aug 20 19:36:33 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:36:33: Device '0x00158d00013d3202' requested OTA
Aug 20 19:36:48 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:36:48: Device '0x00158d00013d320c' requested OTA
Aug 20 19:37:08 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:37:08: Device '0x00158d00013d3ce5' requested OTA
Aug 20 19:37:26 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:37:26: Device '0xccccccfffe26a5d9' requested OTA
Aug 20 19:37:26 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:37:26: Device '0x00158d00013d3cfc' requested OTA
Aug 20 19:37:34 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:37:34: Device '0x00158d00013d3202' requested OTA
Aug 20 19:37:50 zacate npm[1468358]: Zigbee2MQTT:debug 2021-08-20 19:37:50: Device '0x00158d00013d320c' requested OTA

Right now, they're about 10-ish seconds apart, so it's hard to tell if there's a correlation.

tungmeister, 2021-08-20:
I've not noticed a single flicker for the last 18 days with mine - In that time I've migrated to a slaesh stick and the flickering still hasn't returned. It's really strange as initially, I had it pretty much constantly before moving to the source routing fw on the cc2531.

Koenkk, 2021-08-21:
I doubt more and more this has something to do with z2m, in the sniff you don't see any commands being send to the bulb so they seem to do it on their own. Can somebody check if it also happens with the original gateway?

vdkeybus, 2021-08-21:
I think iCasa does not make gateways. Their website says: "Easy to add to your Zigbee gateway"

Is there a way to actually perform the OTA update ?

sammyke007, 2021-08-22:
They don't seem to have their own gateway AFAIK. Don't know how to perform OTA then...
Flickering has stopped over here for 72h now. The flickering began 5 days ago after a power outage and stopped all of a sudden when I got my sniffer setup ready 2 days later.

Koenkk, 2021-08-22:
@vdkeybus 

> Is there a way to actually perform the OTA update ?

If they provide them yes, otherwise no.

vdkeybus, 2021-08-28:
@kisseler If you're able to sniff the packets, could you check the Zigbee chipset manufacturer. I'm not sure, but the Extended ID for my lamps seems to be from Jennic.

ballwax, 2021-09-13:
Hi all,
just as Informationen, I have exactly the same issues with some Paulmann 929.63 and CC2531 stick.

kisseler, 2021-09-13:
I did not have time to go deeper in that issue with Paulman devices. So my solution was: I bought a iluminize Zigbee 3.0 LED Controller, replaced the Paulman controller and everything works as expected, no flickering, good response and additionally the possibility to connect a physical light switch (pulse).
I'm still curious what might have been the problem. If anyone needs my spare Paulman controllers, let me know. ;-)

paulgbrichardson, 2021-10-15:
@tungmeister  Thanks - that might be the route I take.  I'm on an old TI CC2530 and been thinking of upgrading as the number of devices grow.  fingers crossed it works!

vdkeybus, 2021-10-15:
But the issue is: I have the slaesh stick as well, and Ikea stuff in
between. So essentially the same as Daniel.

Daniel, where are the iCasa lights in your network topology ? With me,
they're at the edge and cannot communicate directly with the coordinator.

sammyke007, 2021-10-18:
> @tungmeister @sammyke007 have you solved your issues? I'm still having flickering and now its getting darker in the UK have the lights on more. Getting to the point where I might have to replace them with something more stable. Very frustrating.

For me it stopped by itself after a week or so...
I'm using a ZZH stick (CC2652R). It started after a power outage in my street.
Never found out why it started, never found out why it stopped :-D

xaziwb, 2021-11-01:
Same problem. I have 2 flickering bulbs:
- Oeeone Edison E27 (paired as Paulmann 50064 ???; but working); name: oeeoneBulb01
- Müller Licht Tint retro E27; name: retrotint01
Coordinator: Slaeshs Stick
See Network Map attached; Oeeone is near by the coordinator; Müller across the house.
[image]

johanvh83, 2021-11-15:
Same problem with 3 iCasa bulbs. After reset and re-joint the problem is gone for some days and then it restarts.
Informed manufacturer about my setup: HA and Z2M and they responded they don't support Z2M.
A guy from iCasa support uses HA with a Conbee II stick and never had problems.
Don't know if this is useful but this is my map:
[image]
The bulbs are connected straight to the coordinator: Slae.sh CC2652RB_stick
Hopefully a solution can be found because this is very annoying!
Please instruct if further information is necesarry.

closed 2021-12-23 by github-actions[bot]

ASNNetworks, 2022-01-18:
I wouldn't be surprised if more people had this issue. I also have this problem. Also a few bulbs are defective and never go above 20% brightness anymore. It will on software side, but the actual hardware won't output more lumens.

I also notice these bulbs are out of stock for months at many retailers. Currently there isn't a single retailer that sells them. I returned all the bulbs and got a refund. Too bad, since they had amazing temperatur and color with filament design. Haven't found a proper replacement...

sammyke007, 2022-03-08:
And its back after an update to v1.24. Really strange! It has been long gone this time, but only the iCasa lights keep flickering (to 100% brightness and back off in 1 or 2 seconds). I'll restart my Pi and see if it helps. Log is not showing anything special in debug mode except the OTA requests from the bulbs.

johanvh83, 2022-03-08:
I'm on 1.23.0-1 and I have to agree, it's weeks or months ago that the bulbs were flickering. Based on your post @sammyke007, I keep my 1.23.0-1

tungmeister, 2022-03-08:
I'm not seeing any issues with mine in 1.24

SvenVerheesen, 2022-04-06:
I'm on 1.22.2-1 and whenever i update to the newest version i'm experiencing the flickering. I reconnected them but didn't solve the problem.

jaspernlp, 2022-04-22:
I've found an issue with the code. For over a week it is now running fine, although after one reboot the problem reoccurred. Running with docker compose the node.js in the zigbee2mqtt container the process number of the node.js process is not the same after each start. I've seen process number 7, 8 and 9. If it reoccurs I will have a look into this. But for now I'm happy with the fix.

I had four bulbs with the following behaviour: every 6-7 minutes the light did briefly go on while the state was off and went briefly off while the state is on. Very frustrating as one of them is in the bedroom. Also the log (on debug level) shows every minute a request for a firmware update.

This were my thoughts on what is happening. It looks like the bulbs are running a developer/debug/test firmware or are in a developer/debug/test mode. There is no need in production to check for new firmware that often, once a day, once a week or even once a month will do the job. The interruption of the on/off state is an alert, telling the developer/tester that is something wrong. The alert is produced after x-number of times a request for an update was sent and no response was received by the bulb. 

The code as in 1.25 does not send a reponse in the following cases, the automatic update is disabled and if zigbee2mqtt does not support OTA for that device. The javascript code that I updated will send a response on every QueryNextImageRequest (if zigbee2mqtt knows device). The code for the actual OTA update has not been changed.

The code (after update to typescript) goes into https://github.com/Koenkk/zigbee2mqtt/blob/master/lib/extension/otaUpdate.ts as start of 
    @bind private async onZigbeeEvent(data: eventdata.DeviceMessage): Promise<void> {
on line 72. I hope someone (@Koenkk ?) can move this forward by creating a PR for it.

        if (data.type !== 'commandQueryNextImageRequest' || !data.device.definition)                                                               
            return; 
                                                                                                                               
        const ABORT = {code:0x95, descr:'ABORT'};
        const NO_IMAGE_AVAILABLE = {code:0x98, descr: 'NO_IMAGE_AVAILABLE'};
        const endpoint = data.device.zh.endpoints.find((e) => e.supportsOutputCluster('genOta'));

        let otaIsDisabled = settings.get().ota.disable_automatic_update_check;
        let supportsOTA = data.device.definition.hasOwnProperty('ota');                                                                            
        
        logger_1.default.debug(`Device '${data.device.name}', type: queryNextImageRequest, ` + 
            `device supports OTA: ${supportsOTA}, ` + 
            `system auto update disabled: ${otaIsDisabled}`); 

        let status = null;
        if (otaIsDisabled){
            status = ABORT;
        }
        if (!supportsOTA && status === null) {
            status = NO_IMAGE_AVAILABLE;
        }
        if (endpoint && status !== null) {
            await endpoint.commandResponse('genOta', 'queryNextImageResponse', {status: status.code});
            logger_1.default.debug(`Device '${data.device.name}', type: queryNextImageResponse, ` + 
                `state: 0x${status.code.toString(16)}, descr: '${status.descr}' `); 
        }
                                                                 
        if (supportsOTA && !otaIsDisabled) {
            // no changes to the code as in the repo
        }

        // Respond to the OTA request:
        // this code needs to be deleted as the code was not reached for each request and now at the start of the method

Koenkk, 2022-04-22:
@jaspernlp well found, can you apply the changes to otaupdate.ts from https://github.com/Koenkk/zigbee2mqtt/pull/12247/files and run a `npm run build` in the z2m directory and see if the issue is fixed? If not, I guess it also depends if we respond with `NO_IMAGE_AVAILABLE` or `ABORT`?

jaspernlp, 2022-04-22:
I don't have a development environment setup, if I had, I would have created the PR. Reviewing the change: looks like the response is also send if the ota process is started, which means that in that case there might be two responses, the updated itself and the ABORT/NO_IMAGE_AVAILABLE. As based on the setting of the automatic disabled flag I tested with both the ABORT and the NO_IMAGE_AVAILABLE response. Both automatic update disabled true and false ran a couple of days successful.

Koenkk, 2022-04-22:
@jaspernlp fixed that now, can you try to change `otaUpdate.js` on your local system to https://gist.github.com/Koenkk/474413ea391b3f38cc4f4f364cf5f226 and see if that fixes the issue?

jaspernlp, 2022-04-22:
I've copied the file into my deployment. Added one debug statement just before the response is sent. It logs now like:
Zigbee2MQTT:debug 2022-04-22 21:12:47: Received Zigbee message from 'lights/utilityRoom', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' fr
om endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-22 21:12:47: Device 'lights/utilityRoom' requested OTA
Zigbee2MQTT:debug 2022-04-22 21:12:47: Device 'lights/utilityRoom' is getting a response
Zigbee2MQTT:debug 2022-04-22 21:12:54: Received Zigbee message from 'lights/livingRoom/tv', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' 
from endpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-22 21:12:54: Device 'lights/livingRoom/tv' requested OTA
Zigbee2MQTT:debug 2022-04-22 21:12:54: Device 'lights/livingRoom/tv' is getting a response
Zigbee2MQTT:debug 2022-04-22 21:13:01: Received Zigbee message from 'lights/master', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from en
dpoint 1 with groupID 0
Zigbee2MQTT:debug 2022-04-22 21:13:01: Device 'lights/master' requested OTA
Zigbee2MQTT:debug 2022-04-22 21:13:01: Device 'lights/master' is getting a response
Zigbee2MQTT:debug 2022-04-22 21:13:17: Received Zigbee message from 'lights/wc', type 'commandQueryNextImageRequest', cluster 'genOta', data '{"fieldControl":0,"fileVersion":537199895,"imageType":4364,"manufacturerCode":4137}' from endpoi
nt 1 with groupID 0
Zigbee2MQTT:debug 2022-04-22 21:13:17: Device 'lights/wc' requested OTA
Zigbee2MQTT:debug 2022-04-22 21:13:17: Device 'lights/wc' is getting a response
So far no flashes/interruptions. I will post an update after a couple of days.

I cannot test an OTA update, my ikea lights are uptodate. Please check the code that it is not sending both firmware update and an ABORT or NO_IMAGE_AVAILABLE.

Koenkk, 2022-04-23:
@sammyke007 @SvenVerheesen did you have the `disable_automatic_update_check` option set to `true`? If yes, does disabling this fix it for you?

sammyke007, 2022-04-23:
[image]

Yes. The flickering is gone again. It only happens/starts after a power surge or when I completely remove power myself. It also stops flickering by itself after a week or so.

Koenkk, 2022-04-23:
I see, merged the fix now so flickering should be gone for good in the dev branch (and next release on 1 May)

Changes will be available in the dev branch in a few hours from now. (https://www.zigbee2mqtt.io/advanced/more/switch-to-dev-branch.html)

jaspernlp, 2022-04-23:
I still think the last block should be wrapped to avoid responding while the ota is started/update as well. I run the code as:
        if (endpoint && (!supportsOTA || automaticOTACheckDisabled)) {                                                                                                                                                                        
            // Some devices send OTA requests without defining OTA cluster as input cluster.                                                                                                                                                  
            logger_1.default.debug(`Device '${data.device.name}' is getting a response`);                                                                                                                                                     
            await endpoint.commandResponse('genOta', 'queryNextImageResponse', { status: supportsOTA ? 0x95 : 0x98 });                                                                                                                        
        }

Koenkk, 2022-04-24:
@jaspernlp that is already prevented by https://github.com/Koenkk/zigbee2mqtt/blob/808671dd598c6fe5b7266b44755436b8f620bbeb/lib/extension/otaUpdate.ts#L74

jaspernlp, 2022-04-24:
Thanks for the explanation. Through out the weekend I've removed the power from the pi, restarted the pi, removed the power from 2 of the bulbs, changed the disable automatic update  setting. I have not seen any flickering. I'm happy with your change in the code.

[3 more replies]