Koenkk/zigbee2mqtt #4564: MQTT message to request property on /get of a group fails

TilmanK, 2020-10-05:
## What happened
Reading properties via /get topic on MQTT fails, when used with a group.

#033[34mZigbee2MQTT:debug#033[39m 2020-10-05 18:05:23: Received MQTT message on 'zigbee2mqtt/Küche/Oberschränke/get' with data '{"brightness":255,"state":"on"}'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-05 18:05:23: Publishing get 'get' 'state' to 'Küche/Oberschränke'
#033[31mZigbee2MQTT:error#033[39m 2020-10-05 18:05:23: Publish 'get' 'state' to 'Küche/Oberschränke' failed: 'TypeError: entity.read is not a function'
#033[34mZigbee2MQTT:debug#033[39m 2020-10-05 18:05:23: TypeError: entity.read is not a function
    at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:582:34)
    at EntityPublish.onMQTTMessage (/opt/zigbee2mqtt/lib/extension/publish.js:262:37)
    at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44)
    at Controller.onMQTTMessage (/opt/zigbee2mqtt/lib/controller.js:268:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/opt/zigbee2mqtt/lib/mqtt.js:94:14)
    at MqttClient.emit (events.js:315:20)
    at MqttClient._handlePublish (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:1271:12)
    at MqttClient._handlePacket (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:410:12)
    at work (/opt/zigbee2mqtt/node_modules/mqtt/lib/client.js:321:12)

## What did you expect to happen
Either a useful error message or it works (maybe just read the requested property of the groups first device?)

## How to reproduce it (minimal and precise)
Send a msg to zigbee2mqtt/[group_name]/get with proper formatted JSON payload.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Koenkk, 2020-10-06:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-06 by Koenkk

TilmanK, 2020-10-06:
Nice, thank you ;)