Koenkk/zigbee2mqtt #4567: Strip color properties based on color_mode

sjorge, 2020-10-05:
## Is your feature request related to a problem? Please describe
Now that color_mode is correctly being reported, it would be nice to have an option to strip the non active color options.

```{"brightness":254,"color":{"hue":0,"saturation":100,"x":0.7006,"y":0.2993},"color_mode":1,"linkquality":47,"state":"OFF"}```

We can remove either hue/sat or x/y based on color_mode, and color_temp too.
Might be good to gate this behind an option though so the user can choose.

## Describe the solution you'd like
Using filter_attributes,  it this is hard coded and won’t change if the color_mode changes.

## Describe alternatives you've considered
 Currently filtering in node-red but the code is in multiple paths and it’s rather a pain.

## Additional context
Add any other context or screenshots about the feature request here.

Koenkk, 2020-10-06:
Instead of stripping I prefer to have an additional conversion to keep all properties up to date, e.g. set hue/saturation will also be reflected in color_temp and x/y.

sjorge, 2020-10-06:
Not sure we can convert these? Hue/Sat <-> xy might work but not sure we can get color_temp -> xy + hue/sat or the reverse.

Perhaps a toggle for both would be better, convert for those who want it and strip for those who don't?

sjorge, 2020-10-06:
@Koenkk how about `filtered_attributes` in 2 phases
- allow filtering of nested attributes e.g. color.hue or color.x,...
- allow a simple conditional syntax before filtering .e.g. `%(state==OFF)power` (where it will filter power when state = OFF)

Not sure on the exact syntax though, but the first one of this should not be to hard to implement I think (will give it go next weekend) for syntax I was either considering.

```
    filtered_attributes:
      - color.hue
      - color.saturation
```

or

```
    filtered_attributes:
      - color:
        - hue
        - saturation
```

The later seems easier to implement as we can abstract the filtering to a separate function that would then just call itself again on the subset of the data.

Making this more generic might overall be a better approach than just doing it for color_mode and puts control in the hands of the end user.

Koenkk, 2020-10-06:
> Not sure we can convert these? Hue/Sat <-> xy might work but not sure we can get color_temp -> xy + hue/sat or the reverse.

Yes, this is possible and in fact already done for xy <-> color temp (https://github.com/Koenkk/zigbee-herdsman-converters/blob/904c6019defb872a268cd09564fdd15739b4a3b4/converters/toZigbee.js#L900). I think it would be nice to have this conversion. When e.g. using Home Assistant (which uses x/y by default) and setting the color via hue/saturation, Home Assistant would show an incorrect color.

sjorge, 2020-10-06:
That would explain the odd behavior I am seeing with color_temp/color updates when there is no report of them coming in.

TilmanK, 2020-10-07:
> Currently filtering in node-red but the code is in multiple paths and it’s rather a pain.

Just out of interest @sjorge, since I'm using NodeRed too. What are you doing exactly? Passing the stuff to some Homekit node? And what are you doing with all the other stuff you might not need? 

I implemented a parser function for every device that implements a positive filter... Seems you do it the other way round.

sjorge, 2020-10-07:
I do some filtering on the mqtt input, then have a 'mapper' subflow for each device type.

I am using reporting I currently filter what I send back to homekit based on color_mode... if the bulb is in hue/sat mode I pass that without coverting, xy mode? I convert and pass it back, color temperature mode? I pass back color temperature.

If I don't filter and always pass all data homekit gets a bit confused.

closed 2020-11-15 by github-actions[bot]