Koenkk/zigbee2mqtt #4568: Groups view crashes frontend if a device has been removed

nbostrom, 2020-10-06:
# Bug Report
## What happened
Fronted crashes when clicking the "Groups" link after forcefully removing a device that was included in a group.

## What did you expect to happen
I expected the groups view to render.

## How to reproduce it (minimal and precise)
1. Forcefully remove a device that is included in a zigbee group.
2. Click the "Groups" menu link.

## Debug Info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2652RB
Adapter firmware version: unknown.. don't think it matters in this case

nbostrom, 2020-10-06:
It might be a good idea to at least wrap every view in a [ErrorBoundary](https://reactjs.org/blog/2017/07/26/error-handling-in-react-16.html#introducing-error-boundaries) so that an error won't crash the whole ui.

Koenkk, 2020-10-06:
Please create frontend issues here: https://github.com/nurikk/z2m-frontend/issues (I will update https://github.com/Koenkk/zigbee2mqtt/issues/new/choose to better indicate this).

closed 2020-10-06 by Koenkk

nbostrom, 2020-10-06:
@Koenkk aight, thanks, will do that! You might want to update this comment as well: https://github.com/Koenkk/zigbee2mqtt/issues/4266#issue-693381866

Koenkk, 2020-10-06:
@nbostrom thanks, done!