Koenkk/zigbee2mqtt #4569: Xiaomi smoke route problems

pedroke, 2020-10-06:
## What happened

Coordinator on ZZH! running the latest firmware CC26X2R1_20200925.zip and the latest zigbee2mqtt 1.15.0.
I've already reported the issue for the previous version of the firmware but it seems to me that the problem persists.

When I pair the smoke detector for a very short period of time it shows as attached in the network map and it is possible to send a message to it e.g. self-test. But then a few seconds later it is suddenly detached and unable to receive messages again with "No route" error.
I suppose that the coordinator removes it from the neighbor list and also from the routing table.

I tried to sniff the traffic and the only thing that might be related I see that there is are "Route request" issued by the coordinator and this request is not answered by a smoke detector.

I know that the fix made in the firmware adapted the timeout for indirect messages to be 20 seconds. But does this timeout also apply to the "Route request" message? Is it possible to set the longer timeout for such requests?
I don't know why the coordinator decides to check the route, but it is nor related to the network map because it happens no independently whether I trigger a map update or not...

## What did you expect to happen

Expected is that the detector remains attached in the neighbor list and able to receive incoming messages.

## How to reproduce it (minimal and precise)

please see the above.

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: ZZH!
Adapter firmware version: CC26X2R1_20200925

Koenkk, 2020-10-06:
Can you provide the herdsman debug logging starting from the point where selftest succeeds until it fails?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

pedroke, 2020-10-07:
Hi Koen,
Thanks a lot for your help. Please see the logs: https://hastebin.com/imumeqiker.yaml
I force-removed the device and then I pressed the button 3 times to join.
The log begins when the smoke detector joins the network. Then I had 2 successful self-tests and then after few seconds, no self-test was possible anymore with NO ROUTE error. As I can see in the log there are several problems. It seems that the interview was not completely successful (Interview - completed for device '0x00158d0002cb0c6a' because of quirks), probably because of IAS related stuff and there are also some data confirmations missing, which probably triggered the route discovery process and removal of the device from the neighbor list. All in all, it seems that this device is really having some problems working properly in the network.  I also don't understand why there are two interviews captured, maybe because of unsuccessful first one.. Anyway, do you have any idea what might be going wrong?

Koenkk, 2020-10-07:
Somehow the hastebin is empty, can you post it on pastebin instead?

[image]

pedroke, 2020-10-07:
Hi Koen,
strange.. so another try:

https://pastebin.com/zTyY4CV1

line 737 is interesting and then failures about route and data confirmations missing..

pedroke, 2020-10-07:
Hmm..I studied the logs little bit deeper and it seems that device itself asked to be removed:

2020-10-07T09:31:28.890Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - leaveInd - {"srcaddr":54736,"extaddr":"0x00158d0002cb0c6a","request":0,"removechildren":0,"rejoin":0}
2020-10-07T09:31:28.890Z zigbee-herdsman:controller:log Device leave '0x00158d0002cb0c6a'
2020-10-07T09:31:28.890Z zigbee-herdsman:controller:log Removing device from database '0x00158d0002cb0c6a'

It happened after controller asked for the fifth time for ias zone state 'after enrolling'.
I don't understand why it asks in the loop for the ias zone state?Is it supposed to be changed? Do I understand correctly that the device itself asked with the leave request?

pedroke, 2020-10-07:
By studying herdsman code I found that the device is supposed to send enroll request... which does not happen here.. That's why coordinator checks for status change... Instead device removes itself from the network... Very strange. Could it be that the device does not like zigbee 3.0?

pedroke, 2020-10-08:
Could it be that the problem is related to https://github.com/Koenkk/zigbee-herdsman/commit/4b0c1e8266cbf85c2416d259a0d439447790f4ef#diff-3a4a6c17eb7d440e4529b0721a7836fa
I googled somewhere that IAS enrollment starts when coordinator sends "enroll response" and then device replies with "enroll request" and coordinator again with "enroll response".. I know it does not sound logical but it would explain why the smoke detector is not sending the enroll request, since in the old code it seems that coordinator was sending enroll response without prior request and maybe it was necessary to start the enrollment. Could it be?

Koenkk, 2020-10-08:
That's indeed what should happen in the enrolment, but probably the Xiaomi device doesn't do this because Xiaomi devices are often not compliant with the Zigbee spec. I've tested this device with my ZZH CC2652R and it works with that, maybe the firmware version on the smoke sensors are different.

Can you try disabling the enrollment code? You can do this by editing `/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/controller/model/devices.js` and change `for (const endpoint of this.endpoints.filter((e) => e.supportsInputCluster('ssIasZone'))) {` to `for (const endpoint of this.endpoints.filter((e) => e.supportsInputCluster('ssIasZoneDONT ENROLL'))) {`

pedroke, 2020-10-08:
Thanks Koen I will try it. I will also try to send ias response first by coordinator just to see if it trigger the process. 
It's pitty that I cannot factory reset the smoke detector. Even when I take out the batteries and discharge capacitors I think it is still in different state than it was as new...

pedroke, 2020-10-08:
Hi Koen,
Here's the take out from ZCL specification:

IAS Zone servers SHALL support either:
 Trip-to-pair AND Auto-Enroll-Response, OR
 Auto-Enroll-Request
An IAS Zone client SHALL support either: 
 Trip-to-pair AND Auto-Enroll-Response, OR
 Auto-Enroll-Request

Auto-Enroll-Response
1. After an IAS Zone server (end device) is commissioned to a network, the IAS CIE (coordinator) MAY perform service dis-
covery.
2. If the IAS CIE determines it wants to enroll the IAS Zone server, it SHALL send a Write Attribute 
command on the IAS Zone server’s CIE_IAS_Address attribute with its IEEE address.
3. The IAS Zone server MAY configure a binding table entry for the IAS CIE’s address because all of 
its communication will be directed to the IAS CIE.
4. The IAS CIE SHALL send a Zone Enroll Response, which assigns the IAS Zone server’s ZoneID
attribute.
5. The IAS Zone server SHALL change its ZoneState attribute to 0x01 (enrolled).

As you can see in this scenario coordinator is supposed to send Enroll response right after it writes CIE adress.
In the current implementation in device.ts it is missing. Before your fix in march it was like that.
So I would propose to add the following after CIE address is written:

await Wait(300); 
const payload = {enrollrspcode: 0, zoneid: 23};
await endpoint.command('ssIasZone', 'enrollRsp', payload, {disableDefaultResponse: true}); 
debug.log(`Interview - IAS - sent enroll response`);
await Wait(300);

Basically the current implementation provides Auto-Enroll-Request but also Auto-Enroll-Response should be supported.

Koenkk, 2020-10-09:
Can you test if that works by modifying your device.js locally? Code for sending the enroll response can be found in the `onZclData` method: https://github.com/Koenkk/zigbee-herdsman/commit/4b0c1e8266cbf85c2416d259a0d439447790f4ef#diff-3a4a6c17eb7d440e4529b0721a7836faR170

pedroke, 2020-10-09:
Hi Koen,

so I tested. I edited device.js and added the following code after CIE address is written  and just before the loop for checking is starts:

                    debug.log("Interview - IAS - sending enroll response (Auto enroll response)");
                    yield utils_1.Wait(500);
                    const payload = { enrollrspcode: 0, zoneid: 23 };
                    yield endpoint.command('ssIasZone', 'enrollRsp', payload, { disableDefaultResponse: true });

                    let enrolled = false;
                    for (let attempt = 0; attempt < 20; attempt++) {

After my change, the interview was successful and smoke detector set the zoneStatus to 1 (stripped version):

2020-10-09T08:38:59.093Z Interview - IAS - before enrolling state: '{"iasCieAddr":"0x0000000000000000","zoneState":0}'
2020-10-09T08:38:59.093Z Interview - IAS - not enrolled, enrolling
2020-10-09T08:38:59.097Z Write 0x00158d0002cb0c6a/1 ssIasZone({"iasCieAddr":"0x00124b0021600658"}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) 
2020-10-09T08:39:02.325Z Interview - IAS - wrote iasCieAddr
2020-10-09T08:39:02.326Z Interview - IAS - sending enroll response (Auto enroll response)
2020-10-09T08:39:02.830Z zigbee-herdsman:controller:endpoint Command 0x00158d0002cb0c6a/1 ssIasZone.enrollRsp({"enrollrspcode":0,"zoneid":23}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})  
2020-10-09T08:39:08.626Z Interview - IAS - after enrolling state (0): '{"iasCieAddr":"0x00124b0021600658","zoneState":1}'
2020-10-09T08:39:08.626Z Interview - IAS successfully enrolled '0x00158d0002cb0c6a' endpoint '1'
2020-10-09T08:39:08.627Z Interview - completed for device '0x00158d0002cb0c6a' 

So as you can see as soon as sensor received enroll response from coordinator, it set its zoneState to 1 and it was detected on the first run and interview was considered successful. So it seems that my version of detector really expects method defined as "Auto enroll response" in ZCL documentation. Maybe it would be good if herdsman supports both of those methods. If it sends the enroll response just after writing CIE address it should trigger the process for devices which expect it and devices which are using different method should be hopefully fine with it...

Koenkk, 2020-10-09:
Great, implemented. Can you confirm?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

pedroke, 2020-10-10:
Cool! Confirmed. Seems to be working as expected. Thanks a lot.
Hopefully it will not interfere with devices which use different enroll mechanism. Specification seems to be pretty loose.

Anyway. Regarding different issue. It seems to me that ZZH firmware is still not able to deliver messages to longer sleeping devices correctly. Although you changed the timeout to 20 seconds for indirect messages, I'm getting error code 240 = MAC transaction expired after 7-8 seconds and it triggers resending after 2s delay.. when it fails again, device is marked as unreachable and removed from tables.. So it is matter of luck, whether devices like smoke detector even catch such messages when waking up every 15 seconds.. I will investigate deeper what other possibilities are there for firmware to set the timeout for MAC transaction to be longer and if I find something I will open the new issue. I just wanted to let you know that there is probably such issue with Zigbee 3 firmware. Thanks a lot for your great work!

Koenkk, 2020-10-10:
I've changed the timeout based on https://e2e.ti.com/support/wireless-connectivity/zigbee-and-thread/f/158/p/919346/3402037#3402037 so I expect it to work. After applying this I've tested this multiple times with my Xiaomi sensor and it worked as expected (or I was incredible lucky, which I doubt).

However I believe if you restore a backup (`data/coordinator_backup.json`) of a non 20200925 firmware the timeout values are kept to 7 seconds. Can you try if it works with a freshly flashed 20200925 firmware WITHOUT restoring a `data/coordinator_backup.json` (just remove or rename that file).

pedroke, 2020-10-10:
Aaaah.. that could be a trick..I will try..Thanks for hint!

pedroke, 2020-10-10:
Hmmm..loking at the backup code it seems to me it will be part of NIB part. By the way it is very tempting for me to implement also the backup and restore of device list ( ZCD_NV_DEVICE_LIST), binding table(ZCD_NV_BINDING_TABLE), etc... if possible, otherwise it is always necessary to pair all devices after firmware update to have correct network map, routing, binding..

pedroke, 2020-10-11:
Fixed in dev by Koen.

closed 2020-10-11 by pedroke

sibero80, 2020-11-02:
I'm struggling as well to get the Smoke Sensor paired with my CC2531 Coordiantor Running Zigbee 3.0.
I'm at 1.16.0-dev branch, the logs dont even show the device showing up for pairing. (Battery and Sensor are OK).