Koenkk/zigbee2mqtt #4570: TUYA ZIGBEE LIGHT SWITCH

dror23, 2020-10-06:
## Information about the device + link
I bought the device from here:

https://www.aliexpress.com/item/4001181765345.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&tt=MG&aff_platform=default&sk=_dVohEV8&mergeHashcode=74176412736&description=%E2%82%AA+46.10++40%25OFF+%7C+ZigBee+smart+home++wall+touch+switch%2Cwireless+Intelligent+light+switch%2C2%2F3+Way%2CAPP+Control%2Cworks+with+google+home+aleax%2CUSUK&aff_trace_key=8b2c3630d35d4aba9e01d9c67ddc4599-1601976071950-07892-_dVohEV8&businessType=ProductDetail&title=%E2%82%AA+46.10++40%25OFF+%7C+ZigBee+smart+home++wall+touch+switch%2Cwireless+Intelligent+light+switch%2C2%2F3+Way%2CAPP+Control%2Cworks+with+google+home+aleax%2CUSUK&platform=AE&terminal_id=6eacd466c10c4ba799ef598104b0ed06&templateKey=white_backgroup_101

## data/database.db entry of the device
my log:

https://paste.ofcode.org/TuhvQc6ib3XFruJqPMwtS9

Koenkk, 2020-10-06:
Added, Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-06 by Koenkk

dror23, 2020-10-06:
Thank you very much!