Koenkk/zigbee2mqtt #4571: Eurotronic Spirit zigbee won't accept commands?

gielfeldt, 2020-10-06:
Hi

I'm trying to control my Eurotronic Spirit Zigbee thermostats from zigbee2mqtt. Getting data from the thermostat works fine, but sending messages to the thermostat does not work. Or if it does, then I don't know how to do it properly.

## What happened
Got error when trying to change temperature setpoint on Eurotronic Spirit thermostat.

`2020-10-06T11:31:59.165140021Z Zigbee2MQTT:error 2020-10-06 13:31:59: Publish 'set' 'current_heating_setpoint' to 'sove_radiator' failed: 'Error: Write 0x00158d00053d357d/1 hvacThermostat({"16387":{"value":1900,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')'`

## What did you expect to happen
I expected the temperature setpoint to bet set.

## How to reproduce it (minimal and precise)
Post `{"current_heating_setpoint":19}` to `zigbee2mqtt/device_friendly_name/set`

`2020-10-06T11:31:59.165140021Z Zigbee2MQTT:error 2020-10-06 13:31:59: Publish 'set' 'current_heating_setpoint' to 'sove_radiator' failed: 'Error: Write 0x00158d00053d357d/1 hvacThermostat({"16387":{"value":1900,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')'
2020-10-06T11:31:59.169145852Z Zigbee2MQTT:info  2020-10-06 13:31:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'current_heating_setpoint' to 'sove_radiator' failed: 'Error: Write 0x00158d00053d357d/1 hvacThermostat({\"16387\":{\"value\":1900,\"type\":41}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4151,\"transactionSequenceNumber\":null}) failed (Status 'INVALID_VALUE')'","meta":{"friendly_name":"sove_radiator"},"type":"zigbee_publish_error"}'`

## Debug info
Zigbee2MQTT version: dev latest
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

I noticed that when posting properties in payload that are unknown, zigbee2mqtt repsonds with a "No converter available". In the logs above, it also seems that it converts the 19 into 1900. Maybe that is a hint?

gielfeldt, 2020-10-06:
Hmmm. I may have found some relevant code over at https://github.com/Koenkk/zigbee-herdsman-converters

Perhaps this issue should go there? I'm not certain, please advise.

sjorge, 2020-10-06:
19 -> 1900 is AFAIK, I don't have mine hooked up currently but e.g. 18.5 will become 1850.

It's also within the accepted range of 10 -> 30. I'll see if I can find mine, maybe I can replicate the error.

sjorge, 2020-10-06:
Sadly I can replicate it, I now also get INVALID_VALUE back. That's both good and bad.
good -> you're not alone
bad -> somehow this broke

gielfeldt, 2020-10-06:
Thanks. I consistently get the INVALID_VALUE, almost instantaneously.

However, I'm facing other challenges now, which may or may not be related. After having paired that Spirit thermostat, I seem to be unable to pair devices anymore. It's like announcements are not even coming through?
All my existing paired devices still work fine though.

It's all very weird. I'll try to remove the thermostat again, and see if that does anything.

Side-note: It also seems to be pretty chatty for a battery powered device?

TilmanK, 2020-10-06:
@gielfeldt: Yes, the batteries don't last a full year in mine. Nevertheless, the AA ones are easy to replace.

I just tried on one of mine and sent:

{"current_heating_setpoint" : 23}

Works flawlessly.

sjorge, 2020-10-06:
Yeah I got around 6-8 months in mine.

gielfeldt, 2020-10-06:
I've managed to get it to pair again. I had to unplug the coordinator, and then replug it. I used a different ub port this time, not sure if that matters. Everything pairs fine now. I have two of those thermostats, and they paired without problems.

However, I still consistently get the INVALID_VALUE error when trying to manipulate any of the two thermostats.

I'm probably giving up on these thermostats, or I'm trying out another zigbee stick, or ultimately perhaps a different hub.

What are you guys using? I'm currently using a Pi 4 + a CC2531 stick.

I have a CC2530 lying around, so I might try that one out to see if it makes any difference.

I've also read about the samsung smartthings hub and the homey hub being able to run zigbee+zwave to mqtt locally? That might be something I will try out, if I don't get this to work.

sjorge, 2020-10-06:
I'm also using a CC2531, but your issue sounds familiar: https://github.com/Koenkk/zigbee2mqtt/issues/3177

Brunes, 2020-10-06:
Hi guys,

it's interesting because my two new thermostats don't work either. The one I bought before the summer works and I can control it, today I received new and the same but it says failed (Status' INVALID_VALUE ')'

How can this be fixed?

sjorge, 2020-10-06:
Does older and newer one have different firmware? Mines pretty told so maybe why it mostly works

Brunes, 2020-10-06:
It's 1.15.0 both.

-> Old
"manufacturer":"Eurotronic","model":"Spirit Zigbee wireless heater thermostat (SPZB0001)","name":"Termostat1","sw_version":"Zigbee2MQTT 1.15.0"}

-> New
"manufacturer":"Eurotronic","model":"Spirit Zigbee wireless heater thermostat (SPZB0001)","name":"TermostatBH","sw_version":"Zigbee2MQTT 1.15.0"}

gielfeldt, 2020-10-07:
Mine are brand new as well, so it could make sense that new ones don't work.

1.15.0 is the zigbee2mqtt version, isn't it?

Not sure if or where you can see the firmware version for the thermostats.

gielfeldt, 2020-10-07:
My thermostats report the following I think

Hardware version is 35
Firmware version is 22190930

sjorge, 2020-10-07:
> My thermostats report the following I think
> 
> Hardware version is 35
> Firmware version is 22190930

Yep this looks more correct than 1.15.0 which is indeed de zigbee2mqtt version, my mostly working (old) one is
Firmware: 22190930
Hardware: 32

gielfeldt, 2020-10-07:
Then the question is: How do we figure out the protocol difference between these two?

sjorge, 2020-10-07:
> Then the question is: How do we figure out the protocol difference between these two?

Last time someone who was fluent in German managed to get them to send over a implementation document, sadly he did not share it. I forgot who it was. But probably one of the first committers for this device.

Brunes, 2020-10-07:
I think this is a serious problem. Can zigbee2mqtt developers fix it or thermostat developers? People usually buy this device because it is on the device support list.

TilmanK, 2020-10-07:
I've got three of them and got a new one recently:

"appVersion":18,"stackVersion":5,"hwVersion":34,"dateCode":"20190408","swBuildId":"18181120",
"appVersion":22,"stackVersion":5,"hwVersion":35,"dateCode":"20191014","swBuildId":"22190930",
"appVersion":18,"stackVersion":5,"hwVersion":34,"dateCode":"20190408","swBuildId":"18181120"

And indeed, correcting my statement above, the second one does not react when trying to set current_heating_setpoint:

ERROR: Publish 'set' 'current_heating_setpoint' to 'Heizung Hobbykeller links' failed: 'Error: Write 0x00158d00053d4419/1 hvacThermostat({"16387":{"value":2300,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Data request failed with error: 'Timeout' (9999))'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'current_heating_setpoint' to 'Heizung Hobbykeller links' failed: 'Error: Write 0x00158d00053d4419/1 hvacThermostat({\"16387\":{\"value\":2300,\"type\":41}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4151,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'Timeout' (9999))'","meta":{"friendly_name":"Heizung Hobbykeller links"},"type":"zigbee_publish_error"}'

Nevertheless, it reports its values correctly, so there is connection (not good, but it's there):
INFO: MQTT publish: topic 'zigbee2mqtt/Heizung Hobbykeller links', payload '{"last_seen":1602056788015,"linkquality":6,"unoccupied_heating_setpoint":16}'

I'll flash my Sniffer stick and see if I can find anything about it.

sjorge, 2020-10-07:
In the past we also asked Eurotronics for firmware images which they refused to provided. So also no downgrading.

But based on the data here it looks like current models have a different hardware revision and firmware on them, capturing the traffic via a sniffer to/from both the old and new devices might indeed be useful.

TilmanK, 2020-10-07:
> In the past we also asked Eurotronics for firmware images which they refused to provided. So also no downgrading.
>
> But based on the data here it looks like current models have a different hardware [...]

Yeah, we'll see. Expect results in the afternoon, I hope I'll find the time then. :)

gielfeldt, 2020-10-07:
@TilmanK 

You do, however, get a different error than @Brunes and me. I hope though that the sniffing proves useful anyways.

TilmanK, 2020-10-07:
> @TilmanK 
> 
> 
> 
> You do, however, get a different error than @Brunes and me. I hope though that the sniffing proves useful anyways.

I know. But it's highly suspicious that those commands fail reproducibly and others not. I haven't had a look at the debug logs though...

TilmanK, 2020-10-07:
Ok, guys, so far I know the following:

I can't see any messages from Z2M to the device with sw 22190930. Neither setting the attribute current_heating_setpoint nor trying to read it produces a logged zigbee message. It works like a charm for the other device with sw 18181120.

I set the temperature manually on both devices and sniffed the Report Attributes message. The Zigbee payload is identical apart from sequence number and counter. So the device still reports a 16-bit integer value.

Here's the relevant zigbee debug output:

`[34mZigbee2MQTT:debug[39m 2020-10-07 17:04:36: Received MQTT message on 'zigbee2mqtt/Heizung Hobbykeller rechts/set' with data '{"current_heating_setpoint":19}'
[34mZigbee2MQTT:debug[39m 2020-10-07 17:04:36: Publishing 'set' 'current_heating_setpoint' to 'Heizung Hobbykeller rechts'
  [38;5;220;1mzigbee-herdsman:controller:endpoint [0mWrite 0x00158d000192392d/1 hvacThermostat({"16387":{"value":1900,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) [38;5;220m+17s[0m
  [38;5;27;1mzigbee-herdsman:adapter:zStack:adapter [0msendZclFrameToEndpointInternal 0x00158d000192392d:50047/1 (0,0,1) [38;5;27m+17s[0m
  [38;5;205;1mzigbee-herdsman:adapter:zStack:znp:SREQ [0m--> AF - dataRequest - {"dstaddr":50047,"destendpoint":1,"srcendpoint":1,"clusterid":513,"transid":3,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[20,55,16,4,2,3,64,41,108,7]}} [38;5;205m+17s[0m
  [38;5;33;1mzigbee-herdsman:adapter:zStack:unpi:writer [0m--> frame [254,20,36,1,127,195,1,1,1,2,3,0,30,10,20,55,16,4,2,3,64,41,108,7,173] [38;5;33m+17s[0m

  [38;5;169;1mzigbee-herdsman:adapter:zStack:znp:AREQ [0m<-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":50047,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":6847669,"transseqnumber":0,"len":6,"data":{"type":"Buffer","data":[28,55,16,4,4,0]}} [38;5;169m+19ms[0m
  [38;5;161;1mzigbee-herdsman:controller:log [0mReceived 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":4,"manufacturerCode":4151,"commandIdentifier":4},"Payload":[{"status":0}]},"address":50047,"endpoint":1,"linkquality":84,"groupID":0}' [38;5;161m+20s[0m

  [38;5;169;1mzigbee-herdsman:adapter:zStack:znp:AREQ [0m<-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":50047,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":84,"securityuse":0,"timestamp":6893332,"transseqnumber":0,"len":10,"data":{"type":"Buffer","data":[28,55,16,252,10,3,64,41,108,7]}} [38;5;169m+731ms[0m
  [38;5;161;1mzigbee-herdsman:controller:log [0mReceived 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":252,"manufacturerCode":4151,"commandIdentifier":10},"Payload":[{"attrId":16387,"dataType":41,"attrData":1900}]},"address":50047,"endpoint":1,"linkquality":84,"groupID":0}' [38;5;161m+731ms[0m
[34mZigbee2MQTT:debug[39m 2020-10-07 17:04:40: Received Zigbee message from 'Heizung Hobbykeller rechts', type 'attributeReport', cluster 'hvacThermostat', data '{"16387":1900}' from endpoint 1 with groupID 0

[34mZigbee2MQTT:debug[39m 2020-10-07 17:05:27: Received MQTT message on 'zigbee2mqtt/Heizung Hobbykeller links/set' with data '{"current_heating_setpoint":19}'
[34mZigbee2MQTT:debug[39m 2020-10-07 17:05:27: Publishing 'set' 'current_heating_setpoint' to 'Heizung Hobbykeller links'
  [38;5;220;1mzigbee-herdsman:controller:endpoint [0mWrite 0x00158d00053d4419/1 hvacThermostat({"16387":{"value":1900,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) [38;5;220m+6s[0m
  [38;5;27;1mzigbee-herdsman:adapter:zStack:adapter [0msendZclFrameToEndpointInternal 0x00158d00053d4419:39154/1 (0,0,1) [38;5;27m+6s[0m
  [38;5;205;1mzigbee-herdsman:adapter:zStack:znp:SREQ [0m--> AF - dataRequest - {"dstaddr":39154,"destendpoint":1,"srcendpoint":1,"clusterid":513,"transid":6,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[20,55,16,6,2,3,64,41,108,7]}} [38;5;205m+6s[0m
  [38;5;33;1mzigbee-herdsman:adapter:zStack:unpi:writer [0m--> frame [254,20,36,1,242,152,1,1,1,2,6,0,30,10,20,55,16,6,2,3,64,41,108,7,124] [38;5;33m+6s[0m
  [38;5;93;1mzigbee-herdsman:adapter:zStack:unpi:parser [0m<-- [254,1,100,1,0,100] [38;5;93m+6s[0m
  [38;5;93;1mzigbee-herdsman:adapter:zStack:unpi:parser [0m--- parseNext [254,1,100,1,0,100] [38;5;93m+1ms[0m
  [38;5;93;1mzigbee-herdsman:adapter:zStack:unpi:parser [0m--> parsed 1 - 3 - 4 - 1 - [0] - 100 [38;5;93m+0ms[0m
  [38;5;56;1mzigbee-herdsman:adapter:zStack:znp:SRSP [0m<-- AF - dataRequest - {"status":0} [38;5;56m+6s[0m
  [38;5;93;1mzigbee-herdsman:adapter:zStack:unpi:parser [0m--- parseNext [] [38;5;93m+0ms[0m
  [38;5;27;1mzigbee-herdsman:adapter:zStack:adapter [0mData confirm error (0x00158d00053d4419:39154,9999,0) [38;5;27m+10s[0m
[... cut ...] ber":null}) failed (Data request failed with error: 'Timeout' (9999))
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:319:27)
    at Generator.next (<anonymous>)
    at fulfilled (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)`

I'll try to repair the one device to see if something changes. Another difference in my setup is, that I use a CC26X2R1. Just for the record.

TilmanK, 2020-10-07:
I rejoined the device several times. I got it working now. This doesn't make sense at all...

I have absolutely no clue.

gielfeldt, 2020-10-07:
Can anybody recommend another zigbee radiator thermostat that I can use instead of these?

sjorge, 2020-10-07:
> Can anybody recommend another zigbee radiator thermostat that I can use instead of these?

I'm still looking my self, I have the eurotronic, viessmann, schnieder eletronic and tuya one. I'm not really satisfied of any of them.

viessmann is currently the best hardware wise, but the firmware is problematic. tuya worked but does the weird tuya stuff, the se one never really worked well for me.

gielfeldt, 2020-10-07:
Perhaps the thermostat market is better/more mature on z-wave?

TilmanK, 2020-10-07:
> > Then the question is: How do we figure out the protocol difference between these two?
> 
> Last time someone who was fluent in German managed to get them to send over a implementation document, sadly he did not share it. I forgot who it was. But probably one of the first committers for this device.

Is someone of you able to sniff the Zigbee traffic for this error? I'll be happy to contact them in German for you to see if they can help.

TilmanK, 2020-10-07:
@sjorge, @gielfeldt @Brunes I fixed the problem that the thermostat is not reacting by rejoining it. I had problems when doing this in one software lifecycle. You might therefore want to try the following:

1. Remove the device (force remove if necessary).
2. Restart Z2M
3. Reset the device, pass the interview and then CONFIRM that it was correctly configured.

I had failures in configuration all the time and once the configuration was successful, I could reach the device as expected.

gielfeldt, 2020-10-07:
@TilmanK  Perhaps I'll give it a try. However, I doubt it will change anything. It does pair correctly it seems, and it also does transmit data regularly. It's only when I try to send it commands, it fails.

gielfeldt, 2020-10-07:
@sjorge What does "weird tuya stuff" imply? :-)

gielfeldt, 2020-10-07:
@TilmanK I'm waiting for a programmer for my CC2530. When I get that, perhaps I could flash the CC2531 with a sniffer firmware and take a look at the traffic.

sjorge, 2020-10-07:
weird tuya stuff => having a custom tuya protocol on top of zigbee, instead of using the zigbee hvac stuff, so it is not open and most things have been pieced together. But looks like many tuya devices have slightly different implementations. Which IMHO makes it a mess.

gielfeldt, 2020-10-08:
Still invalid_value, here's the debug log. Can someone make any sense of this?

`
zigbee2mqtt_1  | 2020-10-08T08:01:27.355749330Z Zigbee2MQTT:debug 2020-10-08 10:01:27: Received MQTT message on 'zigbee2mqtt/0x00158d00053d357d/set' with data '{
zigbee2mqtt_1  | 2020-10-08T08:01:27.355899013Z   "current_heating_setpoint": 20
zigbee2mqtt_1  | 2020-10-08T08:01:27.355968401Z }
zigbee2mqtt_1  | 2020-10-08T08:01:27.356028233Z '
zigbee2mqtt_1  | 2020-10-08T08:01:27.360818752Z Zigbee2MQTT:debug 2020-10-08 10:01:27: Publishing 'set' 'current_heating_setpoint' to '0x00158d00053d357d'
zigbee2mqtt_1  | 2020-10-08T08:01:27.364487085Z 2020-10-08T08:01:27.363Z zigbee-herdsman:controller:endpoint Write 0x00158d00053d357d/1 hvacThermostat({"16387":{"value":2000,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null})
zigbee2mqtt_1  | 2020-10-08T08:01:27.367127026Z 2020-10-08T08:01:27.366Z zigbee-herdsman:adapter:zStack:adapter sendZclFrameToEndpointInternal 0x00158d00053d357d:61861/1 (0,0,1)
zigbee2mqtt_1  | 2020-10-08T08:01:27.368894111Z 2020-10-08T08:01:27.368Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":61861,"destendpoint":1,"srcendpoint":1,"clusterid":513,"transid":35,"options":0,"radius":30,"len":10,"data":{"type":"Buffer","data":[20,55,16,34,2,3,64,41,208,7]}}
zigbee2mqtt_1  | 2020-10-08T08:01:27.371521589Z 2020-10-08T08:01:27.370Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,20,36,1,165,241,1,1,1,2,35,0,30,10,20,55,16,34,2,3,64,41,208,7,255]
zigbee2mqtt_1  | 2020-10-08T08:01:27.380149124Z 2020-10-08T08:01:27.379Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
zigbee2mqtt_1  | 2020-10-08T08:01:27.381180979Z 2020-10-08T08:01:27.380Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
zigbee2mqtt_1  | 2020-10-08T08:01:27.381684119Z 2020-10-08T08:01:27.381Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
zigbee2mqtt_1  | 2020-10-08T08:01:27.383935862Z 2020-10-08T08:01:27.382Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
zigbee2mqtt_1  | 2020-10-08T08:01:27.384057768Z 2020-10-08T08:01:27.382Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-10-08T08:01:27.482519524Z 2020-10-08T08:01:27.482Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,35,229]
zigbee2mqtt_1  | 2020-10-08T08:01:27.483221347Z 2020-10-08T08:01:27.482Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,35,229]
zigbee2mqtt_1  | 2020-10-08T08:01:27.484501883Z 2020-10-08T08:01:27.483Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,35] - 229
zigbee2mqtt_1  | 2020-10-08T08:01:27.484609640Z 2020-10-08T08:01:27.484Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":35}
zigbee2mqtt_1  | 2020-10-08T08:01:27.484949987Z 2020-10-08T08:01:27.484Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-10-08T08:01:27.497014469Z 2020-10-08T08:01:27.496Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,28,68,129,0,0,1,2,165,241,1,1,0,131,0,157,4,153,0,0,8,28,55,16,34,4,135,3,64,165,241,29,149]
zigbee2mqtt_1  | 2020-10-08T08:01:27.497305669Z 2020-10-08T08:01:27.497Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,28,68,129,0,0,1,2,165,241,1,1,0,131,0,157,4,153,0,0,8,28,55,16,34,4,135,3,64,165,241,29,149]
zigbee2mqtt_1  | 2020-10-08T08:01:27.497718792Z 2020-10-08T08:01:27.497Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 28 - 2 - 4 - 129 - [0,0,1,2,165,241,1,1,0,131,0,157,4,153,0,0,8,28,55,16,34,4,135,3,64,165,241,29] - 149
zigbee2mqtt_1  | 2020-10-08T08:01:27.498269006Z 2020-10-08T08:01:27.497Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":513,"srcaddr":61861,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":131,"securityuse":0,"timestamp":10028189,"transseqnumber":0,"len":8,"data":{"type":"Buffer","data":[28,55,16,34,4,135,3,64]}}
zigbee2mqtt_1  | 2020-10-08T08:01:27.500678691Z 2020-10-08T08:01:27.499Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":true,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":34,"manufacturerCode":4151,"commandIdentifier":4},"Payload":[{"status":135,"attrId":16387}]},"address":61861,"endpoint":1,"linkquality":131,"groupID":0}'
zigbee2mqtt_1  | 2020-10-08T08:01:27.501183165Z 2020-10-08T08:01:27.500Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
zigbee2mqtt_1  | 2020-10-08T08:01:27.501270052Z 2020-10-08T08:01:27.500Z zigbee-herdsman:controller:endpoint Write 0x00158d00053d357d/1 hvacThermostat({"16387":{"value":2000,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')
zigbee2mqtt_1  | 2020-10-08T08:01:27.502740419Z Zigbee2MQTT:error 2020-10-08 10:01:27: Publish 'set' 'current_heating_setpoint' to '0x00158d00053d357d' failed: 'Error: Write 0x00158d00053d357d/1 hvacThermostat({"16387":{"value":2000,"type":41}}, {"timeout":10000,"disableResponse":false,"disable [... cut ...]
zigbee2mqtt_1  | 2020-10-08T08:01:32.297691507Z 2020-10-08T08:01:32.297Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,36,226]
zigbee2mqtt_1  | 2020-10-08T08:01:32.297824802Z 2020-10-08T08:01:32.297Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,36] - 226
zigbee2mqtt_1  | 2020-10-08T08:01:32.298131186Z 2020-10-08T08:01:32.297Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":36}
zigbee2mqtt_1  | 2020-10-08T08:01:32.298214759Z 2020-10-08T08:01:32.297Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
`

sjorge, 2020-10-08:
@Koenkk nothing immediately pops out to me, do you spot anything?

fedebruni84, 2020-10-11:
I have more or less the same problem. I tried everything: reset/rejoin, flashing again CC2531, changing USB port. It reports correctly the actual status but commands fail:
```

Oct 11 10:12:13 raspberrypi npm[609]: Zigbee2MQTT:info  2020-10-11 10:12:13: MQTT publish: topic 'zigbee2mqtt/eurotronic_thermostat01', payload '{"linkquality":65,"local_temperature":22.5,"occupied_heating_setpoint":21,"pi_heating_demand":0,"unoccupied_heating_setpoint":16}'
Oct 11 10:13:14 raspberrypi npm[609]: Zigbee2MQTT:error 2020-10-11 10:13:14: Publish 'set' 'system_mode' to 'eurotronic_thermostat01' failed: 'Error: Read 0x00158d0003c461c0/1 hvacThermostat([16392], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'
Oct 11 10:13:14 raspberrypi npm[609]: Zigbee2MQTT:info  2020-10-11 10:13:14: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'system_mode' to 'eurotronic_thermostat01' failed: 'Error: Read 0x00158d0003c461c0/1 hvacThermostat([16392], {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4151,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'","meta":{"friendly_name":"eurotronic_thermostat01"},"type":"zigbee_publish_error"}'

```

I don’t know if it’s 1.15 fault, but I have this thermostat for more than a year and everything worked correctly at least until some weeks ago.

Brunes, 2020-10-11:
Hi guys, I tried several times rejoin device, but I have same issue

`Zigbee2MQTT:error 2020-10-11 12:03:00: Publish 'set' 'current_heating_setpoint' to 'TermostatM' failed: 'Error: Write 0x00158d00053d35a2/1 hvacThermostat({"16387":{"value":1800,"type":41}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')'
`

Koenkk, 2020-10-11:
> I have more or less the same problem. I tried everything: reset/rejoin, flashing again CC2531, changing USB port. It reports correctly the actual status but commands fail:
>
> ```
>
> Oct 11 10:12:13 [...]

Can you try downgrading and see at which version it stopped working?

gielfeldt, 2020-10-12:
@fedebruni84  Sadly it's not the invalid_value error you're getting. However, if we find out which version the error you are getting arose, then it might hint to solving the other error. Who knows, they could be related.

I'm not home for the next couple days to try out some downgrading, unfortunately.

@Brunes Do you have the option to try out downgrading?

Brunes, 2020-10-12:
I can try, but I don't see reason why we should downgrading when one thermostat (same type) is working right.

fedebruni84, 2020-10-12:
@Koenkk @gielfeldt 
After I deleted state.json and database.db from data folder, reflashed CC2531 and changed usb port I had to pair again every single zigbee device. After that the thermostat works fine, it's quite weird.

Bibo13, 2020-10-15:
Are there some new Information about the controling of the Thermostat?
As you can see i have the same Messages when i try to control DPs.
Thanks for any Info about it.

2020-10-15 09:15:30.251  - error:  zigbee.0 (2496) Error on send command to 0x00158d00032f6d80. Error:  Error: Write 0x00158d00032f6d80/1  hvacThermostat({"16387":{"value":2250,"type":41}},  {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null})  failed (Status 'INVALID_VALUE')
--
at  Endpoint.checkStatus  (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:140:23)
at Endpoint. (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:169:26)
at Generator.next ()
at fulfilled  (/opt/iobroker/node_modules/iobroker.zigbee/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)

Palcooky, 2020-10-15:
I am also with you on this one. I have 5 of these thermostats purchased the last one i needed for the kitchen.

Pared Correctly, I receive the link quality and current temps. Weirdly enough not the battery level.

But as soon as i try to set the set point:

"{"type":"zigbee_publish_error","message":"Publish 'set' 'current_heating_setpoint' to '0x00158d00032f7cf2' failed: 'Error: Write 0x00158d00032f7cf2/1 hvacThermostat({\"16387\":{\"value\":1850,\"type\":41}}, {\"timeout\":10000,\"manufacturerCode\":4151,\"disableDefaultResponse\":true}) failed (Error: Status 'INVALID_VALUE')'","meta":{"friendly_name":"0x00158d00032f7cf2"}}"

gielfeldt, 2020-10-15:
I’m “glad” I’m not alone in this matter. I would like to help out with solving the issue, but I’m not sure what I can do.

Palcooky, 2020-10-15:
Ok so i managed to set the temp. In node-red I sent a message to this

zigbee2mqtt/0x00158d00032f7cf2/set/occupied_heating_setpoint : msg.payload : string[2]
"19"

zigbee2mqtt/<Name or ID of Unit>/set/occupied_heating_setpoint

I know this is not the what we normally send which is: zigbee2mqtt/<Name or ID of Unit>/set/current_heating_setpoint

But i do not use occupied and not occupied and this seems to work to set my current setpoint.

drdownload, 2020-10-15:
I've got the same problem. tried the tuya/moes before, had to upgrade their firmware to get them to work (accept commands, still no battery reported). I already have the Zwave Eurotronics which worked like a charm, for our holiday home I thought to get the Zigbee Version (because I wanted a Zigbee/Wifi only installation instead of my 10+ protocols home now ;) )

With the first inclusion it worked (kinda), battery status and setting the mode, but not the temperature, I tried switching to dev-branch of zigbee2qtt, restartarting, readding, every combination but it seems to just get worse, no not even battery status reported.

Error: Write 0x00158d00053d3967/1 hvacThermostat({"16392":{"value":33,"type":34}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')

drdownload, 2020-10-15:
Or the debug log when trying to set the mode:
Zigbee2MQTT:info  2020-10-15 21:27:58: MQTT publish: topic 'zigbee2mqtt/0x00158d00053d3967', payload '{"current_heating_setpoint":18.5,"elapsed":253868,"eurotronic_error_status":0,"eurotronic_host_flags":{"boost":false,"child_protection":false,"mirror_display":false,"window_open":false},"eurotronic_system_mode":1,"last_seen":"2020-10-15T21:27:58+02:00","linkquality":49,"local_temperature":27.5,"occupied_heating_setpoint":18.5,"pi_heating_demand":0,"system_mode":"auto","unoccupied_heating_setpoint":16}'
Zigbee2MQTT:debug 2020-10-15 21:27:58: eurotronic: host_flags object converted to 33
Zigbee2MQTT:error 2020-10-15 21:27:58: Publish 'set' 'system_mode' to '0x00158d00053d3967' failed: 'Error: Write 0x00158d00053d3967/1 hvacThermostat({"16392":{"value":33,"type":34}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')'
Zigbee2MQTT:debug 2020-10-15 21:27:58: Error: Write 0x00158d00053d3967/1 hvacThermostat({"16392":{"value":33,"type":34}}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":4151,"transactionSequenceNumber":null}) failed (Status 'INVALID_VALUE')
    at Endpoint.checkStatus (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:202:26)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:24:58)
Zigbee2MQTT:info  2020-10-15 21:27:58: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'system_mode' to '0x00158d00053d3967' failed: 'Error: Write 0x00158d00053d3967/1 hvacThermostat({\"16392\":{\"value\":33,\"type\":34}}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":4151,\"transactionSequenceNumber\":null}) failed (Status 'INVALID_VALUE')'","meta":{"friendly_name":"0x00158d00053d3967"},"type":"zigbee_publish_error"}'

drdownload, 2020-10-15:
> Ok so i managed to set the temp. In node-red I sent a message to this
> 
> zigbee2mqtt/0x00158d00032f7cf2/set/occupied_heating_setpoint : msg.payload : string[2]
> "19"
> 
> zigbee2mqtt/0x00158d00032f7cf2/set/occupied_heating_setpoint
> 
> I know this is not the what we normally send which is: zigbee2mqtt/0x00158d00032f7cf2/set/current_heating_setpoint
> 
> But i do not use occupied and not occupied and this seems to work to set my current setpoint.

Tried this with the zigbee2mqtt webUI and can confirm. occupied setpoint works and also the trv is updated immediatly.

drdownload, 2020-10-15:
This paragraph in the manual states that:
```
Die übertragenen Solltemperaturen wie Occupied/Unoccupied Heating Setpoint Attribute (0x0012 oder 0x0014)
werden auf das Attribut Current Temperature Setpoint (0x4003) kopiert, um den TRV ohne hersteller spezifische
Attribute verwenden zu können
```
the occupied heating setpoint is copied to current temperature setpoint. We could assume that it depends on the mode if it allowed to write to current temperature setpoint directly or we need to use the others

Palcooky, 2020-10-15:
Yea I found the English manual: 
https://eurotronic.org/wp-content/uploads/2019/11/Spirit_ZigBee_BAL_web_EN_November-2019.pdf

That was what made me try the option to set the occupied value.

drdownload, 2020-10-15:
I can't find anything strange with the converter, and also the documentation seems very good for this TRV (in zigbee2mqtt and from Eurotronics), I cant wrap my head around what doesnt work.

sjorge, 2020-10-15:
Oof there are some contradictions in there with regard to reporting, piHeatingDemand is listed as both reportable at 600s default and non reportable capable in the table earlier in the dock :|

Looks like if we implement the manufacturing specific cluster we should be able to grab some more things like manual valve position control!

Not that this will help us with the odd INVALID_VALUE, I wonder if it is expecting some sort of transaction number. I noticed most were null when sending.

~ sjorge

drdownload, 2020-10-15:
To tease me the battery status for the first time showed up in zigbee2mqtt without any change ;)

Palcooky, 2020-10-15:
Yep im seeing battery aswell now.

{
    "current_heating_setpoint": 18,
    "eurotronic_host_flags": {
        "mirror_display": false,
        "boost": false,
        "window_open": false,
        "child_protection": false
    },
    "system_mode": "auto",
    "eurotronic_system_mode": 1,
    "linkquality": 92,
    "local_temperature": 20.5,
    "occupied_heating_setpoint": 18,
    "unoccupied_heating_setpoint": 16,
    "pi_heating_demand": 0,
    "eurotronic_error_status": 0,
    "battery": 100
}

drdownload, 2020-10-15:
now occupied and local setpoint have decoupled ;) 
[image]

gielfeldt, 2020-10-15:
Incredible. "occupied_heating_setpint" seems to work! ... sometimes :-).

In any case, this is a step in the right direction.

gielfeldt, 2020-10-15:
I played around with some commands. After sending a eurotronic_trv_mode, my local_temperature reports weird numbers. Like 0.04.

Anyone knows how I get back to proper temperature in celcius?

gielfeldt, 2020-10-15:
Okay, temperature is back on track. All by itself. Not sure what was wrong.

Using "occupied_heating_setpoint" works well enough for me. Hope it doesn't stop working.

Brunes, 2020-10-16:
When I tried set occupied_heating_setpoint

`ERROR: Publish 'set' 'occupied_heating_setpoint' to '0x00158d00053d260a' failed: 'Error: Write 0x00158d00053d260a/1 hvacThermostat({"occupiedHeatingSetpoint":2800}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC transaction expired' (240))'`

drdownload, 2020-10-16:
Ah, I noticed, that I get an Error Code because I demounted the TRV and used my ZWave one which is already integrated ;). Seems when in Error Mode it wont update temperature nor take temperature commands.

[121 more replies]

closed 2021-06-23 by github-actions[bot]