Koenkk/zigbee2mqtt #4577: Support for new device Zemismart ZW-EU-01 (zigbeeModel: LXN59-1S7LX1.0)

nanojavi, 2020-10-06:
## Information about the device + link

Zemismart ZW-EU-01: Coordinator + Switch 1 Gang (ZigbeeModel: LXN59-1S7LX1.0 )

## data/database.db entry of the device

Sorry if request is in the wrong place, is my first pull request

It works as router, but when activating switch, "non supported device" error appears.

I added the code below on the devices.js, now mqtt messages are received correctly, but not discovered on HA.

CODE:
    {
        zigbeeModel: ['LXN59-1S7LX1.0'],
        model: 'LXN59-1S7LX1.0',
        vendor: 'Nue / 3A', 
        description: 'Smart light relay - 1 gang',
        extend: generic.switch,
        exposes: [exposes.switch().withEndpoint('left')],
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZW-EU-01'},
        ],
        endpoint: (device) => {
            return {'left': 1};
        },
        meta: {configureKey: 1, multiEndpoint: true},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint1 = device.getEndpoint(1);
            await bind(endpoint1, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint1);
        },
   },

Zigbee2MQTT log when switch activated manually:

Zigbee2MQTT:debug 2020-10-06 21:32:01: Received Zigbee message from 'RS_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2020-10-06 21:32:01: MQTT publish: topic 'zigbee2mqtt/RS_1', payload '{"linkquality":10,"state":"ON","state_left":"OFF"}'

Zigbee2MQTT:debug 2020-10-06 21:32:01: Received Zigbee message from 'RS_1', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0

Zigbee2MQTT:info  2020-10-06 21:32:01: MQTT publish: topic 'zigbee2mqtt/RS_1', payload '{"linkquality":13,"state":"ON","state_left":"ON"}'

Koenkk, 2020-10-07:
Added!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-07 by Koenkk

nanojavi, 2020-10-08:
After a few test, now is working with this code in devices.js

    {
        zigbeeModel: ['LXN59-1S7LX1.0'],
        model: 'LXN59-1S7LX1.0',
        vendor: 'Nue / 3A',
        description: 'Smart light relay - 1 gang',
        extend: generic.switch,
        whiteLabel: [
            {vendor: 'Zemismart', model: 'ZW-EU-01'},
        ],
        meta: {configureKey: 1},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1) || device.getEndpoint(3);
            await bind(endpoint, coordinatorEndpoint, ['genOnOff']);
            await configureReporting.onOff(endpoint);
        },
   },

In addition, also added to homeassistant.js the device (it was missing from my side)

'LXN59-1S7LX1.0': [cfg.switch],

Thanks for all the support