Koenkk/zigbee2mqtt #4579: Can't add new router to CC2531 coordinator - "Failed to call 'Availability'"

EfficientSetting, 2020-10-07:
# What happened
I've been trying to add a CC2531 (FW router-cc2531-std) to my CC2531 Coordinator (Default FW).
The router changes LED indicating a successful pairing/join, but the router never shows up on the zigbee2mqtt base.

## What did you expect to happen
Router joins network and shows up on coordinator network

## How to reproduce it (minimal and precise)
Have 2 CC2531 - 1 Router, 1 Coordinator and join the router to the coordinator network

## Debug info
Zigbee2MQTT version: 1.15.0 (current master pulled)
Adapter hardware: CC2531 Coordinator + CC2531 Router
Adapter firmware version: router-cc2531-std + Zigbee 1.2HA CC2531ZNP-Prod (both fresh flashed, Zigbee Network is 4 Bulbs, 4 end devices)

Error message from journalctl 
```
Oct 06 20:42:23 iotpi node[3066]: Zigbee2MQTT:error 2020-10-06 20:42:23: Failed to call 'Availability' 'onZigbeeEvent' (TypeError: Cannot read property 'friendlyName' of null
Oct 06 20:42:23 iotpi node[3066]:     at Availability.inPasslistOrNotInBlocklist (/opt/zigbee2mqtt/lib/extension/availability.js:47:37)
Oct 06 20:42:23 iotpi node[3066]:     at Availability.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/availability.js:180:18)
Oct 06 20:42:23 iotpi node[3066]:     at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:371:44))
```

Koenkk, 2020-10-07:
Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

If you see something in the logging when pairing, please provide a complete debug log (not jus a few lines).

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

EfficientSetting, 2020-10-07:
> Make sure to check https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair
>
> If you see something in the logging when pairing, please provide a complete debug log (not [...]

Here's the debug log. Note, that I use the new API, the problem existed prior already though

```
Oct 07 17:57:27 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:57:27: Received MQTT message on 'zigbee2mqtt/bridge/request/permit_join' with data '{"value":true}'
Oct 07 17:57:27 iotpi node[4696]: Zigbee2MQTT:info  2020-10-07 17:57:27: Zigbee: allowing new devices to join.
Oct 07 17:57:28 iotpi node[4696]: Zigbee2MQTT:info  2020-10-07 17:57:28: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"e30b257","config":{"advanced":{"adapter_concurrent":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":60,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":11,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"ISO_8601","legacy_api":true,"log_directory":"/opt/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{"0x00158d000233935d":{"friendly_name":"Single Clicker Double Rocker Switch"},"0x00158d000276f9ae":{"friendly_name":"Xiaomi Mi Cube"},"0x00158d000283cc94":{"friendly_name":"Alika Single Button"},"0x00158d0002fd1b9b":{"friendly_name":"Double Clicker Double Rocker Switch"},"0x00158d0003a5b3c6":{"friendly_name":"Humidity and Temperature Sensor - Sunroom","retain":true},"0x0017880103adfff2":{"friendly_name":"Wardrobe Ceiling 4"},"0xf0d1b800000574e9":{"friendly_name":"Kitchen Ceiling 1"},"0xf0d1b8000005826e":{"friendly_name":"Kitchen Ceiling 4"},"0xf0d1b800000583d0":{"friendly_name":"Kitchen Ceiling 3"},"0xf0d1b80000059347":{"friendly_name":"Kitchen Ceiling 2"}},"experimental":{"new_api":true,"output":"json"},"external_converters":[],"frontend":{"port":8080},"groups":{},"homeassistant":true,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","include_device_information":false,"password":">YUKVgFDvh1sA5T[sx","server":"mqtt://192.168.178.115","user":"user"},"passlist":[],"permit_join":false,"serial":{"disable_led":true,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":11,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":true,"version":"1.15.0"}'
Oct 07 17:57:28 iotpi node[4696]: Zigbee2MQTT:info  2020-10-07 17:57:28: MQTT publish: topic 'zigbee2mqtt/bridge/response/permit_join', payload '{"data":{"value":true},"status":"ok"}'
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Received Zigbee message from 'Kitchen Ceiling 3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Successfully pinged 'Kitchen Ceiling 3'
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Received Zigbee message from 'Kitchen Ceiling 4', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Successfully pinged 'Kitchen Ceiling 4'
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Received Zigbee message from 'Kitchen Ceiling 1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 17:58:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:06: Successfully pinged 'Kitchen Ceiling 1'
Oct 07 17:58:16 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:16: Received Zigbee message from 'Kitchen Ceiling 4', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Oct 07 17:58:16 iotpi node[4696]: Zigbee2MQTT:info  2020-10-07 17:58:16: MQTT publish: topic 'zigbee2mqtt/Kitchen Ceiling 4', payload '{"brightness":254,"color":{"hue":"0","saturation":"0","x":0.323,"y":0.329},"color_temp":370,"last_seen":"2020-10-07T21:58:16.683Z","linkquality":31,"state":"OFF","update":{"state":"idle"},"update_available":false}'
Oct 07 17:58:23 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:23: Received Zigbee message from 'Kitchen Ceiling 2', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 17:58:23 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:58:23: Successfully pinged 'Kitchen Ceiling 2'
Oct 07 17:59:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 17:59:06: Received Zigbee message from 'Kitchen Ceiling 3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
[... cut ...]
Oct 07 18:01:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 18:01:06: Received Zigbee message from 'Kitchen Ceiling 3', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 18:01:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 18:01:06: Successfully pinged 'Kitchen Ceiling 3'
Oct 07 18:01:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 18:01:06: Received Zigbee message from 'Kitchen Ceiling 1', type 'readResponse', cluster 'genBasic', data '{"zclVersion":2}' from endpoint 1 with groupID 0
Oct 07 18:01:06 iotpi node[4696]: Zigbee2MQTT:debug 2020-10-07 18:01:06: Successfully pinged 'Kitchen Ceiling 1'
```

Koenkk, 2020-10-08:
Thanks, I need some more logging as I can't see what goes wrong yet. can you provide the herdsman debug logging. 

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

EfficientSetting, 2020-11-14:
Sorry for the late reply, elections and a broken MacBook kept me busy at work.

Here's the log from the INFO log output again (maybe good for comparison):

```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Coordinator"},"type":"device_announced"}'
ERROR: Failed to call 'Availability' 'onZigbeeEvent' (TypeError: Cannot read property 'friendlyName' of null at Availability.inPasslistOrNotInBlocklist (/opt/zigbee2mqtt/lib/extension/availability.js:47:37) at Availability.onZigbeeEvent (/opt/zigbee2mqtt/lib/extension/availability.js:178:18) at Controller.callExtensionMethod (/opt/zigbee2mqtt/lib/controller.js:378:44))
```

And here's the Zigbee-herdsman debug log (I killed the process right after the availability failure):
https://pastebin.com/5yKDTXRM

Let me know if you need anything else

Koenkk, 2020-11-14:
@EfficientSetting can you check if the ieeeAddr of the coordinator is the same as the one of the router? `0x00124b0018e1a7a9` Can be checked in the flash programmer: https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html#windows

closed 2020-11-22 by github-actions[bot]

EfficientSetting, 2020-11-23:
@Koenkk Again, I'm late :( I'm sorry and would understand if this issue isn't worth your time. 

I just put the programmer together, unfortunately I don't have a windows machine close to check it in the GUI, but using the -a option on the cc-tool gives me

`MAC addresses, primary: 00:12:4B:00:18:E1:A7:A9, secondary: FF:FF:FF:FF:FF:FF:FF:FF`

Seems like your assumption that the router has the same IEEE address is correct. Any fix for that?

Koenkk, 2020-11-23:
@EfficientSetting you should check if it's possible to change the ieeeAddress with cc-tool, otherwise do it via the GUI on a windows machine.