Koenkk/zigbee2mqtt #4580: Cannot pair old Hue bulb on latest-dev

gielfeldt, 2020-10-07:
On latest-dev my old hue bulb cannot be paired. It works on 1.15.0

## What happened
`
Successfully interviewed '0x0017880100bca120', device has successfully been paired
Device '0x0017880100bca120' with Zigbee model 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
`

## What did you expect to happen
`
Successfully interviewed '0x0017880100bca120', device has successfully been paired
Device '0x0017880100bca120' is supported, identified as: Philips Hue white and color ambiance E26/E27/E14 (9290012573A)
`

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: latest-dev
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-10-07:
Can you provide the herdsman debug logging of pairing this device?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

gielfeldt, 2020-10-07:
I'll attempt to do so later tonight or tomorrow. I'll produce a log for both the working and non-working.

gielfeldt, 2020-10-07:
Heisenbug....

I tried pairing it to latest-dev, and it failed consistently. I then enabled herdsman-debug-logging, and then it paired fine?

And now, I cannot get it to fail, even after I disable debug logging?

Could it be a timing issue somewhere?

I'll make some more attempts and see if I can get it to fail again, with debug logging enabled.

gielfeldt, 2020-10-07:
I cannot reproduce this anymore. I have no idea what was wrong.

closed 2020-10-07 by gielfeldt

Sab3rRid3r, 2020-10-11:
Adapter hardware: zzh! CC2652R 
Zigbee2MQTT version: 1.15.0

Same situation here while trying to pair a Hue White LWB010

Zigbee2MQTT:warn  2020-10-11 17:46:24: Device '0x00178801026fcd44' with Zigbee model 'undefined' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

After that I enabled herdsman logging and tried to remove the pairing. normal remove did not work so I used force_remove and did a factory reset by using the dimmer switch. After that the pairing was successfull.

Regards, Andreas