Koenkk/zigbee2mqtt #4581: No network map on 1.15.0

ThePirat91, 2020-10-07:
## What happened
No more network map on 1.15.0.
Without it, I cannot be sure that my endpoints are passing through the router and not the coordinator.

[image]
[image]

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: Coordinator CC2531, Router CC2530+CC2591
Adapter firmware version: CC2531 fw source-routing v.20200619, CC2530+CC2591 fw router v.201902 (router-cc2530-cc2591-std.hex)

danieledwardgeorgehitchcock, 2020-10-07:
If you bring the router closer to your co-ordinator, do you get the same error? My guess is that it may be out of range..

ThePirat91, 2020-10-07:
It's just few meters away, with a hollow bricks wall between....
On 1.14.x and before I have never had this kind of issues and they were both on same place....
After 1.15.0 upgrade, it worked once and never more....

Thanks

ThePirat91, 2020-10-14:
Just tried putting my router closer to the coordinator.
No luck.
Also tried to put coordinator away from my rpi and wifi ap.
No luck.

What's wrong with 1.15.0 and network map?

Thanks

ThePirat91, 2020-11-14:
UPDATA

Just tried with 1.16.1....the problem is still here
i tried to reflash both coordinator and router...no luck at all

closed 2020-12-23 by github-actions[bot]