Koenkk/zigbee2mqtt #4582: Moes HY369RT (AKA TuYa TS0601) Battery Status Unavailable In Home Assistant

danieledwardgeorgehitchcock, 2020-10-07:
## What happened
Added the Moes HY369RT (AKA TuYa TS0601) in Zigbee2MQTT and can see that status of the battery in the new beta frontend.

## What did you expect to happen
Battery status reported in HomeAssistant - it is shown as unavailable (battery with exclamation mark)

## How to reproduce it (minimal and precise)
Add Moes HY369RT (AKA TuYa TS0601) to Zigbee2MQTT and enable MQTT integration with auto device generation in HomeAssistant.

## Debug info
Zigbee2MQTT version: 1.15 beta
Adapter hardware: CC26X2R1

drdownload, 2020-10-11:
Same for 1.14 and CC2531

mcsshs, 2020-10-11:
I have the same problem
Z2M 1.15.0 and CC2531

danieledwardgeorgehitchcock, 2020-10-12:
Just to add, it appears that Zigbee2MQTT only detects these devices as being powered by batteries (no battery discharge status is collected) - Looking at a sniff of one of these devices on a TuYa network, it would appear that the status is reported so, Zigbee2MQTT just doesn't seem to be converting these values.

I can provide sniff logs if required..

ARNik, 2020-11-16:
The same issue with zigbee2mqtt and CC2538 coordinator.

Blueforcer, 2020-11-27:
Same with conbee II

groenmarsmannetje, 2020-12-13:
It does report the battery_low status but not the %. I can see that now flashing on and off when the motor is busy adjusting the valve.

Kalypox, 2020-12-18:
Same with PiZiGate

markheloking, 2021-01-10:
> The same issue with zigbee2mqtt and CC2538 coordinator.

Same here also with zigbee2mqtt and CC2538 coordinator.

closed 2021-02-18 by github-actions[bot]