Koenkk/zigbee2mqtt #4584: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

ayasystems, 2020-10-07:
I can not start zigbee2mqtt

## Debug info
Zigbee2MQTT version: 1.15.0 (commit #e30b257)
Adapter hardware: CC2531
Adapter firmware version:  CC2531ZNP-Prod.hex

Log response
https://pastebin.com/raw/8QtEBN9b

TilmanK, 2020-10-07:
Can you try to stop zigbee2mqtt, unplug the adapter, plug it in again and restart Z2M?

YamiFrankc, 2020-10-08:
Im having this same issue, it started happening at some point this morning.
I updated to the latest version last night.
I uplug the adapter and reboot. But result is the same

YamiFrankc, 2020-10-08:
Z2MQTT was able to start after I flashed the stick, and the devices appear as online on HA, however they do not update and I also can not change their state from HA. Here is the log file:
[](https://paste.ubuntu.com/p/2xNdTq9wc3/)

ayasystems, 2020-10-08:
Hi @TilmanK 

I have did some test.....

It works if i plug the stick, press the reset button for 5 seconds and start the software. But if i stop the software i need repeat the process... 

I have other raspberry 4 (same rasberry with the same raspbian version). I have test it with this raspberry and it works without doing any thing... plug it and works allways....

What's the problem?? I don't understand....

Best regards

YamiFrankc, 2020-10-08:
So as an update, maybe @ayasystems will have the same results as me, if you want to try.

After updating the firmware on the stick, zigbee2mqtt was able to run again. Devices on HA became online again, but they were not updating, and they took like 5 minutes to respond to commands(turning on a light, for example).

I just left it running for a few hours while I was doing something else, and then I noticed that my automations that depend on one or more zigbee devices started working again. I guess the software/hardware was just mapping out the devices again, or something again.

TLDR: I had the same problem, and  errors in my log as you. I fixed it by re flashing the stick and running Z2MQTT for a few hours.
Maybe it will work with you too

ayasystems, 2020-10-08:
@YamiFrankc 

I reflashed the stick yesterday and did not help for me.

I have used CC2531ZNP-Prod 10.06.2019, Are you using the same release firmware?

YamiFrankc, 2020-10-08:
I used the one that can be downloaded here [link](https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html)

Which seems to be from other date, 2019.06.08 as peer the zip file name

ayasystems, 2020-10-08:
We are using the same.....

ayasystems, 2020-10-08:
I've checked that only works if i start the software immediately after plug the stick...

Strange....

YamiFrankc, 2020-10-09:
Z2MQTT keeps failing on my side, after a few hours of working. It does not stop running, but all devices become offline.
The systemctl journal seems fine, however.  I cant post my log at the moment because im not home.
But I f i restart it a few times, it works again for some time

timdonovanuk, 2020-10-11:
Oh joy, this again. Z2M been working fine for 6 months untouched. Then this rears its ugly head again and brings down my entire house, and nothing works to restore the network. Sigh.

whocarezpt, 2020-10-14:
Hi,

I have the same error since i've upgraded.

whocarezpt, 2020-10-15:
After unplug & reset is working again

iFloris, 2020-10-15:
Using a CC1352P-2, I also had this problem a few times. Updated the firmware for the adapter etc. My solution ended up connecting the CC1352P-2 to a Pi Zero W and using ser2net to connect from Zigbee2Mqtt (in Hassio)

ccorderod, 2020-10-15:
I also use CC1352P_2. As the network is big, I have 2 Z2M instances running on Pi4. I have this issue with one: SRSP - SYS - Ping ... 
Rebooting fix the problem; restarting Z2M does not. I do not need to unplug, plug or push the reset button.
The other Z2M instance is running fine; same Pi4, same power supply, same patch level.

larbigj, 2020-11-19:
I have also the problem with the CC2531 using a pi4, pi4 and pi zero. Reflashing the stick does not fix the problem. The same stick is running without problems on a zotac zbox

CountOlaf87, 2020-11-28:
I'm also having this problem, I have the CC2531 stick on a Intel NUC running Hassio in proxmox.

877dev, 2020-12-09:
Same here after updating to 1.16.2, tried the usual stop container, unplug, wait 10s, plug back in, restart container.

`
npm ERR! Failed at the zigbee2mqtt@1.16.2 start script.

npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:

npm ERR!     /root/.npm/_logs/2020-12-09T08_42_08_520Z-debug.log

Using '/app/data' as data directory

> zigbee2mqtt@1.16.2 start /app

> node index.js

Zigbee2MQTT:debug 2020-12-09 08:42:11: Loaded state from file /app/data/state.json

Zigbee2MQTT:info  2020-12-09 08:42:11: Logging to console and directory: '/app/data/log/2020-12-09.08-42-10' filename: log.txt

Zigbee2MQTT:debug 2020-12-09 08:42:11: Removing old log directory '/app/data/log/2020-12-09.08-37-56'

Zigbee2MQTT:info  2020-12-09 08:42:11: Starting Zigbee2MQTT version 1.16.2 (commit #04c15f7)

Zigbee2MQTT:info  2020-12-09 08:42:11: Starting zigbee-herdsman (0.13.37)

Zigbee2MQTT:debug 2020-12-09 08:42:11: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'

Zigbee2MQTT:error 2020-12-09 08:42:31: Error while starting zigbee-herdsman

Zigbee2MQTT:error 2020-12-09 08:42:31: Failed to start zigbee

Zigbee2MQTT:error 2020-12-09 08:42:31: Exiting...

Zigbee2MQTT:error 2020-12-09 08:42:31: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)

    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)

    at Generator.throw (<anonymous>)

    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)

npm ERR! code ELIFECYCLE

npm ERR! errno 1

npm ERR! zigbee2mqtt@1.16.2 start: `node index.js`

npm ERR! Exit status 1


`

yves-gaignard, 2020-12-09:
I am also having the same issue.
My configuration is 

Zigbee2MQTT version: 1.16.2.2
Adapter hardware: zig-a-zig-ah! (CC2652 Stick)
Adapter firmware version: CC26X2R1_20201026.hex

Logs of the issue
[zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/5668304/zigbee2mqtt.log)
[home_assistant_log_viewer.log](https://github.com/Koenkk/zigbee2mqtt/files/5668305/home_assistant_log_viewer.log)

Here is the configuration of zigbee2mqtt: 

[zigbee2mqtt_config_yaml.txt](https://github.com/Koenkk/zigbee2mqtt/files/5668346/zigbee2mqtt_config_yaml.txt)

For your information, I tried unplug and plug the adapter several times, rebooted the Raspberry Pi several times, I have always the same issue.

Before I had this issue, I succeeded to run this adapter with a previous version of firmware (CC26X2R1_20200805.hex), but after the first reboot I had this same issue and the adapter was never recognized by zigbee2mqtt. That's why I decided to flash it again with the latest version, but it has never worked. 
I tried to flash the adapter to the previous version and it still not worked again.

I red a lot of issue related this kind of error, and I am desperate. I really think that the solution is not reliable at all.
Is someone succeeded to work with this kind of adapter and with zigbee2mqtt ?
Do you recommend to use another adapter and/or another solution than zigbee2mqtt ?

Thank you to all of you for your help.

yves-gaignard, 2020-12-10:
Omer from Electrolama team provides me a script to test the zzh adapter without Home Assistant and zigbee2mqtt. 
Use this script:  [ https://gist.github.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23](https://gist.github.com/omerk/0ee0e447a9e36786b4ff71d8f8126a23). 
The test passed.
So I should conclude that it is due to another issue.

I unplug and plug again the zzh to the Pi 4 and, miraculously, the error message changed.
Now it is : 
 AREQ - ZDO - stateChangeInd after 60000ms

I looked for it in the zigbee2mqtt issue database but I didn't find something interesting.

Here is the complete log of the issue: 
[2020-12-10-08-10-zigbee2mqtt.log](https://github.com/Koenkk/zigbee2mqtt/files/5671139/2020-12-10-08-10-zigbee2mqtt.log)

If it can help

timdonovanuk, 2020-12-10:
I was thinking it was just limited to CC1352P-2 boards but I see people having issues even with the newer CC2652's which is troubling. For me, I think its the stupid design of CC1352P-2 where you have to press a physical button on the device every time it power cycles. I was going to try changing to CC2652 but now I'm not so sure...

iFloris, 2020-12-10:
>  CC1352P-2 where you have to press a physical button on the device every time it power cycles.
I have now been using the CC1352P-2 on a Pi Zero W and using ser2net from my z2m installation since august and I have not had a problem with this anymore. In addition, z2m / hassio updates no longer cause the zigbee network to be rebuilt.

timdonovanuk, 2020-12-10:
@iFloris it only seems to affect that adapter when connected to non-pi systems, such as intel nucs. It's kind of a show stopper to be honest, because if your house loses power, it means you lose your zigbee network until you're physically able to push the button. I'm tempted to try ser2net but I honestly want less complexity in my system not more :/

Stooovie, 2020-12-21:
My CC2531 worked flawlessly for over a year, it started flaking out last night. There were no updates to anything, no downtime, no reboot. Z2M can't communicate with the stick anymore. Getting the usual:

Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.16.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)

kiat-bswg, 2020-12-21:
Me too. I've a CC2531 that worked fine on my Ubuntu 20.04 box where I flashed it with z2m (zigbee2mqtt).  A Wiser Smartplug was detected (showed up as a Heiman manufactured device) and I could send power on/off commands via MQTT to control the device.  Then I transferred it to its intended destination a Raspbian (RPi 3) where it has not worked at all, despite all the tips and tricks modprobe, udev, pressing the S1 button (ditto S2), both stable and dev versions of z2m, rebooting, etc. I've tried everything I can find and further searching is showing me the same proposed solutions.  At my wits end and patience nearly expended after spending way too many hours trying to resolve the issue of z2m not recognizing the CC

benvandijk84, 2020-12-25:
Same issue here. CC2531 with the zigbee2mqtt docker container worked for a very long time without any issue. But all of a sudden i'll getting the same errors as described as above. Reset of the CC2531, rebuild of container doesn't help at all. Weird thing is that in the debug log the container finds the stick and is able to open the serial connection.

[debug.txt](https://github.com/Koenkk/zigbee2mqtt/files/5741998/debug.txt)

Any thoughts/idea's on this one?

Koenkk, 2020-12-25:
If you get troubles with the CC2531 after it worked for some time likely the firmware got corrupted. This can be solved by reflashing. It's recommended to flash the source routing firmware in this case:  https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing. If after that your stick still crashes upgrade to a more powerful coordinator: https://www.zigbee2mqtt.io/getting_started/what_do_i_need.html#supported-zigbee-adapter

codalttjalling, 2020-12-26:
Hi Koenkk. Experiencing this problem here too using the Electrolama zig-a-zig-ah. Using https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_3.x.0/bin as firmware. Tested with two adapters, two raspberries (V3.1), multiple sd cards, multiple power supplies.

Any idea how to fix or debug? I upgraded to the zig-a-zig-ah to get a more stable network. However, nothing is working in my home right now.

araczkowski, 2021-01-15:
> likely the firmware got corrupted

@Koenkk sorry for noob question ;) but how this is possible? If something is written to the memory of the adapter while it is being used? 
Could you point me to the code where this is done? Thanks in advance for answer.

Koenkk, 2021-01-16:
@araczkowski this is all done within the firmware itself (see https://github.com/Koenkk/Z-Stack-firmware/blob/master/coordinator/Z-Stack_Home_1.2/COMPILE.md for obtaining the source). Since the firmware is EOL it will not be fixed anymore by Texas Instruments.

araczkowski, 2021-01-18:
Thank you @Koenkk for clarification. I had an impresion that this problem not only affect the CC2531 but it is more common (more adapters have this issue). But probably the "Error: SRSP - SYS - ping after 6000ms" is very general - in 99% there is something wrong with usb /dev port number or system is busy or permissions etc.. 
Before your comment, I had an impresion that this can't be a problem with adapter firmware... but now after you comment  "likely the firmware got corrupted" I'm not so sure...

Does anybody know is there any way to reset the CC2531 adapter without reflashing?
Thanks again!

renaiku, 2021-01-18:
I also got this 600ms error but with a Electrolama zig-a-zig-ah! (zzh!) Up to date with its latest firmware.

Sometimes it just stop working and throw this error. Every 24h+ I have to hard reboot the computer because resetting the usb dongle is not helping.

araczkowski, 2021-01-20:
Today I received such CC2531 device, the device has Z-Stack_Home_1.2 (default) firmware, worked well for couple of months... until sudently CC2531 stopped to communicate with zigbee2mqtt.
I kept it in USB for couple of hours trying to connect via zigbee2mqtt (resetting the service ) in loop - no answer.

Interesting was that after device was pluged to USB the led of device was constantly on, but when zigbee2mqtt process started the led of device was powering off after couple of seconds . 
After the led on adapter was power off, the device were still enumerated in linux as /dev/ttyACM0
I kept it in USB for couple of hours trying to connect via zigbee2mqtt (resetting the service ) in loop - no answer.

I had a plan to sniff the communication via packet sniffer. My goal was to put this device back online without reflashing.
I started to play with zigbee2mqtt settings, I added debug log, soft_reset_timeout... the device was still not responding, than I finally set the channel to 25 (from default channel: 11).

After the channel change to 25 the device is back online :+1: 

Now I switch back change to 11, the device is starting and communicating with zigbee2mqtt without problem.

@Koenkk any idea what is going on? If you want I can do some tests? But for me it is still like black magic.

```
~/Projects/zigbee2mqtt$ node index.js 
Zigbee2MQTT:debug 2021-01-20 15:10:58: Loaded state from file /home/andrzej/Projects/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-01-20 15:10:58: Logging to console and directory: '/home/andrzej/Projects/zigbee2mqtt/data/log/2021-01-20.15-10-58' filename: log.txt
Zigbee2MQTT:info  2021-01-20 15:10:58: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Zigbee2MQTT:info  2021-01-20 15:10:58: Starting zigbee-herdsman (0.13.46)
Zigbee2MQTT:debug 2021-01-20 15:10:58: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/home/andrzej/Projects/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/home/andrzej/Projects/zigbee2mqtt/data/database.db.backup","databasePath":"/home/andrzej/Projects/zigbee2mqtt/data/database.db","network":{"channelList":[25],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:info  2021-01-20 15:11:00: zigbee-herdsman started
Zigbee2MQTT:info  2021-01-20 15:11:00: Coordinator firmware version: '{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"}'
Zigbee2MQTT:debug 2021-01-20 15:11:00: Zigbee network parameters: {"channel":25,"extendedPanID":"0xdddddddddddddddd","panID":6754}
Zigbee2MQTT:info  2021-01-20 15:11:00: Currently 0 devices are joined:
Zigbee2MQTT:info  2021-01-20 15:11:00: Zigbee: disabling joining new devices.
Zigbee2MQTT:info  2021-01-20 15:11:00: Connecting to MQTT server at mqtt://localhost
Zigbee2MQTT:debug 2021-01-20 15:11:00: Using MQTT anonymous login
Zigbee2MQTT:info  2021-01-20 15:11:00: Connected to MQTT server
Zigbee2MQTT:info  2021-01-20 15:11:00: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
Zigbee2MQTT:info  2021-01-20 15:11:00: MQTT publish: topic 'zigbee2mqtt/bridge/info', payload '{"commit":"07cdc9d","config":{"advanced":{"adapter_concurrent":null,"adapter_delay":null,"availability_blacklist":[],"availability_blocklist":[],"availability_passlist":[],"availability_timeout":0,"availability_whitelist":[],"cache_state":true,"cache_state_persistent":true,"cache_state_send_on_startup":true,"channel":25,"elapsed":false,"ext_pan_id":[221,221,221,221,221,221,221,221],"homeassistant_discovery_topic":"homeassistant","homeassistant_legacy_triggers":true,"homeassistant_status_topic":"hass/status","last_seen":"disable","legacy_api":true,"log_directory":"/home/andrzej/Projects/zigbee2mqtt/data/log/%TIMESTAMP%","log_file":"log.txt","log_level":"debug","log_output":["console","file"],"log_rotation":true,"log_syslog":{},"pan_id":6754,"report":false,"soft_reset_timeout":0,"timestamp_format":"YYYY-MM-DD HH:mm:ss"},"ban":[],"blocklist":[],"device_options":{},"devices":{},"experimental":{"output":"json"},"external_converters":[],"groups":{},"homeassistant":false,"map_options":{"graphviz":{"colors":{"fill":{"coordinator":"#e04e5d","enddevice":"#fff8ce","router":"#4ea3e0"},"font":{"coordinator":"#ffffff","enddevice":"#000000","router":"#ffffff"},"line":{"active":"#009900","inactive":"#994444"}}}},"mqtt":{"base_topic":"zigbee2mqtt","force_disable_retain":false,"include_device_information":false,"server":"mqtt://localhost"},"passlist":[],"permit_join":false,"serial":{"disable_led":false,"port":"/dev/ttyACM0"},"whitelist":[]},"coordinator":{"meta":{"maintrel":3,"majorrel":2,"minorrel":6,"product":0,"revision":20190608,"transportrev":2},"type":"zStack12"},"log_level":"debug","network":{"channel":25,"extended_pan_id":"0xdddddddddddddddd","pan_id":6754},"permit_join":false,"version":"1.17.0"}'
```

Koenkk, 2021-01-20:
@renaiku @araczkowski please provide the herdsman debug logs from starting till the error.

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

araczkowski, 2021-01-22:
I can not reproduce this problem 

>  than I finally set the channel to 25 (from default channel: 11).
After the channel change to 25 the device is back online

Now it looks like the device is unblocked and works without a problem, even when I switching back to 11.
I asked on our forum to deliver the logs in case that someone will face this problem - the discussion is here (sorry, the site is in polish only):
https://ai-speaker.discourse.group/t/zigbee2mqtt-cc2531-przestalo-dzialac/743/94

@Koenkk sorry for bothering you again, but you probably the best person to answer for two things that interesting me: 

1. In your opinion switching the zigbee channel is something with can help in some cases for Error: SRSP - SYS - ping after 6000ms. If yes I can add this to FAQ (If you agree).
2. Why the default ZigBee channel is 11? 11 it coincides with the wifi, maybe better choice is 25? I'm thinking about switch to 25 in our project and I thought that I'll ask you first, maybe there is a good reason to stick with 11 as default?

Thanks for this great project :1st_place_medal: have a good day!

Koenkk, 2021-01-22:
> In your opinion switching the zigbee channel is something with can help in some cases for Error: SRSP - SYS - ping after 6000ms. If yes I can add this to FAQ (If you agree).

I don't think so, a replug or reflash is better, also changing channel requires repairing all devices.

> Why the default ZigBee channel is 11? 11 it coincides with the wifi, maybe better choice is 25? I'm thinking about switch to 25 in our project and I thought that I'll ask you first, maybe there is a good reason to stick with 11 as default?

Both 11 and 25 can collide with WiFi, it depends on the environment what the best channel is: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

danielo515, 2021-02-02:
I am also experiencing this problem.
I tried changing the tty port to the AMA0 becuase the original one I was using was not visible, but after plug and unplug the usb stick the original (and I guess correct one) appeared again.

My log is quite shor, so I will just paste it here:

<details>

```
pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.17.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:debug 2021-02-02 19:47:07: Loaded state from file /opt/zigbee2mqtt/data/state.json
Zigbee2MQTT:info  2021-02-02 19:47:07: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-02-02.19-46-56' filename: log.txt
Zigbee2MQTT:debug 2021-02-02 19:47:07: Removing old log directory '/opt/zigbee2mqtt/data/log/2021-02-02.19-23-27'
Zigbee2MQTT:info  2021-02-02 19:47:08: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
Zigbee2MQTT:info  2021-02-02 19:47:08: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:debug 2021-02-02 19:47:08: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","databasePath":"/opt/zigbee2mqtt/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
  zigbee-herdsman:adapter Path '/dev/ttyACM0' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +10s
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +16s
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-02-02 19:47:58: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-02 19:47:58: Failed to start zigbee
Zigbee2MQTT:error 2021-02-02 19:47:58: Exiting...
Zigbee2MQTT:error 2021-02-02 19:47:58: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```
</details>

This was working for long (6 months or so) until it stopped working so I updated to the latest zigbee2mqtt and this started happening.

danielo515, 2021-02-02:
Ok, this was easy, I just had to:

- unscrew the box where I have my rpi-zero (downstaris!)
- Try all sort of stupid combinations of plug,unplug,reboot, restart
- uninstall and re-install a lot of dependencies on my computer trying to understand why the hell the previously working cc-tool was now complaining about missing libs
- completely wipe the cc-tool and start from scratch
- Download the latest firmware and flash that one onte the CC2531
- Try again
- WORKS!! 🎉🎉

Jokes aside, I don't want to repeat this painful process in a month, so, is there any more reliable sniffer? Or they are all subject to the same kind of problems?

danielo515, 2021-02-03:
Ok, I probably talked to early.
I was so excited to see all those OK logs...
None of my devices is now responding. Do I need to re-pair them? Please, tell me that I don't have to re-pair them again...

paciuk, 2021-02-03:
```
~/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.17.0 start /data/data/pl.sviete.dom/files/home/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-02-03 07:42:46: Logging to console only'
Zigbee2MQTT:info  2021-02-03 07:42:46: Starting Zigbee2MQTT version 1.17.0 (commit #07cdc9d)
Zigbee2MQTT:info  2021-02-03 07:42:46: Starting zigbee-herdsman (0.13.46)
  zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/database.db","databaseBackupPath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/database.db.backup","backupPath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +5s
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +15s
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-02-03 07:43:27: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-03 07:43:27: Failed to start zigbee
Zigbee2MQTT:error 2021-02-03 07:43:27: Exiting...
Zigbee2MQTT:error 2021-02-03 07:43:27: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/data/data/pl.sviete.dom/files/home/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/data/data/pl.sviete.dom/files/home/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /data/data/pl.sviete.dom/files/home/.npm/_logs/2021-02-03T07_44_02_316Z-debug.log
~/zigbee2mqtt $ 
```

danielo515, 2021-02-03:
Ok, finally after turning off and then all all my lights (which acts as routers) my network is up again.
Seems that such thing forced the nodes to put themselves on search mode, and then they finally connected to each other.

vdelattre, 2021-02-07:
Hi
I have same error with conbee2
I plugged and unplugged 
Hass.io 2021.2.1
Zigbee2mqtt 1.7.1
Is It possible to increase timeout ?

RT2204, 2021-02-11:
Hello everybody,

I'm a new user of a Pi4 and i chave the same type of issue. I want to run Zigbee2MQTT on a Pi4 in order to connect it to jeedom. I use a Conbee2 as adapter which i plugged un unplugged several times

I changed the write access to the port as advise here : [https://www.zigbee2mqtt.io/information/FAQ.html#verify-that-the-user-you-run-zigbee2mqtt-as-has-write-access-to-the-port](url)

`SUBSYSTEM=="tty", ATTRS{idVendor}=="1cf1", ATTRS{idProduct}=="0030", ATTRS{serial}=="DE2405290", SYMLINK="ttyUSB.CC2531-01", OWNER="jeedom"`

Still have this when i start to start zigbee2MQTT

`> zigbee2mqtt@1.17.1 start /opt/zigbee2mqtt
> node index.js

Zigbee2MQTT:info  2021-02-11 15:36:40: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-02-11.15-36-39' filename: log.txt
Zigbee2MQTT:info  2021-02-11 15:36:40: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
Zigbee2MQTT:info  2021-02-11 15:36:40: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:error 2021-02-11 15:37:00: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-11 15:37:00: Failed to start zigbee
Zigbee2MQTT:error 2021-02-11 15:37:00: Exiting...
Zigbee2MQTT:error 2021-02-11 15:37:00: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/jeedom/.npm/_logs/2021-02-11T14_37_00_600Z-debug.log`

Here my configuration.yaml file

`#  Home Assistant integration (MQTT discovery)
homeassistant: false

# allow new devices to join
permit_join: true

# MQTT settings
mqtt:
  # MQTT base topic for zigbee2mqtt MQTT messages
  base_topic: zigbee2mqtt
  # MQTT server URL
  server: 'mqtt://localhost'
  # MQTT server authentication, uncomment if required:
  # user: my_user
  # password: my_password

# Serial settings
serial:
  port: /dev/ttyUSB.CC2531-01
`
Anyone has an idea about how to solve this ?

vdelattre, 2021-02-11:
I added this line in config and it worked for me:
serial:
  adapter: deconz

RT2204, 2021-02-11:
Ok it works ! Thank you !

srgazevedo, 2021-02-13:
Having same issue using Docker Container with **Zigbee2MQTT version 1.17.1 (commit #99274c7)**

CC2531 is flashed with Z-Stack_Home_1.2 (default) from here: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/default

Logs (using DEBUG=zigbee-herdsman* & advanced: log_level: debug)

```
Using '/app/data' as data directory

> zigbee2mqtt@1.17.1 start /app
> node index.js

Zigbee2MQTT:debug 2021-02-13 22:01:18: Loaded state from file /app/data/state.json
Zigbee2MQTT:info  2021-02-13 22:01:18: Logging to console and directory: '/app/data/log/2021-02-13.22-01-18' filename: log.txt
Zigbee2MQTT:debug 2021-02-13 22:01:18: Removing old log directory '/app/data/log/2021-02-13.21-53-32'
Zigbee2MQTT:info  2021-02-13 22:01:18: Starting Zigbee2MQTT version 1.17.1 (commit #99274c7)
Zigbee2MQTT:info  2021-02-13 22:01:18: Starting zigbee-herdsman (0.13.59)
Zigbee2MQTT:debug 2021-02-13 22:01:18: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null,"delay":null},"backupPath":"/app/data/coordinator_backup.json","databaseBackupPath":"/app/data/database.db.backup","databasePath":"/app/data/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
2021-02-13T22:01:18.820Z zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/app/data/database.db","databaseBackupPath":"/app/data/database.db.backup","backupPath":"/app/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}'
2021-02-13T22:01:18.821Z zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false}
2021-02-13T22:01:24.070Z zigbee-herdsman:adapter:zStack:znp:log Serialport opened
2021-02-13T22:01:40.091Z zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload
2021-02-13T22:01:40.091Z zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239]
2021-02-13T22:01:41.097Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-02-13T22:01:41.099Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-02-13T22:01:47.106Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-02-13T22:01:47.107Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
2021-02-13T22:01:53.113Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1}
2021-02-13T22:01:53.114Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32]
Zigbee2MQTT:error 2021-02-13 22:01:59: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-02-13 22:01:59: Failed to start zigbee
Zigbee2MQTT:error 2021-02-13 22:01:59: Exiting...
Zigbee2MQTT:error 2021-02-13 22:01:59: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.17.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.17.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2021-02-13T22_02_04_277Z-debug.log
```

Do I need to reflash it with Z-Stack_Home_1.2 (source_routing) @Koenkk ?

Thank you so much for your help in advance :)!
Awesome project.

srgazevedo, 2021-02-15:
Ok problem fixed, I reflashed it with the same firmware and it worked.

- fyi in my case this is a brand new cc2531 and never worked before.

Like proposed on this issue (https://github.com/jmichault/flash_cc2531/issues/10) , I needed to reflash it this way:

```
# 1. Build (Optional)
$ make

# 2. Test
$ ./cc_chipid -m 300

# 3. Save the flash content
$ ./cc_read save.hex -m 300

# 4. Erase
$ ./cc_erase -m 300

# 5. Flash
./cc_write -m 300 CC2531ZNP-Pro.hex
```

Thanks!

m0wlheld, 2021-03-01:
#6517 Having this error for the first time with 1.18.0. No need to re-flash the adapter (CC2652RB), going back to 1.17.1 avoids this error.

krzysztof-drobek, 2021-03-02:
> #6517 Having this error for the first time with 1.18.0. No need to re-flash the adapter (CC2652RB), going back to 1.17.1 avoids this error.

Same here. Reverting back to 1.17.1 fixes the problem.

Koenkk, 2021-03-02:
Let's continue in #6517 (this is also about non 1.18.0 releases)

closed 2021-03-02 by Koenkk

jszkiela72, 2021-03-21:
Zigbee2MQTT version: 1.18.1
Adapter hardware: CC2531
I have issue, zigbee keeps rebooting all the time since yesterday. logs attached. What could be the cause? 

Zigbee2MQTT:info  2021-03-21 12:44:27: Logging to console only'
Zigbee2MQTT:info  2021-03-21 12:44:28: Starting Zigbee2MQTT version 1.18.1 (commit #a43d1f3)
Zigbee2MQTT:info  2021-03-21 12:44:28: Starting zigbee-herdsman (0.13.71)
  zigbee-herdsman:adapter Failed to validate path: 'Error: spawn udevadm ENOENT' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[25]},"serialPort":{"path":"/dev/ttyACM0"},"databasePath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/database.db","databaseBackupPath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/database.db.backup","backupPath":"/data/data/pl.sviete.dom/files/home/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null,"delay":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyACM0 and {"baudRate":115200,"rtscts":false,"autoOpen":false} +0ms
  zigbee-herdsman:adapter:zStack:znp:log Serialport opened +22ms
  zigbee-herdsman:adapter:zStack:znp:log Writing skip bootloader payload +530ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> buffer [239] +0ms
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +0ms
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +1s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
  zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - ping - {"capabilities":1} +6s
  zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,0,33,1,32] +6s
Zigbee2MQTT:error 2021-03-21 12:44:48: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-03-21 12:44:48: Failed to start zigbee
Zigbee2MQTT:error 2021-03-21 12:44:48: Exiting...
Zigbee2MQTT:error 2021-03-21 12:44:48: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/data/data/pl.sviete.dom/files/home/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:94:31)
    at Generator.throw (<anonymous>)
    at rejected (/data/data/pl.sviete.dom/files/home/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.18.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /data/data/pl.sviete.dom/files/home/.npm/_logs/2021-03-21T11_44_48_539Z-debug.log
~/zigbee2mqtt $

greghesp, 2021-03-25:
> I added this line in config and it worked for me:
> 
> ```
> serial:
>   adapter: deconz
> ```

Ahhhh perfect, it works now thanks!

iNDeX999, 2021-04-06:
> > I added this line in config and it worked for me:
> > ```
> > serial:
> >   adapter: deconz
> > ```
> 

Hi,

"config" ??? 

What file do you mean??!

greghesp, 2021-04-06:
> > > I added this line in config and it worked for me:
> > > ```
> > > serial:
> > >   adapter: deconz
> > > ```
> 
> Hi,
> 
> "config" ???
> 
> What file do you mean??!

https://www.zigbee2mqtt.io/information/configuration.html

LittleBigDev, 2021-08-31:
> I added this line in config and it worked for me:
> 
> ```
> serial:
>   adapter: deconz
> ```

6 hours pulling my hair before I found this. Thx.

danielo515, 2021-09-05:
I tried adding the adapter deconz to my config file and now I am getting this error (note it just logs undefined at certain point):

```
Zigbee2MQTT:info  2021-09-05 11:31:41: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2021-09-05.11-31-40' filename: log.txt
Zigbee2MQTT:info  2021-09-05 11:31:42: Starting Zigbee2MQTT version 1.18.3 (commit #669489fe)
Zigbee2MQTT:info  2021-09-05 11:31:42: Starting zigbee-herdsman (0.13.92)
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2021-09-05 11:31:52: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2021-09-05 11:31:52: Failed to start zigbee
Zigbee2MQTT:error 2021-09-05 11:31:52: Check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start for possible solutions
Zigbee2MQTT:error 2021-09-05 11:31:52: Exiting...
Zigbee2MQTT:error 2021-09-05 11:31:52: undefined
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.18.3 start: `node index.js`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the zigbee2mqtt@1.18.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
```

I added `adapter:deconz` to my previous configuration, not sure if I should remove the rest. It looks like this now:

```
serial:
  adapter: deconz
   port: >-
      /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0008BDFED0-if00
advanced:
  channel: 11
```

szebenyib, 2022-01-22:
Ohh and don't be a n00b as me, some other container was using it about which I have completely forgotten.