Koenkk/zigbee2mqtt #4585: Aqara curtain(blinds) motor (ZNCLDJ11LM) Xiaomi does not publish its state after receiving the STOP command...

Trigger001, 2020-10-07:
### What happened

Aqara curtain(blinds) motor (ZNCLDJ11LM) Xiaomi does not publish its state after receiving the STOP command..
I made an automation in Node Red with a curtain and a switch. I wanted to make automation like in mihome one button open-stop-close

### How to reproduce it (minimal and precise)

Add Aqara curtain(blinds) motor (ZNCLDJ11LM) Xiaom to Zigbee2MQTT and enable MQTT integration with auto device generation in HomeAssistant. Then send a stop command to the curtains and watch the log

### Debug info
[image]

Zigbee2mqtt Current version: 1.15.0
Adapter hardware: CC2538

Koenkk, 2020-10-08:
Fixed. Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Trigger001, 2020-10-08:
Super!!! Coffee from me :-))

closed 2020-10-08 by Trigger001