Koenkk/zigbee2mqtt #4586: Add state "keep"

renne, 2020-10-07:
## Is your feature request related to a problem? Please describe
In case you want to change e.g. brightness and color at 6:00 o'clock am and back at 9 o'clock pm for all lights in the home but you don't want to change the state as you do not know which lights/rooms are in use one has to loop through all devices, query the state and set the new values with the queried state.

## Describe the solution you'd like
Additionally to "state": "ON|OFF|TOGGLE" I suggest to add "state": "KEEP" to the set command which should not change the current state of the device if other properties are changed.

## Describe alternatives you've considered
Setting brightness and color without state switches devices to state "ON".
Looping through all devices querying the current state.

Koenkk, 2020-10-08:
I've tried to implement this in https://github.com/Koenkk/zigbee-herdsman-converters/commit/645b53bf24c72a1da85aae5b1e8337e4e95a09c1, however it seems that Zigbee bulbs don't allow changing brightness when OFF rendering this option quite useless.

According to the Zigbee spec this behaviour can be overridden via the optionsoverrides, but the bulb I tested this with (IKEA, TRADFRI and Inrr) didn't support this unfortunately. 
[image]

sjorge, 2020-10-08:
Data point of note: Pre zigbee 3 trådfri bulbs did take color changes when off, but everything update to zb3 firmware no longer does. Only float panels still do because they are stuck on old firmware.

But changing the brightness would just turn on the bulbs on those too. Although that might have been z2m implementation.

renne, 2020-10-08:
How about storing the properties in Zigbee2MQTT and sending them when the device is switched on?

sjorge, 2020-10-08:
> How about storing the properties in Zigbee2MQTT and sending them when the device is switched on?

That would probably work, but I don't really think this is z2ms job to do this, this should be a level higher e.g. node-red or homeassistant level IMHO.

If you do implement this on z2m level it opens a whole new can of worms as to what to do with the device state... do you report it as what it currently is (IMHO, we should) or what is it going to be once it comes back online, etc.

renne, 2020-10-11:
Polling the current state, detect it's off, store the new values and wait for the bulb to change to off state is quite a complicated flow in e.g. Node-Red which has to be implemented over and over again for each bulb.

I suggest to send the properties (brightness, color, etc.) to the device and do a retrieve_state. If state is "OFF" Zigbbe2MQTT stores the values. Via MQTT Zigbee2MQTT should announce the new values with "OFF" state. The moment the device switches to "ON" state Zigbee2MQTT should send the stored values and return the retrieve_state to MQTT.

closed 2020-11-18 by github-actions[bot]