Koenkk/zigbee2mqtt #4590: TS0601_thermostat calibration does not support negative values

henrikwils, 2020-10-07:
## What happened
TS0601_thermostat calibration does not support negative values

## What did you expect to happen
Setting negative value to calibrate in negative direction.

## How to reproduce it (minimal and precise)
Trying to set calibration with i.e.:

topic: zigbee2mqtt/FRIENDLY_NAME/set/local_temperature_calibration
payload: -6.0

The set temperature responds as if it was a positive value. Checking the confirmation MQTT message, shows that the value is indeed positive.

Received 9:59:49 PM
QoS: 0
Payload: auto_lock: MANUAL
away_preset_days: 1
away_preset_temperature: 15
boost_time: 300
child_lock: UNLOCKED
comfort_temperature: 20
current_heating_setpoint: '22.5'
eco_temperature: 15
force: normal
holidays:
  - hour: 134
    minute: 0
    temperature: 20
  - hour: 8
    minute: 0
    temperature: 15
  - hour: 11
    minute: 30
    temperature: 15
  - hour: 12
    minute: 30
    temperature: 15
  - hour: 17
    minute: 30
    temperature: 20
  - hour: 22
    minute: 0
    temperature: 15
linkquality: 72
local_temperature: '29.5'
local_temperature_calibration: '6.0'
max_temperature: 35
min_temperature: 5
position: 0
preset: manual
system_mode: manual
week: 5+2
window_detection: 'OFF'
window_detection_params:
  minutes: 10
  temperature: 5
workdays:
  - hour: 6
    minute: 0
    temperature: 20
  - hour: 8
    minute: 0
    temperature: 15
  - hour: 11
    minute: 30
    temperature: 15
  - hour: 12
    minute: 30
    temperature: 15
  - hour: 17
    minute: 30
    temperature: 20
  - hour: 22
    minute: 0
    temperature: 15

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC26X2R1
Adapter firmware version: 20191106

artem-sedykh, 2020-10-11:
I will add an error log:
`Publish 'set' 'local_temperature_calibration' to 'tvr_test' failed: 'RangeError [ERR_OUT_OF_RANGE]: Command 0xbc33acfffe571ef9/1 manuSpecificTuyaDimmer.setData({"status":0,"transid":196,"dp":556,"fn":0,"data":[4,0,0,0,-10]}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (The value of "value" is out of range. It must be >= 0 and <= 255. Received -10)'
Zigbee2MQTT:debug 2020-10-11 14:24:11: RangeError [ERR_OUT_OF_RANGE]: The value of "value" is out of range. It must be >= 0 and <= 255. Received -10
    at writeU_Int8 (internal/buffer.js:729:11)
    at Buffer.writeUInt8 (internal/buffer.js:739:10)
    at BuffaloZcl.writeUInt8 (/app/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:30:21)
    at BuffaloZcl.writeListUInt8 (/app/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:157:18)
    at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/dist/buffalo/buffalo.js:247:18)
    at BuffaloZcl.write (/app/node_modules/zigbee-herdsman/dist/zcl/buffaloZcl.js:303:26)
    at ZclFrame.writePayloadCluster (/app/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:144:21)
    at ZclFrame.toBuffer (/app/node_modules/zigbee-herdsman/dist/zcl/zclFrame.js:74:18)
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:307:163)
    at Generator.next (<anonymous>)
`

danieledwardgeorgehitchcock, 2020-10-12:
I am pretty sure that the temperature calibration parameter should be set at the actual temperature of the room rather than an offset thereof. This way, the sensor automatically calculates the offset be it positive, or negative.

henrikwils, 2020-10-12:
I was pretty sure about that as well, but it DOESN'T WORK THAT WAY! If you set that temperature, it just adds it to the current measured value. Otherwise, I wouldn't have spent hours trying to figure it out, and never made this bug report.

danieledwardgeorgehitchcock, 2020-10-12:
Apologies for trying to help - you did not mention in your OP, anything about trying this approach hence, my suggestion. At least people now know to rule it out of their diagnoses and prevent further abrupt replies!

> I was pretty sure about that as well, but it DOESN'T WORK THAT WAY! If you set that temperature, it just adds it to the current measured value. Otherwise, I wouldn't have spent hours trying to figure it out, and never made this bug report.

henrikwils, 2020-10-12:
It looks like you even have it - Why don't you try it to see what you get? Could be there are different models that behave differently? I know I have one thermostat, that doesn't play well with Z2M at all.

bkupidura, 2020-10-13:
@henrikwils indeed looks like its offset, not target temperature. Looks like valve is working in 0-100 degrees range (at least in Celsius).

So if thermostat is showing current temp as ex. 25, and you want to change it to 20, just send `94` (100-25+20-1). Im not sure if this is expected (im almost sure that its not :)). But, looks like it works. Please dont ask why '-1', i dont have clue. Just started playing with this thermostat.

insipiens, 2020-10-14:
@henrikwils This was mentioned in #3821 

A small change to your code should solve it in the toZigbee.js file located here: /opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js

add the line: _if (value < 0) value = 4096 + value;_
```
    moes_thermostat_calibration: {
        key: ['local_temperature_calibration'],
        convertSet: async (entity, key, value, meta) => {
            if (value < 0) value = 4096 + value;
            const payloadValue = utils.convertDecimalValueTo2ByteHexArray(value);
            sendTuyaCommand(entity, 539, 0, [4, 0, 0, ...payloadValue]);
        },
    },
```

if 4096 doesn't work try 128

If that fails raise a question in #3821 someone there will have solved it.

closed 2020-11-21 by github-actions[bot]

Quapcio, 2020-12-13:
couple minutes ago I managed to set negative value of local temperature compensation
you shuld set value for 256 - [negative value to set]

-1 -> 255
-9 -> 247

it looks like the variable use unsigned char instead of signed char but it's more complicated, bebause fractional numers are accepted resulting in negitive numer rounded to 0.5 degree.

Quapcio, 2020-12-13:
maybe rather (256 + [negtive number to set]) % 256 to be more precise...

un1te, 2020-12-22:
I found new solution for calibration.
I noticed that if thermostat already has calibration value ayou need to calibrate it calibration always worn.
So need first to set calibration to 128 (=0) the delay several seconds, then set new value.

so my automation is:

````
- alias: Calibrate thermostat child
  trigger:
    - platform: state
      entity_id: sensor.temp_child_temperature
  action:
    - service: mqtt.publish
      data:
        topic: 'zigbee2mqtt/thermo_child/set/local_temperature_calibration'
        payload: 128
    - delay: '00:00:03'
    - service: mqtt.publish
      data_template:
        topic: 'zigbee2mqtt/thermo_child/set/local_temperature_calibration'
        payload_template: >-
          {% if states('sensor.temp_child_temperature')|float - state_attr('climate.thermo_child_climate', 'current_temperature')|float >= 0.5 %}
            {{ 128 + (states('sensor.temp_child_temperature')|float - state_attr('climate.thermo_child_climate', 'current_temperature')|float) | round (1) }}
          {% elif states('sensor.temp_child_temperature')|float - state_attr('climate.thermo_child_climate', 'current_temperature')|float <= (-0.5) %}
            {{ 128 - (state_attr('climate.thermo_child_climate', 'current_temperature')|float - states('sensor.temp_child_temperature')|float) | round (1)}}
          {% endif %}
```

Stooovie, 2024-02-11:
Could we get a larger offset than +- 3C? Mine is off by 5C (lower than actual).