Koenkk/zigbee2mqtt #4597: ZIGBEE tuya light switch (4 buttons)

dror23, 2020-10-08:
## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

dror23, 2020-10-08:
First of all thank you for your support!
Recently you added the following switch to the supporting devices:

https://www.aliexpress.com/item/4001181765345.html?srcSns=sns_WhatsApp&spreadType=socialShare&bizType=ProductDetail&tt=MG&aff_platform=default&sk=_dVohEV8&mergeHashcode=74176412736&description=%E2%82%AA+46.10++40%25OFF+%7C+ZigBee+smart+home++wall+touch+switch%2Cwireless+Intelligent+light+switch%2C2%2F3+Way%2CAPP+Control%2Cworks+with+google+home+aleax%2CUSUK&aff_trace_key=9c0d7cbb2e27488790bd95c491d5c198-1602151252496-01262-_dVohEV8&businessType=ProductDetail&title=%E2%82%AA+46.10++40%25OFF+%7C+ZigBee+smart+home++wall+touch+switch%2Cwireless+Intelligent+light+switch%2C2%2F3+Way%2CAPP+Control%2Cworks+with+google+home+aleax%2CUSUK&platform=AE&terminal_id=7e336818915a425d83ad68b0f4c9ffc2&templateKey=white_backgroup_101

But there is a little problem, I'm using 4 button switch and the 4th button does not send indication to HA when I'm pressing on it physically. The rest of the buttons works fine.

Again, thank you for your time and support 🙏

Koenkk, 2020-10-08:
- Can you share the data/database.db entry of this device?
- Can you try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and check the log if it's ok?

dror23, 2020-10-08:
Of cours, I've try to reconfigure but I'm not sure how exactly...
Here is the data from the log:

https://paste.ofcode.org/G8X62W5CA5KZiQKePUp9XM

Hope it will help!

dror23, 2020-10-13:
> * Can you share the data/database.db entry of this device?
> * Can you try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and check the log if it's ok?

Hey Koenkk Have you manage to fix it?

Koenkk, 2020-10-13:
- I don't see a configure being executed in your log
- Please share the `data/database.db` entry.

dror23, 2020-10-14:
> * I don't see a configure being executed in your log
> * Please share the `data/database.db` entry.
Log:
https://paste.ofcode.org/HR5f6U7n3DQgH8V4qV4a68

[database.zip](https://github.com/Koenkk/zigbee2mqtt/files/5379313/database.zip)

Koenkk, 2020-10-14:
Is the problem with model `TS0601_switch` , `ieeeAddr`: `0xbc33acfffe189ace`?

dror23, 2020-10-14:
> Is the problem with model `TS0601_switch` , `ieeeAddr`: `0xbc33acfffe189ace`?

Yes, that the one.

Koenkk, 2020-10-14:
Can you provide the herdsman debug logging when pressing the 4th button?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

dror23, 2020-10-14:
> Can you provide the herdsman debug logging when pressing the 4th button?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

OK I did it and than restarted the addon and I'm attaching the file after the changes.
[ZIGBEE2MQTT.zip](https://github.com/Koenkk/zigbee2mqtt/files/5379865/ZIGBEE2MQTT.zip)

Koenkk, 2020-10-14:
Can you point me to what is exactly generated when you click the 4th button?

dror23, 2020-10-14:
> Can you point me to what is exactly generated when you click the 4th button?

Thats the problem, when I'm pressing on the 4th button (physically) nothing happens on HA, the status stay the same, not llike the other buttons on the same switch...
A friend of my suggested that the problem is Cluster but I have no idea...

Koenkk, 2020-10-15:
Normally a 4 gang Zigbee switch has 4 endpoints, each endpoint represents one gang. To have feedback for all endpoints reporting is setup on each endpoint.

Unfortunately TuYa devices often deviate from this standard. In this case your device only has 1 endpoint (you see `"epList":[1]` in the database.db entry). Therefore I have no clue about:
- Why does reporting work for gang 2 and 3, I expect it to only work for gang 1, since only for that reporting is setup.
- Why does reporting not work for gang 4, it does for 2 and 3 while no reporting is setup for that.

To further investigate, you should check how this works with the original TuYa gateway:
- Does it have the same issue? (reports for gang 1, 2 and 3 but not for 4). If yes, it's an issue in the device not solvable by Zigbee2MQTT
- If no, sniffing the traffic might reveal what we need to do in order to fix this (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

ARafaelSF, 2021-01-16:
> * Does it have the same issue? (reports for gang 1, 2 and 3 but not for 4). If yes, it's an issue in the device not solvable by Zigbee2MQTT

Yes, works ok with tuya hub.

> * If no, sniffing the traffic might reveal what we need to do in order to fix this (https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html)

I don't have a debugger, but I already bought it. When it arrives I will do it.

ARafaelSF, 2021-01-21:
@Koenkk 

I made a sniff

The sequence is:

1 - Paring
2 - On Gang 1 - Off Gang 1 
3 - On Gang 2 - Off Gang 2 
4 - On Gang 3 - Off Gang 3 
5 - On Gang 4 - Off Gang 4 (this not work in zigbee2mqtt)

[https://1drv.ms/u/s!AtfNPxDs_iedo85HPTjfYg79pM3DpQ?e=niGsQw](url)

Please see if it is enough or if I need to do anything else.

Thanks

Koenkk, 2021-01-22:
@ARafaelSF cool that you've managed to sniff. Can you make 2 sniffs (one with TuYa hub and one with Z2M) which contain: pair of the device + wait 20 seconds + click on 3th gang, click on 4th gang.

ARafaelSF, 2021-01-22:
@Koenkk I Tried it.

But it looks like I’m not listening tuya hub, even though it’s working with the app. I'm not sure if I need to change any settings to listen tuya.  Please tell me if i did something wrong

[image]

Attached are the two files, with cc5231 + Z2M and Tuya. When I did with tuya I turned off the cc2531

Pair of the device + wait 20 seconds + click on 3th gang (on and off), click on 4th gang (on and off)

CC2531+Z2M - https://1drv.ms/u/s!AtfNPxDs_iedo85RY4-IInK5Yq6t_Q?e=7ysfjI
Tuya - https://1drv.ms/u/s!AtfNPxDs_iedo85QMFmy_b_-h__5fA?e=4O6771

Other information that may interest you.

In Z2M the device looks like ok and I can controll all buttons. But it still doesn't get the status update if I press on the device.

[image]

Koenkk, 2021-01-23:
Can you update to the latest dev branch, and apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2137/files to devices.js

Next pair the device with Z2M and sniff the traffic from it. Also make a sniff when controlling the 4 and 3th gang of this device and provide me the packet send (left column in wireshark "No.")

ARafaelSF, 2021-01-24:
@Koenkk 

I did as you asked. Update Z2M edge and change devices.js

Here it sniff: https://1drv.ms/u/s!AtfNPxDs_iedo85sbVjO9AUz4FYBYw?e=Oe5V8w

The sequence is:

Nº 279 - Gang 3 - ON
Nº 334 - Gang 3 - OFF
Nº 348 - Gang 4 - ON
Nº 365 - Gang 4 - OFF

Koenkk, 2021-01-24:
@ARafaelSF can you apply the changes from https://github.com/Koenkk/zigbee-herdsman-converters/pull/2137/files factory reset the device and pair it, do you get updates of the 4th gang now?

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

ARafaelSF, 2021-01-24:
@Koenkk soyry, but I didn't understand how to do this part.

I´m using hass.io. So as I understand it I need to use external_converters.

I put this in yalm file

external_converters:
  - new_device.js

And this file, in share\zigbee2mqtt\new_device.js

https://1drv.ms/u/s!AtfNPxDs_iedo85utcxHK4UQiacmeA?e=tjwh8C

But, my file is wrong, and I can´t fix it. 

Sorry for that kind of question, but can you help me?

Koenkk, 2021-01-25:
@ARafaelSF since you have an additional CC2531, probably easier to setup a bare metal environment by following: https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html

ARafaelSF, 2021-01-25:
@Koenkk 

It works

[image]

The sequence of 4 buttons, On and Off, reporting do Mqtt.

Thank you very much,

dror23, 2021-01-25:
@Koenkk Thank you very much vor your assistant!

dror23, 2021-01-26:
> @Koenkk
> 
> It works
> 
> [image]
> 
> The sequence of 4 buttons, On and Off, reporting do Mqtt.
> 
> Thank you very much,

Hey @ARafaelSF how did you manage to fix the problem?
I deleted the addon and reinstalled it and the problem still exist... 
In addition Iv'e delete the switch manually but still, doesn't work...

ARafaelSF, 2021-01-26:
> > @Koenkk
> > It works
> > [image]
> > The sequence of 4 buttons, On and Off, reporting do Mqtt.
> > Thank you very much,
> 
> Hey @ARafaelSF how did you manage to fix the problem?
> I deleted the addon and reinstalled it and the problem still exist...
> In addition Iv'e delete the switch manually but still, doesn't work...

My check was outside HA.
The code is not published yet. he should do it as soon as he sees my confirmation that the changes have worked.

Koenkk, 2021-01-26:
Merged!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2021-01-26 by Koenkk

oferwald, 2021-01-27:
@Koenkk Thanks, I can confirm that it works for me with the latest-dev branch

dbroome1978, 2021-07-04:
I know this is an old thread, just curious if the fix is still only in the dev branch?

I have the same device (or close to it) but with a different fingerprint _TZE200_1ozguk6x

https://www.moeshouse.com/collections/tuya-zigbee-switch/products/zigbee-wall-touch-smart-light-switch-with-neutral-wire-no-neutral-wire-no-capacitor-needed-smart-life-tuya-2-3-way-muilti-control-association-hub-required-4-gang-white-us

Btw, this was the only zigbee2mqtt was the only gateway to pick up my ZTS-US_3gang (it read it as ZTS-EU_3gang but close enough)