Koenkk/zigbee2mqtt #4598: Moes MS-104Z-1 (aka TuYa TS0121_switch)

Boia11, 2020-10-08:
I have install 4 of this switch [Moes MS-104Z-1](https://www.zigbee2mqtt.io/devices/TS0121_switch.html) but log report errors.
## What happened
At addon start the log say that for all switch install
```
INFO: Configuring 'Corridoietto'
ERROR: Failed to configure 'Corridoietto', attempt 1 (Error: ConfigureReporting 0xec1bbdfffefd501c/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34201 - 1 - 6 - 6 - 7 after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7))
ERROR: Failed to configure 'Corridoietto', attempt 2 (Error: ConfigureReporting 0xec1bbdfffefd501c/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34201 - 1 - 6 - 6 - 7 after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7))
ERROR: Failed to configure 'Corridoietto', attempt 3 (Error: ConfigureReporting 0xec1bbdfffefd501c/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 34201 - 1 - 8 - 6 - 7 after 10000ms) at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35) at listOnTimeout (internal/timers.js:549:17) at processTimers (internal/timers.js:492:7))
INFO: MQTT publish: topic 'zigbee2mqtt/Corridoietto', payload '{"linkquality":80,"state":"ON"}'
```

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2530+CC2592
Adapter firmware version: zStack12 - 20190608

Koenkk, 2020-10-08:
When toggling the switch via the physical switch, do you get a state update?

Boia11, 2020-10-08:
yes, but when the addon start do that and slows startup...the frontend is not available until it has failed to configure 3 times for all 4 switches

Koenkk, 2020-10-08:
Fixed the issue.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Boia11, 2020-10-08:
Thanks! I'll wait it

Boia11, 2020-11-08:
Perfect! Now work normal

closed 2020-11-08 by Boia11