Koenkk/zigbee2mqtt #4600: Sonoff BASICZBR3 does not work well if included in a group of lights

Josov, 2020-10-08:
The problem is caused by the fact the Sonoff device only accepts commands like {state: "ON"} and {state: "OFF"}. In groups you will often extend the state command with commands for brightness, transition and or color. The Sonoff device does not accept these commands and the herdsman replies with an error without executing the state command. In my too bad, not necessary and it does not bring anything.
The Philips HUE Power Plug and IKEA Power Switch work different, both accept state commands combined with commands for color or brightness. Of course these commands are not deployed but they do not cause errors. Albeit they do switch ON of OFF. This allows the IKEA and Philips Power Plugs to be included in a group and it works fine!
I believe all power plugs should accept additional commands besides the state command, execute the state command and not return errors.

olialb, 2020-10-08:
Hey @Josov ,

I think this can connected to my issue report:  https://github.com/Koenkk/zigbee2mqtt/issues/4558

Koenkk, 2020-10-09:
@Josov I didn't know there was already an issue open, let's continue in #4558

closed 2020-10-09 by Koenkk