Koenkk/zigbee2mqtt #4601: Cant Pair Ikea Tradfri E1810 remote

coxtor, 2020-10-08:
## What happened
I want to join my new E1810 remote in my network. I followed the instructions and the device led blinks as described, however the device does not pair with zigbee2mqtt - other devices still work and can be paired.

## What did you expect to happen
My E1810 announces itsself to zigbee2mqtt

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2531

NoWhere2Go19, 2020-10-08:
I have seveal of these remotes workng in my network. However the steps you can undertake is:
- make sure that devices can join your network
- reset the remote and put it again into pairing mode. Keep clicking buttons, so the remote keeps awake
- take a look into your log (have at least info enabled for this) and see if anything announces itself

That all it. If you're using a RPi, reboot it before doing all of this.

olialb, 2020-10-08:
I can confirm it works! I just pair 3 new E1810 devices. But it took me also a while until they paired successfully....
Sometimes it paired as unknown device and I had to delete it an try again. But finally all devices work perfectly. 👍

NoWhere2Go19, 2020-10-08:
Cool perfect, could you close this topic then as it's resolved. Th

coxtor, 2020-10-09:
Hi thanks for the replies.
@NoWhere2Go19  I have completed all steps. Also, I am running zigbee2mqtt in docker. 
However something has changed - and now I am completely lost!
1. After holding the pairing button for a long time the log outputted:
`Zigbee2MQTT:warn  2020-10-09 10:11:19: Device '0x7cb03eaa00a75dfa' left the network`
which is a Osram Switch in my network.
2. I then removed the entry in my configuration (commented it out)
3. Now this is really weird:  The remote seems to be in the network somehow - there was no prior config or anything - its fresh out of the box and the only one of its kind in my network - When I push the power button on the e1810 the switch (that left the network and is no longer part of my config) toggles its power state. All other buttons do nothing. Also zigbee2mqtt did not output any information about a new device joining etc. as it is common. 

Has anybody got a clue what is going on?

NoWhere2Go19, 2020-10-10:
I guess you accidentally bound the remote to a device. You don't necessarily need a bridge for that.
For an example, simply look at Ikea's sell of a build and the remote (browse the catalog or look on yt). You can operate a bulb with the remote without having the bridge at all.
I guess this is what you accidentally did. Simply reset the remote and switch, add them to the Z2m network and all good

coxtor, 2020-10-15:
Okay it's sorted out now, I got it connected. However, I still do not know what exactly the issue was. Once I even successfully paired it and zigbee2mqtt stated it was an unknown device. Another attempt made it work as expected in the end. Thank you for your helpful tips.

closed 2020-10-15 by coxtor

flixx, 2021-05-16:
Hello,

I've still problems pairing my E1810. Nothing appears in the logs at all. No error, no unknown device.

Things I tried:
* Pressing the pairing button on the remote 4 times many times
* Placing the Remote very close to the Controler
* Long-Pressing the pairing button many times
* Randomly pressing other buttons after pressing the pairing button 4 times for some minutes
* Removing the battery and inserting it again
* Restarting zigbee2mqtt
* Turning off all other zigbee devices 

Also the controller is in pairing mode all the time and other devices are succesfully paired.

I'm using Zigbee2MQTT version 1.18.3 (commit #f2e39af) and Controller version 20201127...
I don't know the firmware version of th E1810 because I don't have an IKEA gateway...

Any hints for me?
Felix

flixx, 2021-05-19:
Allright, so what helped me was just waiting a few days after reseting. 
I just long-pressed the pair button and it connected (without pressing 4 times first). 
I see now that the batery is at 21% (and who knows if my replacement batery was full). So that might also be it...

pw1124, 2022-01-25:
Just to add to this in case it helps anyone down the line - I spent at least 4 hours trying to pair this thing with no luck, and in the end what worked instantly was putting it next to a router (in my case, a zigbee light bulb) away from the CC2531 coordinator and pressing the pairing button 4 times.  Before this, I had been putting it next to my coordinator and wondering why I wasn't getting anything whatsoever in the logs.

rtrinks, 2022-02-20:
I also have a problem with the E1810 remote. Use two with Zigbee2mqtt and always went well.
Successfully upgraded both via OTA and still worked perfectly after that.
I had to get a new meter from Liander, so the power was off. And then it stopped working.
Then I saw that both remotes were linked to Zigbee at the neighbors in HA Zigbee2mqtt.

I removed them from there and turned off pairing.
Then I paired it again in my own system, but that doesn't work anymore. Both no more.

I removed an Ikea lamp, which was working and I can no longer connect it.
When I grab a new sensor from Xiaomi, it pairs without any problems.

Have already stopped Zigbee2mqtt at the neighbors and even then it does not work.
But if I connect the E1810 to an Ikea lamp, it works and I have already tried all of the above.
Always returns with the message device has not successfully been paired

What could be wrong? For the devices that cannot connect, I see in log:

Info Device '0x000d6ffffe555084' joined
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6ffffe555084","ieee_address":"0x000d6ffffe555084"},"type":"device_joined"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6ffffe555084","ieee_address":"0x000d6ffffe555084","status":"started"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_started","meta":{"friendly_name":"0x000d6ffffe555084"},"type":"pairing"}'
Error Failed to interview '0x000d6ffffe555084', device has not successfully been paired
Info MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"0x000d6ffffe555084","ieee_address":"0x000d6ffffe555084","status":"failed"},"type":"device_interview"}'
Info MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"interview_failed","meta":{"friendly_name":"0x000d6ffffe555084"},"type":"pairing"}'

mathieuf20, 2022-03-13:
@rtrinks I also have the same issue an E1743 after OTA upgrade. Had to re-pair it after a few days for some reason and I never got able to re-pair it... D

rtrinks, 2022-03-17:
@Koenkk Is there a possibility to realize a solution for this? The problem is after an OTA update and disconnection of E1743 that it no longer wants to connect. I also have the same with an Ikea lamp led1732g11

Other Zigbee devices from Xioami, for example, are no problem at all.

oystermon, 2022-03-27:
Hi all, I recently faced a similar problem when I was trying to re-pair my E1810 after doing some hardware upgrades to my home server.

I did most of the things @flixx listed, and didn't get anywhere. 
But then I realised I changed the usb port I connected my zigbee coordinator to, so I swapped it back the original usb port I used and placed the remote right next to it, and held the pairing button and my remote got detected almost instantly.

Not sure if it helps but that was my experience..

rossowl, 2023-07-07:
1. turn off zigbee2mqtt
2. unplug coordinator and wait a few seconds
3. plug coordinator back to usb
4. start zigbee2mqtt
5. enable pairing
6. place ikea remote close to coordinator (10cm)
7. press pairing button 4 times
8. wait for succesfull pairing and configuring
9. check binding to coordinator cluster 1 - OnOff, genPollCtrl, PowerCfg