Koenkk/zigbee2mqtt #4607: Fail to start zigbee-herdsman

poenneby, 2020-10-08:
## What happened

Upgraded to zigbee2mqtt-1.15.0

2020-10-08T22:20:15: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-08 22:20:21: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-08.22-20-21' filename: log.txt
Zigbee2MQTT:debug 2020-10-08 22:20:21: Removing old log directory '/share/zigbee2mqtt/log/2020-10-08.22-18-43'
Zigbee2MQTT:debug 2020-10-08 22:20:23: Loaded state from file /share/zigbee2mqtt/state.json
Zigbee2MQTT:info  2020-10-08 22:20:23: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-08 22:20:23: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-08 22:20:23: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2020-10-08 22:20:25: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-08 22:20:25: Failed to start zigbee
Zigbee2MQTT:error 2020-10-08 22:20:25: Exiting...
Zigbee2MQTT:error 2020-10-08 22:20:25: Error: SREQ '--> AF - register - {"appdeviceid":5,"appdevver":0,"appnuminclusters":0,"appinclusterlist":[],"appnumoutclusters":0,"appoutclusterlist":[],"latencyreq":0,"endpoint":1,"appprofid":260}' failed with status '(0xb8: APS_DUPLICATE_ENTRY)' (expected '(0x00: SUCCESS)')
    at Znp.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:291:27)
    at Generator.next (<anonymous>)
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm 
ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR!
 A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-08T20_20_25_464Z-debug.log

## What did you expect to happen

That the server started

## How to reproduce it (minimal and precise)

Restart service 

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

poenneby, 2020-10-08:
A reboot of the entire system allows zigbee-herdsman to start, but restarting the service from supervisor puts it back into a non starting state continuosly attempting to start a only to fail with the above error.

Koenkk, 2020-10-09:
Can you provide the herdsman debug logging of this?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

poenneby, 2020-10-09:
Hi @Koenkk - thanks for the quick reply, here's the log with debug enabled:

2020-10-09T10:41:13.498Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,0,15,0,37]
2020-10-09T10:41:13.521Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,85,62]
2020-10-09T10:41:13.521Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,85,62]
2020-10-09T10:41:13.522Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,85] - 62
2020-10-09T10:41:13.522Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[85]}}
2020-10-09T10:41:13.522Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-09T10:41:13.523Z zigbee-herdsman:adapter:zStack:startZnp Item 'hasConfigured' is valid
2020-10-09T10:41:13.526Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":132,"len":4,"offset":0,"value":{"type":"Buffer","data":[0,8,0,0]}}
2020-10-09T10:41:13.529Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,132,0,0,174]
2020-10-09T10:41:13.539Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,6,97,8,0,4,0,8,0,0,99]
2020-10-09T10:41:13.540Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,6,97,8,0,4,0,8,0,0,99]
2020-10-09T10:41:13.541Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 6 - 3 - 1 - 8 - [0,4,0,8,0,0] - 99
2020-10-09T10:41:13.542Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":4,"value":{"type":"Buffer","data":[0,8,0,0]}}
2020-10-09T10:41:13.542Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-09T10:41:13.544Z zigbee-herdsman:adapter:zStack:startZnp Item 'channelList' is valid
2020-10-09T10:41:13.545Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":99,"len":1,"offset":0,"value":{"type":"Buffer","data":[0]}}
2020-10-09T10:41:13.546Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,99,0,0,73]
2020-10-09T10:41:13.555Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,97,8,0,1,0,107]
2020-10-09T10:41:13.556Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,97,8,0,1,0,107]
2020-10-09T10:41:13.556Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 3 - 1 - 8 - [0,1,0] - 107
2020-10-09T10:41:13.556Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":1,"value":{"type":"Buffer","data":[0]}}
2020-10-09T10:41:13.557Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-09T10:41:13.557Z zigbee-herdsman:adapter:zStack:startZnp Item 'networkKeyDistribute' is valid
2020-10-09T10:41:13.559Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SAPI - readConfiguration - {"id":98,"configid":98,"len":16,"offset":0,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}
2020-10-09T10:41:13.559Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,1,38,4,98,65]
2020-10-09T10:41:13.576Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,19,102,4,0,98,16,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,0]
2020-10-09T10:41:13.577Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,19,102,4,0,98,16,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13,0]
2020-10-09T10:41:13.577Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 19 - 3 - 6 - 4 - [0,98,16,1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13] - 0
2020-10-09T10:41:13.578Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SAPI - readConfiguration - {"status":0,"configid":98,"len":16,"value":{"type":"Buffer","data":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13]}}
2020-10-09T10:41:13.578Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-09T10:41:13.579Z zigbee-herdsman:adapter:zStack:startZnp Item 'networkKey' is valid
2020-10-09T10:41:13.580Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":131,"len":2,"offset":0,"value":{"type":"Buffer","data":[98,26]}}
2020-10-09T10:41:13.581Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,131,0,0,169]
2020-10-09T10:41:13.603Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,4,97,8,0,2,98,26,23]
2020-10-09T10:41:13.603Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,4,97,8,0,2,98,26,23]
2020-10-09T10:41:13.603Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 4 - 3 - 1 - 8 - [0,2,98,26] - 23
2020-10-09T10:41:13.604Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- SYS - osalNvRead - {"status":0,"len":2,"value":{"type":"Buffer","data":[98,26]}}
2020-10-09T10:41:13.604Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-10-09T10:41:13.605Z zigbee-herdsman:adapter:zStack:startZnp Item 'panID' is valid
2020-10-09T10:41:13.607Z zigbee-herdsman:adapter:zStack:znp:SREQ --> SYS - osalNvRead - {"id":45,"len":8,"offset":0,"value":{"type":"Buffer","data":[221,221,221,221,221,221,221,221]}}
2020-10-09T10:41:13.609Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,3,33,8,45,0,0,7]
2020-10-09T10:41:13.632Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,10,97,8,0,8,221,221,221,221,221,221,221,221,107]
2020-10-09T10:41:13.632Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,10,97,8,0,8,221,221,221,221,221,221,221,221,107]
2020-10-09T10:41:13.633Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 10 - 3 - 1 - 8 - [0,8,221,221,221,221,221,221,221,221] - 107
[... cut ...]
    at fulfilled (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:24:58)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-10-09T10_41_13_765Z-debug.log

oldfart101, 2020-10-09:
same problem, same version.
```
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] done.
[services.d] starting services
[services.d] done.
./run.sh: line 17: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-10-09T11:39:41: PM2 log: Launching in no daemon mode
2020-10-09T11:39:41: PM2 log: App [npm:0] starting in -fork mode-
2020-10-09T11:39:41: PM2 log: App [npm:0] online
> zigbee2mqtt@1.15.0 start /zigbee2mqtt-1.15.0
> node index.js
Zigbee2MQTT:info  2020-10-09 11:39:46: Logging to console and directory: '/share/zigbee2mqtt/log/2020-10-09.11-39-46' filename: log.txt
Zigbee2MQTT:debug 2020-10-09 11:39:46: Removing old log directory '/share/zigbee2mqtt/log/2020-10-09.11-17-19'
Zigbee2MQTT:debug 2020-10-09 11:39:48: Can't load state from file /share/zigbee2mqtt/state.json (doesn't exist)
Zigbee2MQTT:info  2020-10-09 11:39:48: Starting Zigbee2MQTT version 1.15.0 (commit #1.15.0)
Zigbee2MQTT:info  2020-10-09 11:39:48: Starting zigbee-herdsman...
Zigbee2MQTT:debug 2020-10-09 11:39:48: Using zigbee-herdsman with settings: '{"adapter":{"concurrent":null},"backupPath":"/share/zigbee2mqtt/coordinator_backup.json","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","databasePath":"/share/zigbee2mqtt/database.db","network":{"channelList":[11],"extendedPanID":[221,221,221,221,221,221,221,221],"networkKey":"HIDDEN","panID":6754},"serialPort":{"path":"/dev/ttyACM0"}}'
Zigbee2MQTT:error 2020-10-09 11:40:08: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-10-09 11:40:08: Failed to start zigbee
Zigbee2MQTT:error 2020-10-09 11:40:08: Exiting...
Zigbee2MQTT:error 2020-10-09 11:40:08: Error: Failed to connect to the adapter (Error: SRSP - SYS - ping after 6000ms)
    at ZStackAdapter.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:92:31)
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm
 ERR! zigbee2mqtt@1.15.0 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.15.0 start script.
```

Barjavel, 2020-10-09:
same problem, same version, Conbee2
[image]

poenneby, 2020-10-09:
@Barjavel and @oldfart101, with all due respect - your stack trace is different than mine.
@Koenkk I'm not sure this is related to the latest release.
Someone suggested in another issue that I reinitialise the database (the `APS_DUPLICATE_ENTRY` seems somehow related to DB) however I cannot seem to find that information again...

Koenkk, 2020-10-09:
@poenneby your error is indeed different (and I've neverseen it before). What adapter are you using and have you also tried reflashing it? Looks like something is corrupted in the adapter firmware.

@ others, please check https://www.zigbee2mqtt.io/information/FAQ.html#help-zigbee2mqtt-fails-to-start

poenneby, 2020-10-09:
I am using a Texas Instruments CC2531 - have not tried to reflash, seems like drastic measure since it is working after a system reboot. It's only when I restart the plugin from the supervisor in hassio that it fails.
I have around 10 devices connected, Aqara temperature sensors and power relays.
Struggling to pair Ikea tradlos devices failing to pair during interview stage.
Thanks

oldfart101, 2020-10-09:
The problem with Home Assistant (HassOS) is that the expected directory for serial devices is not created.
ls -l /dev/serial/by-id results in no such file or directory
same with ls -l /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0014D956FC-if00

poenneby, 2020-10-09:
I gave [flashing](https://www.zigbee2mqtt.io/information/flashing_the_cc2531.html) a go and this seems to have resolved the issue for me. I can now restart without fail. I am guessing the version of my firmware dating around November 2019 when I first flashed is not compatible with the later versions of zigbee-herdsman.
Thanks again.

oldfart101, 2020-10-14:
solved (for me anyway)
I had the HomeAssistant integration 'ZHA' installed as well as zigbee2mqtt
Removed Zigbee Home Automation
zigbee2mqtt now starts

gonzaechev, 2020-10-21:
Same problem with 1.15 version with CC2531

`0 info it worked if it ends with ok
1 verbose cli [ '/usr/bin/node', '/usr/bin/npm', 'start' ]
2 info using npm@6.14.8
3 info using node@v12.19.0
4 verbose run-script [ 'prestart', 'start', 'poststart' ]
5 info lifecycle zigbee2mqtt@1.15.0~prestart: zigbee2mqtt@1.15.0
6 info lifecycle zigbee2mqtt@1.15.0~start: zigbee2mqtt@1.15.0
7 verbose lifecycle zigbee2mqtt@1.15.0~start: unsafe-perm in lifecycle true
8 verbose lifecycle zigbee2mqtt@1.15.0~start: PATH: /usr/lib/node_modules/npm/node_modules/npm-lifecycle/node-gyp-bin:/srv/openhome-zigbee2mqtt/node_modules/.bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games
9 verbose lifecycle zigbee2mqtt@1.15.0~start: CWD: /srv/openhome-zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.15.0~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.15.0~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.15.0~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.15.0 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:314:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:314:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.15.0
15 verbose cwd /srv/openhome-zigbee2mqtt
16 verbose Linux 4.19.66-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.19.0
19 verbose npm  v6.14.8
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.15.0 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.15.0 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]`

If I reboot the system, the service start OK,

closed 2020-11-29 by github-actions[bot]

imrise, 2021-03-26:
> same problem, same version, Conbee2
> [image]

did u manage to solve this problem. I am having the same problem also..