Koenkk/zigbee2mqtt #4608: Failed to configure 'Friendlyname' (PIR Sensor from Samotech)

Fabrice-v-t, 2020-10-08:
Hello koenk,

i've got a problem with my PIR sensors.

When i connect them everything works.
After a few moments Zigbee2Mqtt beginns to configure the Sensor.
From that moment on the sensor stops working.

and after a few seconds this error arives:

_**Failed to configure 'Bewegung1', attempt 1 (Error: Bind 0xec1bbdfffea37d8c/1 genBasic from '0x00124b000be88ccf/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:554:17)
    at processTimers (internal/timers.js:497:7))**_

I looked in the devices.js and found these Sensors.
So they are supported or not?
Do you know how i can solve this problem?

Greetings Fabrice

Koenkk, 2020-10-09:
Try to force configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and make sure to wakeup the device right before doing this. Check the log if is executed succesfully.

Fabrice-v-t, 2020-10-09:
Hello and thanks for your awnser,
I forgot to say that i tried this allredy.
I pressed the reset Button Just Shirt top wake Up (the devices Woke Up and sendt also Data)
I startest configure and it ended in the Error above.

Fabrice-v-t, 2020-10-09:
Is there a way to Stop the Auto configure because the device works....

Koenkk, 2020-10-09:
Pushed a possible fix, after this try https://github.com/Koenkk/zigbee2mqtt/issues/4608#issuecomment-706383735 again.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Fabrice-v-t, 2020-10-10:
Ok so i will Change to Developer Version and after?

Fabrice-v-t, 2020-10-10:
So i changed the devices.js and tried it again but the error occured again.
On wich way i can show you the whole console output of this try?

Koenkk, 2020-10-11:
You can post it on pastebin.com and link it here.

Fabrice-v-t, 2020-10-11:
So i tried again,
and the error occured again.

Most of the time the error occurs that often that Zigbee2Mqtt stucks.

[https://pastebin.com/zLdvSREv](url)

Koenkk, 2020-10-12:
While configure was executed did you kept pressing a button on this device (to keep it awake)?

Fabrice-v-t, 2020-10-12:
some times i tried it. but in this shell output it was all by it self

Fabrice-v-t, 2020-10-12:
The problem when holding the only button the device will go in pairing mode.
On the device is just the reset button.

Koenkk, 2020-10-12:
@PedroLamas is this device working fine in your case?

PedroLamas, 2020-10-12:
As far as I can see... yes, it's working just fine!

Fabrice-v-t, 2020-10-12:
yes but after a while zigbee stops working.
and after a restart the pir sensor dosn´t react anymore

PedroLamas, 2020-10-12:
Actually, I just noticed mine was manually turned off (had it like that for a while), I will now turn it on and report back my findings.

Fabrice-v-t, 2020-10-12:
another question is there a way to turn the automatic configure off.
so that zigbee2mqtt won´t configure a special device automatic?

PedroLamas, 2020-10-12:
Mine has been working just fine for the past 6h, so I don't see any issue with it!

just a thought, but the last time I saw those errors mentioning "AREQ" stuff, I ended up reflashing the firmware of my zigbee adapter (LAUNCHXL-CC26X2R1) to fix it...

Fabrice-v-t, 2020-10-13:
Yes i have the same.
so i repaird  both sensors a few times.
at the moment everything works and after two or three errors it stops configuring.
thanks for your help guys.
I will close these isue and if i get this problem again i will reopen.

closed 2020-10-13 by Fabrice-v-t