Koenkk/zigbee2mqtt #4609: IKEA zigbee dimmer produces to much activity

Spirituss, 2020-10-08:
## What happened
I connected IKEA zigbee dimmer (https://www.zigbee2mqtt.io/devices/ICTC-G-1.html) to HA via zigbee2mqtt integration. It adds brightness and action sensors (via MQTT). In case I rotate the dimmer it starts to send a series of sensor changes and just after a couple of seconds HA freezes for a time period (apparently, until all events queue is not processed by HA). In case one rotate the dimmer for some seconds more, it hangs up HA totally and docker container needed to be relaunched.
The initial issue was described here https://github.com/home-assistant/core/issues/40728

## What did you expect to happen
Single (final) sensor value update after dimmer rotation.

## How to reproduce it (minimal and precise)
See above.

## Debug info
Zigbee2MQTT version:
Adapter hardware: CC2538
Adapter firmware version: 1.15.0

configuration.yaml:
```
homeassistant: true
permit_join: true
devices: devices.yaml
zigbee_shepherd_devices: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://192.168.0.2'
  user: mqtt
  password: mqtt
  include_device_information: true
  keepalive: 60
serial:
  port: /dev/ttyACM0
  disable_led: true
advanced:
  log_level: debug
  log_rotation: true
  pan_id: ...
  channel: 15
  cache_state: true
  network_key:
    - ...
  availability_blacklist: []
  availability_timeout: 0
  baudrate: 115200
  rtscts: true
  last_seen: epoch
  homeassistant_discovery_topic: homeassistant
  homeassistant_status_topic: hass/status
  homeassistant_legacy_triggers: true
  timestamp_format: 'YYYY-MM-DD HH:mm:ss'
ban: []
whitelist: []
queue: {}
device_options:
  retain: true
```

Log file:
[home-assistant_core_issues_40728_20200930.log](https://github.com/Koenkk/zigbee2mqtt/files/5351273/home-assistant_core_issues_40728_20200930.log)

Koenkk, 2020-10-09:
- Try to set `legacy` false for this device (https://www.zigbee2mqtt.io/devices/ICTC-G-1.html#legacy-integration) 
- If that doesn't work provide the debug logging of Zigbee2MQTT.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Spirituss, 2020-10-09:
@Koenkk 
I used ```legacy: false``` but this doesn't work. See the debug log:
[ikea_dimmer_log.txt](https://github.com/Koenkk/zigbee2mqtt/files/5354633/ikea_dimmer_log.txt)

@danieledwardgeorgehitchcock 
I'm already using ```debounce: 0.4``` but it's not a good way how to solve the problem since it makes the handling of the dimmer very coarse - it just works more like common switch than like dimmer.

Koenkk, 2020-10-09:
Something is wrong with your config, can you share it? Including the device with `legacy: false` set.

Koenkk, 2020-10-09:
Are you sure the legacy part is causing the issue? I recommend chagning it to

```yaml
'0x000....':
  friendly_name: dimmer_cabinet
  legacy: false
```

nikitaindik, 2020-10-12:
@Koenkk Hi Koen! After setting `legacy: false` I stopped receiving brightness updates. Actions like "brightness_move_down", "brightness_move_up" do fire, but "brightness" property stays the same.

What can cause this?

Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608.zip

[configuration.yaml](https://www.dropbox.com/s/2w7g8zzdbkmoyxo/configuration.yaml?dl=1)
[debug_log.txt](https://www.dropbox.com/s/zmrijmpcnicvoii/debug_log.txt?dl=1)

Koenkk, 2020-10-13:
@nikitaindik this was  not added yet for the new integration but is now available in the dev branch under `simulated_brightness`, see https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ICTC-G-1.md#device-type-specific-configuration. (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Spirituss, 2020-10-13:
> Are you sure the legacy part is causing the issue? I recommend chagning it to
> 
> ```yaml
> '0x000....':
>   friendly_name: dimmer_cabinet
>   legacy: false
> ```

@Koenkk 
I have already switched on ```legacy: false``` in my config. What do you mean?

Koenkk, 2020-10-13:
@Spirituss try changing your config to that instead of:
```yaml
'0x000....':
  friendly_name: dimmer_cabinet
  retain: true
  debounce: 0.4
  legacy: false
  homeassistant:
    battery: null
    linkquality: null
```

Spirituss, 2020-10-13:
@Koenkk 
Are you serious? What is the difference between my config and which you propose? Do ```homeassistant``` and ```debounce``` influence it?

Koenkk, 2020-10-14:
It produces the exceptions shown in this log: https://github.com/Koenkk/zigbee2mqtt/issues/4609#issuecomment-706130947

Spirituss, 2020-10-14:
@Koenkk 
Should I comment ```debounce``` as well?

Koenkk, 2020-10-14:
Yes

Spirituss, 2020-11-14:
The issue still wait for fix

Spirituss, 2020-12-15:
The issue is not fixed but still exists.

Koenkk, 2020-12-15:
Did you already try: https://github.com/Koenkk/zigbee2mqtt/issues/4609#issuecomment-707665941 ?

Spirituss, 2020-12-15:
Yes, I tried it just after you propose it. Nothing was changed.

Koenkk, 2020-12-15:
Since the code changed in the meantime, can you again provide the debug log of the issue?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

closed 2020-12-23 by github-actions[bot]