Koenkk/zigbee2mqtt #4611: Define a max and min in opening or closing in roller blind

alcar21, 2020-10-09:
## Is your feature request related to a problem? Please describe
When I raise the Ikea Fyrtur blinds they go all the way up and when they go down the catch on the window handles.
When low it is frustrating to be aware that it does not latch anywhere.

## Describe the solution you'd like
Define a maximum or/and a minimum when you open or close the blinds fully

## Describe alternatives you've considered
Define a new parameter that allows a blind to open or close at a certain percentage.
example:
```yaml
devices:
  '0x00158d0001d82999':
    friendly_name: 'livin_room_cover'
    max_percent: 90
    min_percent: 20
```

## Additional context
Does not apply

alcar21, 2020-11-09:
it can be implemented?

Koenkk, 2020-11-09:
It can if there is enough interest for it. I currently do not see any thumbs up on the OP, if someone else is interested please do this.

alcar21, 2020-11-09:
I understand. Thank you @Koenkk

edubox, 2020-11-09:
I think is an interesting feature

Jaguaza, 2020-11-09:
interesting

Davide1313, 2020-11-09:
I think it's a good feature, it could be very useful.

Jaguaza, 2020-11-12:
Please @Koenkk I've been behind this thread for a long time and I'm very interested. Please implant it, it would make us happy and it would be good news.

Sure there are non-participatory people who would see this integration great.
If I knew about programming it would help to do it, but I'm a chump.

I can only ask and trust that you can help us to do this.

Thanks for everything.

hoggerz, 2020-12-13:
Hi,

The IKEA Fyrtur blinds do have a configurable maximum drop that can be set using the buttons on the blind itself. I have one and it works fine with zigbee2mqtt, here’s the instructions from the manual:

Set maximum level of extension

Move the blinds to the desired position using the Open/Close remote or the buttons on blind. When blind is at the desired position, you can double press the up or down button on the blind to save this position as the new maximum level of extension.
If you would like to clear the maximum level of extension setting, first move the blind to the top position. Afterwards, double press either up or down button on the blind .
You cannot perform this setting using the Open/Close remote.

I know this doesn't solve the minimum opening issue but it’s a start.

hoggerz

alcar21, 2020-12-13:
I didn't know this. It's valid for me but i don't know if it is valid for the rest people.

mambers, 2020-12-22:
It would be interesting for me if this function could be implemented, without it it is much more limited by creating automation for this blind.

Koenkk, 2020-12-22:
I will implement this once I'm finished with current zigbee-herdsman-converters refactoring (almost done)

Koenkk, 2020-12-27:
@alcar21 reading https://github.com/Koenkk/zigbee2mqtt/issues/4611#issuecomment-744065796 now, does this fix your issue? (this sounds better than limiting the position from the z2m side).

Koenkk, 2020-12-27:
I've started to implement this but I don't think this can be (correctly) implemented. The problem is that when you send state: OPEN, a Zigbee command "OPEN" will be send to the cover (so there is no position to min/max out here). I think that in fact the correct way to limit the max/min position is in the blind itself (like described in: https://github.com/Koenkk/zigbee2mqtt/issues/4611#issuecomment-744065796).

mambers, 2020-12-27:
@Koenkk There is no method to natively put the minimum, you can only the maximum.

Koenkk, 2020-12-27:
@mambers I see, but neither can we limit the min position from z2m

alcar21, 2020-12-27:
> I've started to implement this but I don't think this can be (correctly) implemented. The problem is that when you send state: OPEN, a Zigbee command "OPEN" will be send to the cover (so there is no position to min/max out here). I think that in fact the correct way to limit the max/min position is in the blind itself (like described in: [#4611 (comment)](https://github.com/Koenkk/zigbee2mqtt/issues/4611#issuecomment-744065796)).

The native method not working with zigbee2mqtt. 

Ok ok, i understand the problem.

alcar21, 2020-12-28:
> @mambers I see, but neither can we limit the min position from z2m

@Koenkk this feature is useful because when I said to Google or Alexa, "open the cover or close the cover" you can check the command and set max or min position. I believe that if you set position above or below the configured limit (min and max), it's difficult to control this case.

Now, when I push the remote button, I control max with home assistant. This is my action:
```
 action:
    - service: cover.set_cover_position
      data:
        entity_id: cover.cover_salon_derecha_cover
        position: 90
```

Koenkk, 2020-12-28:
@alcar21 that would involve translating commands, e.g. when OPEN command is set use position. In my opinion this look like a big workaround/hack.

alcar21, 2020-12-28:
yes, it's botched. I understand because I'm a Java programmer.

I think that comment #4611 only works with the button [Ikea E1766](https://www.zigbee2mqtt.io/devices/E1766.htm)

@mambers do you have home assistant? Do the Ikea limits of comment #4611 work for you?

Thanks.

mambers, 2020-12-28:
Once added to Z2M, the default maximum limits that ikea has have disappeared, there is no way to configure them

hoggerz, 2020-12-28:
I found that I was able to set the hard limits following the instructions posted above. Mine is integrated with zigbee2mqtt & HA and it works fine

alcar21, 2020-12-28:
> I found that I was able to set the hard limits following the instructions posted above. Mine is integrated with zigbee2mqtt & HA and it works fine, it’s an IKEA blind.

I try later again because now, I'm not at home. When I do the instructions of comment #461. when I made the comment instructions I couldn't define a limit.

mambers, 2020-12-30:
I have managed to set the maximum limit by performing the steps of [#4611](https://github.com/Koenkk/zigbee2mqtt/issues/4611#issuecomment-744065796), now saying to ok google "lower bedroom blind" I don't have to say the percentage to which to go down. only missing some method for the minimum amount.

alcar21, 2021-01-02:
Hi, I check comment 4661 and only working in the bottom limit. For the top not working, go up to the top.

alcar21, 2021-01-10:
@Koenkk it is possible to send the configuration to home assistant such that:
```
cover:
   - platform: "mqtt"
     name: livin_room_cover
     availability_topic: "zigbee2mqtt/bridge/state"
     command_topic: "zigbee2mqtt/livin_room_cover/set"
     position_topic: "zigbee2mqtt/livin_room_cover"
     set_position_topic: "zigbee2mqtt/livin_room_cover/set"
     set_position_template: "{ \"position\": {{ position }} }"
     value_template: "{{ value_json.position }}"
     position_closed: 20
     position_open: 90
```

Koenkk, 2021-01-10:
@alcar21 yes, add the following for your device in `configuration.yaml`.

```yaml
```yaml
devices:
  "0x12345678":
    friendly_name: my_cover
    homeassistant:
      cover:
        position_closed: 20
        position_open: 90
```

alcar21, 2021-01-10:
Perfect, this resolve my problem.
Thanks!

closed 2021-01-10 by alcar21

night199uk, 2021-04-07:
Hey. I'm wondering if the commit in 5cf63da reference here can be pushed to zigbee-herdsman-converters?

I have a non-Ikea blind - a Yoolax / Graywind blind from Amazon powered by the Yookee D10110. Although this blind allows you to set "min" and "max" points, they are not actually correct and act more like "stops" than limits. e.g. issuing a location: 0 will still set the blind fully open and location: 100 set the blind to full length.

I use HomeKIt via homebridge-z2m as my master, not home assistant; homebridge-z2m *always* issues location: <x> settings, and never simply issues open/close commands. Therefore; for this scenario (where the blind can't directly receive open/close) I think the commit in 5cf63da works perfectly and seems to be the only reasonable way to fix this right now. I'm running with that commit applied locally.

alcar21, 2021-04-07:
Finally, if it's usefully, i used the following code:

```
'0x000d6ffffe2ab1e9':
  friendly_name: cover_salon_derecha
  homeassistant:
    cover:
      payload_open: '{ "position": "95" }'
      set_position_template: '{ "position": {{ 95 if position > 95 else position }} }'
      position_open: 95
      position_closed: 0
```

dusopn, 2021-05-16:
> @alcar21 yes, add the following for your device in `configuration.yaml`.
> 
> ```yaml
> devices:
>   "0x12345678":
>     friendly_name: my_cover
>     homeassistant:
>       cover:
>         position_closed: 20
>         position_open: 90
> ```

Do I understand this correctly that this will set bottom but also the top limit of my fyrtur? also for use with Google home "open blinds"? just pasting this code into config file? thanks for info

dusopn, 2021-05-17:
Configuration invalid

Component error: devices - Integration 'devices' not found.

dusopn, 2021-05-17:
Also, do I have to set up anything else (templating) or just paste that code to config. So this is now added functionality of hassio? thank you

sepodele, 2022-10-01:
`'0xasdasdasdaqwrqrwqr0':
    friendly_name: Cover
    homeassistant:
      cover:
        position_closed: 0
        position_open: 85 `

I've added that to Zigbee2mqtt configuration.yaml (and restarted mqtt addon, zigbee2mqtt addon and Home Assistant) but blinds still open all the way. What am I doing wrong?

alcar21, 2022-10-03:
> ```
> '0xasdasdasdaqwrqrwqr0':
>     friendly_name: Cover
>     homeassistant:
>       cover:
>         position_closed: 0
>         position_open: 85
> ```
>
> I've added that to Zigbee2mqtt configuration.yaml [...]

You need to change your code by:
```
'0xasdasdasdaqwrqrwqr0':
    friendly_name: Cover
    homeassistant:
      cover:
        position_closed: 0
        position_open: 85 
        payload_open: '{ "position": "85" }'
        set_position_template: '{ "position": {{ 85 if position > 85 else position }} }'
```

kirillgroshkov, 2022-11-22:
> If you would like to clear the maximum level of extension setting, first move the blind to the top position. Afterwards, double press either up or down button on the blind .

Thank you, I was searching for this instruction for a while! 🙏

sepodele, 2023-04-25:
With this in configuration.yaml I can't stop the cover with 'Open' while cover is opening. Can anyone else confirm? Opened an issue here but not maybe it's not the right repo: https://github.com/zigbee2mqtt/hassio-zigbee2mqtt/issues/477

tjuanma, 2023-08-11:
> 

Amazing, works like a charm. 10 puntazos!

pklebba, 2023-11-09:
I have one issue to resolve.

I think the most stable solution is to configure directly on closures like texted in [this comment](https://github.com/Koenkk/zigbee2mqtt/issues/4611#issuecomment-744065796), so that's why I used it. 
The major reason is the fact the position is not handled by Z2M/HA settings, which happens only in cases when opening/closing are made by it in the middle. So the advantages are:
- My remote(s) can work correctly with just native Zigbee's binding (much faster response),
- Buttons directly on closure work without any problems,

So that had worked until yesterday when I took off the battery to charge it, after that the config for maximum position just disappeared (I thought it was saved on non-volatile memory...). So yes, I can just configure it again (and keep in mind that in the future connect the charger without putting off the batteries from closures), but I wonder if it can be solved in a better way.
The best solution is to set position directly on closures programmatically. Official IKEA gateway supports it without any issues so It probably can be solved automatically via the ZigBee protocol. Has someone looked at that? Or maybe where to start the research?

tjuanma, 2023-12-14:
> You need to change your code by:
>
> ```
> '0xasdasdasdaqwrqrwqr0':
>     friendly_name: Cover
>     homeassistant:
>       cover:
>         position_closed: 0
>         position_open: 85
>         payload_open: '{ [...]

Hi,

i'm using your set_position_template and from last update when I try to open my covers i receive the following error:

`Publish 'set' 'position' to 'EstorIzquierda' failed: 'Error: 'position' is not a number, got string (84)'`

Happnds for both covers "EstorIzquierda" and "EstorDerecha"

My config is:

`  '0x84fd27fffed09516':
    friendly_name: EstorDerecha
    homeassistant:
      cover:
        position_closed: 0
        position_open: 85
        payload_open: '{ "position": "84" }'
        set_position_template: '{ "position": {{ 84 if position > 84 else position }} }'
  '0xcc86ecfffe1610e8':
    friendly_name: EstorIzquierda
    homeassistant:
      cover:
        position_closed: 0
        position_open: 84
        payload_open: '{ "position": "84" }'
        set_position_template: '{ "position": {{ 84 if position > 84 else position }} }''`

Any ideas?

sepodele, 2023-12-14:
Yeah, you just need to change "84" to 84.

tjuanma, 2023-12-14:
> Yeah, you just need to change "84" to 84.

Works again!!

Thank you so much!

lezti2001, 2025-03-28:
For all of people like me, I wanted two differents position of my blind, that it's the code works for me

  '0x000d6ffffe9a7cb1':
    friendly_name: Store fenêtre fixe1
    icon: device_icons/4d6481ea9b3564c9a9a9840ca8311676.png
    homeassistant:
      cover:
        state_open: open
        state_closed: closed
        position_open: 99
        payload_open: '{"position":99}'
        position_closed: 13
        payload_close: '{ "position": 13 }'
        set_position_template: '{ "position"}} }'