Koenkk/zigbee2mqtt #4612: Neo NAS-AB02B0 Alarm

arduino43, 2020-10-09:
Did you read the FAQ?
Yes

Zigbee2MQTT fails to start?
No

Having issues when using a CC2531?
No

Unsupported device?
No

Device does not pair or interview fails?
No

Bug report?

1.) Setting humidity_max sets the value for humidity_min as the values are reversed.

2.) Setting either humidity_alarm and temperature_alarm to false (default is true) does not appear to work. The device responds but does not update the values.

3.) Not sure if this is an issue or just the functionality, but communications have a delay of 3-5 seconds.

## What happened

## What did you expect to happen

## How to reproduce it (minimal and precise)

## Debug info
Zigbee2MQTT version:
CC26X2R1 / v 1.15.0
Adapter firmware version:
3

Koenkk, 2020-10-09:
@kirovilya do you know about this?

kirovilya, 2020-10-09:
check on the weekend

kirovilya, 2020-10-12:
yes. it is a bug :)
621 - humidity_min
622 - humidity_max

closed 2020-11-19 by github-actions[bot]