Koenkk/zigbee2mqtt #4613: Centralite 3400 Keypad

niekveldhuizen, 2020-10-09:
##  What happened

I Orderd a Centralite 3400 Keypad, after some investigation I saw that the supported device is named 3400-D. I don't know that is a compleet differten model? 
I could pair the device but now i get the message "unsupported device with Zigbeee model  '3400' 

## What did you expect to happen
I did expect it to pair and be supported

## How to reproduce it (minimal and precise)
Try to pair the device again. 

## Debug info
Zigbee2MQTT version: V1.15.0
Adapter hardware: CC2531
Adapter firmware version:

Koenkk, 2020-10-09:
Try adding `3400` here https://github.com/Koenkk/zigbee-herdsman-converters/blob/144f32abcb3a1f6083afb3262dc468f94d6cb965/devices.js#L7795 and check if it works.

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

niekveldhuizen, 2020-10-10:
Hi,

Made the modification to the devices.js. Now it is getting paired, but no new devices appear. When i try the arm button on the remote i do see it getting published. But nothing happens.

Here is some of the logging of joining and publish:
https://pastebin.com/McucQ5GD

Koenkk, 2020-10-10:
Check https://www.zigbee2mqtt.io/devices/3400-D.html on how to use this device.

niekveldhuizen, 2020-10-10:
Hi,

I don't really know how to get domoticz to arm the alarm. When i type the pin and the arm symbol the keypad is making a sound en post to zigbee2mqtt. But in domoticz nothing is happening.  

And temperature should be displayed a device?

Koenkk, 2020-10-10:
Probably the device is not in the domitcz z2m plugin yet, you can open an issue for that here. 

I will apply the change of https://github.com/Koenkk/zigbee2mqtt/issues/4613#issuecomment-706385871 so that the device is supported out-of-the-box by Zigbee2MQTT

niekveldhuizen, 2020-10-10:
Does it need te be a different topic? 

Who can build the support? You would think that it should not bee that different from the 3400-d.

Koenkk, 2020-10-10:
@niekveldhuizen I think the 3400-D is also not supported there. It's maintained by someone else and in a different repository so a new issue has to be created.

vanlooverenkoen, 2020-11-12:
I have bought the 3400 as well. But there is no pairing documentation

niekveldhuizen, 2020-11-12:
Step 2: Trigger Manual Rejoin
Remove the keypad from the wall and press the wall
tamper switch 5 times.

See documentation [https://www.centralite.com/downloads/DataSheet-3SeriesSecurityKeypad.pdf](url)

vanlooverenkoen, 2020-11-12:
Thanks, maybe this should be added to the documentation as well + add the 3400 to the documentation as well

niekveldhuizen, 2020-11-12:
Can home assistant work with the 3400 as a device? Domoticz can not..

vanlooverenkoen, 2020-11-12:
My 3400 is still underway. But I wanted to prepare already so I can start pairing when I receive the 3400. I found it second hand for 10 euro and I wanted to give it a shot

kevincaradant, 2020-11-20:
Hi

I bought a 3400-D and I'm currently trying to use it via Home Assistant.

For the moment I can see the payload in Z2M. And I don't know if you find an other way to use it in HA but I the only solution I found to get the action_code is to use some mqtt trigger like that:

```
 - platform: mqtt
      topic: zigbee2mqtt/centralite
      payload: 'disarm'
```

It's weird, I never use devices like that so I don"t know why I don't find an other way to do it.

And if you find a logic way to switch between each mode in the top of the Centralite between a home logo , a person which is running and a logo as a alarm, tell me because I switch between state without erally understand how I do that :P

kevincaradant, 2020-11-22:
Hi again, 

If someone can told me how can we manage the green and red state stick on the Centralite 3400 device. With me, it's always green.

Thank you in advance

Regards

Dinth, 2020-11-24:
> Step 2: Trigger Manual Rejoin
> Remove the keypad from the wall and press the wall
> tamper switch 5 times.

Ive got 3400-X and this method doesn't work for me, but the factory default one works.
1. Remove batteries (both!) 
2. Press the tamper button on the back and keep pressing it.
3. Insert one (!) battery 
4. After keypad is lit up, immediately release the tamper button (doesn't seem to work its kept pressed few more seconds) 
5. I recommend doing that with just one battery, as those batteries don't slid in and out easily and the device works with one as fine as with two, therefore inserting two batteries during setup messes up timing in point 4.

kevincaradant, 2020-11-24:
> Ive got 3400-X and this method doesn't work for me, but the factory default one works.
>
> 1. Remove batteries (both!)
> 2. Press the tamper button on the back and keep pressing it.
> 3. Insert one (!) [...]

Indeed I used it for 3400-D and that's the only way to pair it Ith success.

kevincaradant, 2020-11-24:
BTW someone understand the manual? Sorry if it's a dumb question, but they are talking about state after to have write a good pincode or badpincode.

But how the centralite know that's a good pincode or bad to change his own state ? 

In the aim to have green or red light on the Centralite. 
[image]

Thank you

Dinth, 2020-11-24:
Im not sure if my 3400-X has this functionality. It does have green led on the top and bottom, but so far I've "kind of" managed to change the status between armed and disarmed, but never seen a colour other than green. 
Would you be able to scan your manual? I have found two different manuals online, but none of them is so length as yours. 

Here's my current status:
* I can start different types of arm from keypad and Zigbee2MQTT sees the change. Haven't figured out how to disarm the keypad from the keypad. 
* Every time i arm/change arm type from keypad, it gives me a long beep and disconnects from Zigbee. It connects back if i press the tamper button once. 
* Every time i arm/change arm type from keypad, the top three buttons start blinking. I don't know what it means. 
* Posting a MQTT message to arm/disarm the device from Z2M doesnt seem to work at all.

kevincaradant, 2020-11-25:
Hi

This is my manual.

[image]
[image]

And to answer at yours questions ( as I can because myself I don't undertsand everything :/

> Battery level is not reported to MQTT, but Voltage is and funnily it is always 2900V regardless if one or two batteries are inserted
  Exactly the same thing as you. I'm starting to think 2900 is not the real value because since the begin I have this value :)

> Tampering is not reported to MQTT at all.
   Same or I don't know how to use it but if it's just to press it; in this case I know to do it :)

> Posting a MQTT message to arm/disarm the device from Z2M doesnt seem to work at all.
   Same or I don't know to do it correctly because I don't use often MQTT explorer to do it. But I didn't successfully send to the Centralite a message.

> Every time i arm/change arm type from keypad, the top three buttons start blinking. I don't know what it means.
You too, you had exactly the same problem than me. I think it's because we entered a 'wrong' pincode. The Centralite doesn't receive from HA or Other Domotics App the action code which means that the entered code is correct. But it's just a idea like that, not sure it's for this reason that the 3 buttons starts blinking.

> In fact, keypad doesn't verify code at all. Can all this weird behaviour be because the keypad expects a reply from the network?
I think it's normal. In my case HA receive the pincode and that should be HA which send a mqtt message to centralite as 'yeah man it's good, the pincode is valid'. But the Centralite plays the role of the "dumb component" and he send data and wait to receive data. But I didn't successful to do it :(

> Every time i arm/change arm type from keypad, it gives me a long beep and disconnects from Zigbee. Sometimes it connects back, but usually i need to press the tamper button to reconnect it.
I didn't see this behavior. I think mine is always connected to Z2M. It's the only difference between you and me for the moment ;)

I can start different types of arm from keypad and Zigbee2MQTT sees the change. Haven't figured out how to disarm the keypad from the keypad.
I think I click on the first button (mine is a home picture) yours could be a person stand up. After that I write '123456' and I saw the action 'disarm' in Z2M ;)

I will edit this message more later, I don"t have time to be more explicit :)

Dinth, 2020-11-25:
Thanks for all your comments. 
First of all, i was surprised to see your button layout, as it means that there are actually three different models of 3400, not two as i thought until now.

kevincaradant, 2020-11-28:
@Koenkk , I think when you will have some times, we will need some help with this device to understand what should we do to help you to debbug it if it's necessary :)

Valliess, 2020-11-29:
Yup we start to see the logic from these types of devices, similar to the Iris iL02_1 (probably all the same chinese manufacturers but other brands). 
The device sends a message to the network when pressing ARM (arm away), PARTIAL (arm stay) or OFF (disarm). Every time the device awaits an answer to confirm it's feedback status. Each message needs to be answered with the same message ID. For example you press ON, the device sends something like "arm_all_zones", "zoneID =99", "messageID=37". The alarm central (HA or Loxone or other) needs to interpret this message and send back a confirmation stating that the demand "arm_all_zones" in zoneID"99" with messageID=37 was confirmed. (this is done to avoid false interpretation when someone pushes multiple buttons) 
My syntax and/or clustermentionings and their types are not at all necessarily correct but that's the idea. _This makes it almost unusable for loxone to use in the case of the Z2M plugin for loxberry because Loxone is unable to interpret a message and reply it, recopying certain parts of the received (UDP) message_. 
In case you want to disarm, the ciphercode that you need to type before pushing OFF, is also sent and needs to be replied for as well to confirm the alarmcentrall OFF feedback. Same message structure, but the ciphercode is integrated in the message and needs to be answered, as well as the demand for the "disarm" in zone 99 and the messageID 37. 
The CentraLite seem to be Gen 2 devices and the Iris are Gen 3 devices. 
As I've read, the biggest difference between these two is that the centralite needs the ciphercode before pushing the arm button and before pushing the disarm button as well, the Iris only needs to get the ciphercode when disarming.
So we could very probably say that the CentraLites are similar to the Iris Keypads. But a difference needs to be made in the ARM type interpretation. 
See https://github.com/Koenkk/zigbee2mqtt/issues/4673 for more info

Dinth, 2020-12-04:
I have observed that whenever my keypad disconnects from Zigbee, Zigbee2MQTT shows this in the log:
```
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"Landing keypad","ieee_address":"0x000d6f00052dd68f"},"type":"device_announce"}'
INFO: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"Landing keypad"},"type":"device_announced"}'
```

kevincaradant, 2020-12-12:
Who did a try with this centralite? :)

Someone have successfully did work this device? In HA ect?

Until today, I didn't enought time to check this device but I'm really not sure 3400-D works very well even if the documentation says that this device is supported. Or at least, it does with some issues.

kevincaradant, 2020-12-12:
> Mind you there at at least three different models of 3400, while almost identical from outside, they may behave in a different way.
> Im currently waiting for another keypad, to confirn if the issues im having are not a faulty unit

If we are two (me and you) to have exactly the same issues. Excepted the disconnection, I think for the rest, it will not a faulty issue ;)

kevincaradant, 2020-12-12:
When I click on the first icon and then I type a 6 digits code. I got this log:

```
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:46:44: MQTT publish: topic 'zigbee2mqtt/centralite', payload '{"action":"","linkquality":81,"temperature":19.99,"voltage":2900}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:46:44: MQTT publish: topic 'zigbee2mqtt/centralite/action', payload 'disarm'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:46:47: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centralite","ieee_address":"0x000d6f000d8a5646"},"type":"device_announce"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:46:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"centralite"},"type":"device_announced"}'
```

After to click on the third button and type a 6digits code nothing happens.

After to click on the 2sd icon and then type 6 digits code, this log happens:
```
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:49:22: MQTT publish: topic 'zigbee2mqtt/centralite', payload '{"action":"","linkquality":78,"temperature":19.99,"voltage":2900}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:49:22: MQTT publish: topic 'homeassistant/device_automation/0x000d6f000d8a5646/action_arm_all_zones/config', payload '{"automation_type":"trigger","device":{"identifiers":["zigbee2mqtt_0x000d6f000d8a5646"],"manufacturer":"CentraLite","model":"3-Series security keypad (3400-D)","name":"centralite","sw_version":"Zigbee2MQTT 1.16.2"},"payload":"arm_all_zones","subtype":"arm_all_zones","topic":"zigbee2mqtt/centralite/action","type":"action"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:49:22: MQTT publish: topic 'zigbee2mqtt/centralite/action', payload 'arm_all_zones'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:49:26: MQTT publish: topic 'zigbee2mqtt/bridge/event', payload '{"data":{"friendly_name":"centralite","ieee_address":"0x000d6f000d8a5646"},"type":"device_announce"}'
zigbee2mqtt    | Zigbee2MQTT:info  2020-12-12 21:49:26: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"announce","meta":{"friendly_name":"centralite"},"type":"device_announced"}'
```

I need 2 things. 

1- Link the automation (alarm) in HA with the answer given by the centralite
2- Didn't understand how yet (from HA send a MQTT message at Centrlaite to say it's the good pin or even if in HA the state of the alarm change, it should change on Centralite). I didn't remember where I saw something like that, but I think I should return `action_code: 99` something like that if pincode is good. But I need to find again this text because I didn't remimber how it works exactly.

Dinth, 2020-12-17:
OK i have managed to source a Centralite 3400-D model and unfortunately im seeing a similar problem as with 3400, where the device disconnects from Zigbee after entering a code. 
The difference is:
3400: uses 4 digit pin. Entering any 4 digits at any time causes a long beep and disconnection
3400-D: uses 6 digits pin. Entering 6 digit causes a beep and device stays on the network, but if i press an alarm button and then any 12 digits it also causes a long beep and disconnection.

JanHACS, 2020-12-25:
I would like to buy one of this keypad, but on this state of development I think it is not a good idea :-(

Dinth, 2020-12-25:
Well to be honest we cannot expect the dev to own and implement all 3400 variants on his own. The support is already there or at least most of it, we just need someone to figure it out how to use it. 
My disconnections reportedly dont happen to other users - maybe that's because im using Conbee II.

kevincaradant, 2020-12-26:
Well I will discuss as soon as possible with @Koenkk about this. At least for the 3400-D which is supported normally. Need to fix some others issues on my side before... but when he will have some times for that and me also, it will be with pleasure if I can help to debug it or at least understand how is it working.

I bought it, its not for the decoration. If someone has something useful to share, don't hesitate :P

Dinth, 2020-12-26:
@kevincaradant which ZigBee stick are you using?

kevincaradant, 2020-12-26:
@Dinth Cc1352p-2 board

faceoff34, 2021-01-14:
@Dinth I have a 3400 and can confirm the same disconnect problem after entering a 4 digit code.  My adapter is the CC26X2R1.

niekveldhuizen, 2021-01-14:
@Dinth I have a  CC2531 and also can corfirm the disconnect.

Dinth, 2021-01-14:
Thanks for confirming! So we know its not related to Deconz (for a moment i thought that it may be related to slow network response issue on Deconz adapters).
Do you guys tried sending a response to the keypad with aa confirmation that the code is wrong/correct immediately after receiving the the code from the device?

faceoff34, 2021-01-14:
I am relatively new to all this so I apologize if I am going about this wrong or using the wrong terminology.  I have tried publishing a response and can't seem to find anything useful.  Could it have something to do with the response matching the transaction number ("action_transaction") of the initial keypress?  One thing I noticed is when the device rejoins the network, the three top keys are flashing red sequentially.  If I publish the "arm_mode" to "disarm", next time the motion sensor is activated, the top 3 keys are solid white.

kevincaradant, 2021-01-18:
Hi guys, 

Just to give you some news, yesterday I tried during hours to understand how all of it work. 

I have the good news to announce that I succeded to have a work setup between the centralite and HA (integrated alarm) :)

The only thing which doesn't work, is the little button on the back of the centralite which should tell us if the centralite was unmount or not from the wall. But when I trigger it (the little button), nothing happens on Z2M so apparently it's disabled. Maybe should I talk about it with @Koenkk :)

For people like me who begin with MQTT publish. This is an example with HA.

The documentation is well documented about centralite once you understand the MQTT request.
Just a mistake on the documentation of the 3400-D on Z2M, there is no 'emergency' mode as I could read it somewhere on this page.

```                                                                                             
- alias: "Alarm From Centralite To HA Alarm Disarm"
  trigger:
    - platform: mqtt
      topic: zigbee2mqtt/centralite
  condition:
    - condition: template
      value_template: "{{ 'disarm' == trigger.payload_json.action }}"
    - condition: or
      conditions:
       - condition: state
         entity_id: alarm_control_panel.alarm
         state: armed_home
       - condition: state
         entity_id: alarm_control_panel.alarm
         state: armed_away
       - condition: state
         entity_id: alarm_control_panel.alarm
         state: triggered
       - condition: state
         entity_id: alarm_control_panel.alarm
         state: arming
  action:
    service: alarm_control_panel.alarm_disarm
    data_template:
      entity_id: alarm_control_panel.alarm
      code: "{{ trigger.payload_json.action_code }}"

```

```
- alias: "Alarm From _ To Disarmed"
  trigger:
    - platform: state
      entity_id: alarm_control_panel.alarm
      to: "disarmed"
  action:
    - service: notify.notify_events
      data:
        message: "Alarme désactivée."
        data:
          level: 'warning'
          priority: 'high'
    - service: mqtt.publish
      data:
        topic: zigbee2mqtt/centralite/set
        payload: "{\"arm_mode\":{\"transaction\":99,\"mode\":\"disarm\"}}"
 ```
 
 Enjoy this code which I hope will help you to create all your automation expected using the integrated alarm of HA ;)

faceoff34, 2021-01-19:
@kevincaradant Nice work!  This is very helpful!  I am going to play around with it in my setup.

Dinth, 2021-01-23:
Hmm, i did this (using two keypads): 
```
automation:
  - alias: "Sync alarm from Centralite"
    trigger:
      - platform: mqtt
        topic: zigbee2mqtt/Upper landing keypad
      - platform: mqtt
        topic: zigbee2mqtt/Landing keypad
    action:
      - service: >
          {% if (trigger.payload_json.action == 'disarm') -%}
          alarm_control_panel.alarm_disarm
          {%- elif (trigger.payload_json.action == 'armed_home') -%}
          alarm_control_panel.alarm_arm_home
          {%- elif (trigger.payload_json.action == 'arm_all_zones') -%}
          alarm_control_panel.alarm_arm_away
          {%- endif %}
        data:
          entity_id: alarm_control_panel.ha_alarm
          code: "{{ trigger.payload_json.action_code }}"

  - alias: "Sync alarm from HA "
    trigger:
      - platform: state
        entity_id: alarm_control_panel.ha_alarm
    action:
      - service: mqtt.publish
        data_template:
          topic: zigbee2mqtt/Upper landing keypad/set
          payload: >
            {% if (is_state('alarm_control_panel.ha_alarm','disarmed')) -%}
            '{"arm_mode":{"transaction":99,"mode":"disarm"}}'
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_home')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_day_zones"}}'
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_night')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_night_zones"}}'            
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_away')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_all_zones"}}'
            {%- endif %}
      - service: mqtt.publish
        data_template:
          topic: zigbee2mqtt/Landing keypad/set
          payload: >
            {% if (is_state('alarm_control_panel.ha_alarm','disarmed')) -%}
            '{"arm_mode":{"transaction":99,"mode":"disarm"}}'
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_home')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_day_zones"}}'
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_night')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_night_zones"}}'
            {%- elif (is_state('alarm_control_panel.ha_alarm','armed_away')) -%}
            '{"arm_mode":{"transaction":99,"mode":"arm_all_zones"}}'
            {%- endif %}

```
Unfortunately, it still doesnt seem to work - when im trying to change the arming on my keypad im getting a wrong code (not sure about this?) beep and when im trying to change the arming in HA, it does sent a JSON to keypad, keypad accepts it... and immediatelly changes "action" to "" (empty):
[image]

kevincaradant, 2021-01-23:
About the beep, the centralite can't know it self if it's a bad pincode or not. You have to tell him. 
If you wrote something on keypad but nothing happens on HA. Check if the trigger of this automation is good. Check if the service is called. Try to put directly the real pincode in your automation to be sure HA doesn't respond just because it can't read the pincode variable. 
Indeed every pincode from centralite will trigger the automation as good pincode but at least you will be sure the automation works and it's just the variable for the action code which doesn't work.

From HA to your centralite. It's easy to see if it understand, because only one of the 3 buttons in the top will be red (sticky not blinky).
If the button wanted is not red, it could be because the automation didn't run at all. Bad trigger? Bad condition? Or just your huge if / else in the action section. Please check your logs to see if you see your /set request in logs.
If you see it with the good value and action, I think its good :). Even if few times after you see action empty. I have the same behavior but its not an issue in my case ')

germeraad, 2021-01-27:
Hee all, 

Just got my hands on a 3400 keypad (the one which needs 4 digit codes). And started debugging with Z2M and MQTT. 

@kevincaradant, I see in your HA automation examples that you are always using "transaction":99 is this working for you all the time? Because the way I understand it the keypad generates a random transaction number every time when you type in a code which you need to use to communicate back towards the keypad. 

When you activate the different modes, do they make the buttons light up red on the keypad? When a do a /set request to the keypad I hear a long beeb, but no color changes. 

Also, are you having any errors in your log mentioning no converter available? 

> No converter available for '3400-D' with cluster 'ssIasAce'

kevincaradant, 2021-01-27:
> Hee all,
>
> Just got my hands on a 3400 keypad (the one which needs 4 digit codes). And started debugging with Z2M and MQTT.
>
> @kevincaradant, I see in your HA automation examples that you are always [...]

Hi

Yes I think it worked everytime that I used the transaction: 99.

The button light up red on the keypad as expected. 

And yes I think I saw sometimes, some errors in debug mode only?  about `no converter available for... `
But it works well so I didn't look more these errors. 

The only point where I'm not sure why is that sometimes the device seems to leave and join again the Zigbee network few seconds later and become 'discovered' again for Z2M. But I don't know when this happens really. 
I don't use very often the keypad. I use more often the location with GPS which enable / disable the alarm for me ').

Dinth, 2021-01-27:
@germeraad hi. I have both models and while I the six digit one looks promising, I still struggle with four digit one, so they must differ more than just with different buttons and no of digits. Also ssalasAce error may be responsible for the tamper button or those weird disconnects

KenGrinder, 2021-01-30:
I'm also having an issue with the buttons not turning red when I arm it. All 3 will quickly flash red while it connects, but then turn to white, when using any of the 3 different arms. Everything else is working great though. Thanks!

Dinth, 2021-02-09:
So right now i have ditched my 4 digit model and only playing with my 6 digit (home, man running and a bell in top row of buttons).
Seems that this model has only two zones (disarm and arm_all_zones) what's not cool as i want my home alarm to use more zones, but lets leave that for now. 
Anyone knows what the third button (bell icon) is for? it gives me a beep after a very long press, but nothing visible on Z2M side.

kevincaradant, 2021-02-09:
> So right now i have ditched my 4 digit model and only playing with my 6 digit (home, man running and a bell in top row of buttons).
> Seems that this model has only two zones (disarm and [...]

From what I can remember. The last icon is for arm_day or night mode. Don't remember which one between these two choices.
But I don't use them (only armed and disarmed).

However I'm sure these 3 icons represents one state each time ')

sh00t2kill, 2021-02-13:
Ive got mine very nearly working, use Node Red instead of HA automations because its ... easier to debug :)

For now im just playing with the MQTT in and out, and havent got HA alarm involved yet
I can arm it, pass a response back to the keypad via MQTT and i get a "different" beep, which to me suggests no error.

However i cant get the LED status to change -- its always green suggesting not armed.

Im struggling to get it to disarm, and I assume thats because the keypad wont detect that its armed.

`{"action":"arm_day_zones","action_code":"1234","action_transaction":22,"action_zone":23,"battery":42,"last_seen":1613201896570,"linkquality":114,"temperature":25.3,"voltage":2900}`

I send this response back
`{"arm_mode": {"zone":"23", "transaction":"22", "mode": "arm_day_zones"}}`

The keypad gives a beep as soon as it receives the response, but only when i send arm_day_zones or arm_night_zones. Arm all zones gives nothing from the keypad.

misteriks, 2021-02-21:
Got all my automations setup with the centralite 3400-d

I notices in the zigbee2mqtt logs that there are a few more states that are not mentioned in the documentation. I tested them all to see what they do when sending a mqtt message like 

```
{
    "arm_mode": {
    "transaction": 99,
    "mode": "arm_all_zones" 
    }
}
```

When sending without "transaction": 99 they all do the same except for the first 4. These then don't make any sound

disarm (single beep for 1 second, status led green, all zone buttons white)
arm_day_zones (single long beep, status led red, left zone button red)
arm_night_zones (single long beep, status led red, right zone button red)
arm_all_zones (single long beep, status led red, middle zone button red)
invalid_code (continues short beep every 3 seconds, doesn't update status led or zone buttons)
exit_delay (continues short beep every 1 second, doesn't update status led or zone buttons)
entry_delay (no sound, status led orange, doesn't update zone buttons)
not_ready (no sound, status led red, doesn't update zone buttons)
in_alarm (no sound, status led red, left zone button red)
arming_stay (no sound, status led red, night zone button red)
arming_night (continues short beep every 3 seconds, status led red, middle zone button red)
arming_away (doesn't seem to do anything)

To detect a invalid alarm code I use the HA system log. Fire events needs to be enabled for this.
[https://www.home-assistant.io/integrations/system_log/#fire_event](url)

The "not_ready" is send via mqtt to stop the keypad from making any beep sound.

```
- alias: "Alarm wrong pincode"
  trigger:
    platform: event
    event_type: system_log_event
  condition:
    condition: template
    value_template: '{{ "Invalid code given for" in trigger.event.data.message[0] }}'
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: >
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "not_ready"
          }
        }
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: > 
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "invalid_code"
          }
        }
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: >
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "not_ready"
          }
        }
```

```
- alias: "Alarm keypad in Arming"
  trigger:
    - platform: state
      entity_id: alarm_control_panel.huisalarm
      from: "disarmed"
      to: "arming"
  condition: []
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: > 
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "exit_delay"
          }
        }
```
```
- alias: "Alarm keypad in arm AwayMode"
  trigger:
    - platform: state
      entity_id: alarm_control_panel.huisalarm
      from: "arming"
      to: "armed_away"
  condition: []
  action:
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: >
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "not_ready"
          }
        }
  - service: mqtt.publish
    data:
      topic: zigbee2mqtt/AlarmKeypad/set
      retain: true
      qos: 2
      payload: >
        {
          "arm_mode": {
            "transaction": "99",
            "mode": "arm_all_zones"
          }
        }
```

DennisVonDerBey, 2021-03-15:
Does someone here use the newest `dev` version of z2m with the Centralite 3400-D? I saw a commit changing the key `invalid_code` to `exit_delay`, but my keypad doesn't reliably respond to that code. Also my keypad does never change the system status LED from green to red or something. Could it be the implementation of the device is lacking most functionality?

misteriks, 2021-03-24:
Indeed something seems to be changed/broken with the last update. 

Invalid_code is no longer accepted as valid and behavior of the other actions changed and broke my automations.

Dinth, 2021-04-25:
Still doesn't work for me. Unless I'm doing the automation wrong

closed 2021-06-02 by github-actions[bot]

hvddrift, 2021-07-14:
> Ive got mine very nearly working, use Node Red instead of HA automations because its ... easier to debug :)
> If anyone wants what i think is a pretty comprehensive node red flow, please let me know!
> 
> I can arm it, pass a response back to the keypad via MQTT and i get a "different" beep, which to me suggests no error.
> 
> However i cant get the LED status to change -- its always green suggesting not armed.
> 
> Im struggling to get it to disarm, and I assume thats because the keypad wont detect that its armed.
> 
> `{"action":"arm_day_zones","action_code":"1234","action_transaction":22,"action_zone":23,"battery":42,"last_seen":1613201896570,"linkquality":114,"temperature":25.3,"voltage":2900}`
> 
> I send this response back
> `{"arm_mode": {"zone":"23", "transaction":"22", "mode": "arm_day_zones"}}`
> 
> The keypad gives a beep as soon as it receives the response, a long beep if nothing is received, and a REALLY long beep if invalid_code is received.
> 
> There does seem to be a large number of version of these.
> I think I have what is called the 3400-C, or "3 series security keypad"
> 
> EDIT:: Turns out i was wrong. I can send arm and disarm commands. Just i dont know if i am arming the alarm, or disarming the alarm as the keypad has no indication.

Hi

I would like to see your node red solution for this. Are your able to share it with me?

Regards

Hans

kevincaradant, 2021-10-01:
Hi guys.

I don't know what I broke... but since I migrated from 1.21.1 to 1.22.1. My 3400-D doesn't work anymore as expected :(
I got always "null' as value for the action_code. 
And I try to unpair and pair it again but now nothing works as expected (occupancy ect stays at OFF ) ... 
[image]
I don't know what happens but that seems broken on my side :/

Dinth, 2022-02-23:
Just one thing i have noticed today.
When i was playing with the 6 digit keypad (the one with a bell button), when i am entering the (wrong) code, in Zigbee2MQTT panel, "Exposes" tab, i can see action and action_code changing for a half of a second. But this doesnt get reflected via MQTT at all - nothing gets updated. Im still getting MQTT updates on temperature or link status but not on action. Maybe it changes back to the previous value too quickly to get reflected in MQTT? 
When im doing the same with the 4-digit keypad (the one with a moon button), the same action does get reflected in MQTT.

luisgarcia87, 2022-05-29:
To anyone having the same problem as @Dinth and myself that the Centralite 3400 would disconnect from the network and would not accept any codes until a hard reboot (removing batteries). I've successfully "unblocked" my Centralite 3400 by sending a mqtt message to the device in the "locked/blocked" state like this:

Mqtt publish to 
Topic: zigbee2mqtt/DEVICE NAME/set
payload:
`{
    "arm_mode": {
    "mode": "disarm"
    }
}`

[image]

After that my device can receive and send the action_code via mqtt even if there's no responde from the mqtt server telling the centralite if the code is right or wrong. 

My use for this device was not as an alarm system, but as a gate access device.

Now when someone is at my gate they can just input the code I decide and could easily open the door.

adospinoiu, 2023-06-13:
Hi all. I realize this item is closed, but maybe you can point me to some more resources. I used to have Xfinity Home Security ... got tired of paying for it so gave the main console back. Naturally, I'm now trying to DIY a home system with the door/window sensors (Visonic MCT-350 SMA) and keypad (Centralite 3400). I have a Raspberry Pi 3 Model B+ on which I have successfully installed Zigbee2MQTT (I followed the Linux installation "path") and bought a ZBDongle (CC2652P). I am able to pair the sensors ... it took me a while to find how to force them into pairing-mode; but I was able to get it. Zigbee2MQTT is picking up and pairing the sensors (on the front end).

For the life of me, I cannot figure out how to get the Centralite 3400 to get picked up by Zigbee2MQTT. I have tried every combination I have read of (i.e. remove batteries, hit button on back 10 times, replace batteries... no luck; hold reset button on back while placing batteries in ... no luck; etc).

I just cannot figure why the Centralite 3400 is not going into pairing-mode ... or why Zigbee2MQTT is not picking it up (just like it is the sensors). 

Any guidance would by much appreciated.