Koenkk/zigbee2mqtt #4615: publish to group not working, state not changed in hass

dgreebe, 2020-10-09:
## What happened
Existing binding IKEA E1743 to a group and the state publishing to HASS worked for a long time but its not working correctly anymore. Pushing button will not show any mqtt message to the corresponding group

## What did you expect to happen
its an existing setup and now button is not publishing any group info so in HASS the state is also not changed. Open "report: true" has set but its not reporting back to HASS

## How to reproduce it (minimal and precise)

Its worked for couple of weeks, but suddenly not

debug logs of mqtt;
debug 2020-10-09 10:14:04: Received Zigbee message from 'office_switch', type 'commandOn', cluster 'genOnOff', data '{}' from endpoint 1 with groupID 9
info  2020-10-09 10:14:04: MQTT publish: topic 'zigbee2mqtt/office_switch', payload '{"action":"on","linkquality":5,"update":{"state":"idle"},"update_available":false}'
info  2020-10-09 10:14:04: MQTT publish: topic 'zigbee2mqtt/office_switch', payload '{"action":"","linkquality":5,"update":{"state":"idle"},"update_available":false}'
info  2020-10-09 10:14:04: MQTT publish: topic 'zigbee2mqtt/office_switch/action', payload 'on'

I should expect that is publish to the group "office_light"

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: 20190619 source routing

dgreebe, 2020-10-09:
A weird part is, that another bulb down stairs, does report its state to the group so HASS knows its on... I cannot figure out what is going wrong here...

Koenkk, 2020-10-09:
What devices are in the group office_light? If tradfri, try powercycling them.

dgreebe, 2020-10-10:
> 
> 
> What devices are in the group office_light? If tradfri, try powercycling them.

Hi Koen,

Yes, that did the trick but how does that solve the issue ? I also tried to bind a GU10 bulb of ikea because I want to figure out of my signal was not strong enough or if I hit some kind of boundery, but that joined and paired with the group without issues or powercycles...

Is this an issue in the bulbs firmware or can you do something about it ? I cannot remember I had this issue before and I have a bunch of ikea bulbs.

dgreebe, 2020-10-10:
One more thing; I keep having issues with [these bulbs](https://www.zigbee2mqtt.io/devices/81809_81813.html). Also these are in a  group and [dimmer](https://www.zigbee2mqtt.io/devices/E1743.html) has a bind to the group but that state is again not published somehow. powercycle didn't help. Is there some fix for that ?

Koenkk, 2020-10-10:
IKEA bulbs have to be reconfigured reporting everytime they are powered off, if this fails you won't get state updates.
For the Aduro smart, I'm not sure if these support reporting at all, you can check the log.

dgreebe, 2020-10-10:
> For the Aduro smart, I'm not sure if these support reporting at all, you can check the log.
How can I see that ?

When by Z2M restarts, I see this is my logs;
error 2020-10-09 17:08:56: Failed to setup reporting for '0x00158d0002089807' - Error: ConfigureReporting 0x00158d0002089807/2 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (SRSP - ZDO - extRouteDisc after 6000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

error 2020-10-09 17:09:02: Failed to setup reporting for '0x00158d0002086fd8' - Error: Bind 0x00158d0002086fd8/2 genOnOff from '0x00124b0018e1b0be/1' failed (SRSP - ZDO - bindReq after 6000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

According to this information, I assume report is not supported right ?

Koenkk, 2020-10-11:
Yes looks like it's not supported indeed.

closed 2020-11-18 by github-actions[bot]