Koenkk/zigbee2mqtt #4616: Network Scan Map Problems

wes1993, 2020-10-09:
## What happened
From version 1.15.0 the network scan won't work properly.
I have this errors from some devices:
Zigbee2MQTT:info  2020-10-09 10:49:08: Starting network scan (includeRoutes 'true')
Zigbee2MQTT:error 2020-10-09 10:49:22: Failed to execute LQI for 'Luce Ripostiglio Esterno' (Ikea Device)
Zigbee2MQTT:error 2020-10-09 10:49:32: Failed to execute routing table for 'Luce Ripostiglio Esterno' (Ikea Device)
Zigbee2MQTT:error 2020-10-09 10:49:44: Failed to execute LQI for 'Router Studio' (CC2531 Repeater FW)
Zigbee2MQTT:error 2020-10-09 10:49:55: Failed to execute LQI for 'Faro Parcheggio' (Ikea Device Smart Plug)
Zigbee2MQTT:error 2020-10-09 10:50:05: Failed to execute LQI for 'Faro Lago' (Ikea Device Smart Plug)
Zigbee2MQTT:error 2020-10-09 10:50:21: Failed to execute LQI for 'Sirena Capanno Attrezzi' (Ikea Device Smart Plug)
Zigbee2MQTT:error 2020-10-09 10:50:31: Failed to execute routing table for 'Sirena Capanno Attrezzi' (Ikea Device Smart Plug)
Zigbee2MQTT:info  2020-10-09 10:50:44: Network scan finished
## What did you expect to happen
I can control and this devices respond to the command so i expect that can map this also.
## How to reproduce it (minimal and precise)
Just launch network scan
## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2538 + CC2592
Adapter firmware version: 20200211
I Have 53 Devices Joined

aussiebyrd, 2020-10-14:
I have the same issue...
`Zigbee2MQTT:info  2020-10-14 21:29:16: Starting network scan (includeRoutes 'false')
Zigbee2MQTT:error 2020-10-14 21:29:26: Failed to execute LQI for 'CC2531_router_repeater'
Zigbee2MQTT:error 2020-10-14 21:29:36: Failed to execute LQI for 'telstra_zigbee_plug_01'
Zigbee2MQTT:error 2020-10-14 21:29:46: Failed to execute LQI for 'telstra_zigbee_plug_02'
Zigbee2MQTT:error 2020-10-14 21:29:56: Failed to execute LQI for 'telstra_zigbee_plug_03'
Zigbee2MQTT:info  2020-10-14 21:29:56: Network scan finished`

sourbrambles, 2020-10-15:
I receive the same errors on all my mains powered devices (Mainly Hue bulbs and Ikea plugs)

`ERROR: Failed to execute LQI for '0x0017880110422c76'`
`ERROR: Failed to execute routing table for '0x0017880110422c76'`

Zigbee2MQTT version: 1.15.0
Adapter hardware: ZZH

closed 2020-11-23 by github-actions[bot]