Koenkk/zigbee2mqtt #4617: Aqara Switch QBKG12LM reports 'single_undefined' action every 5 minutes

definitio, 2020-10-09:
## What happened
Aqara Switch QBKG12LM reports 'single_undefined' action every 5 minutes

## What did you expect to happen
Nothing

## How to reproduce it (minimal and precise)
1. Pair device
2. Get spam:
```
INFO: MQTT publish: topic 'zigbee/switch', payload '{"action":"single_undefined","consumption":3.33,"linkquality":105,"power":0,"state_left":"OFF","state_right":"OFF","temperature":24}'
INFO: MQTT publish: topic 'zigbee/switch', payload '{"action":"","consumption":3.33,"linkquality":105,"power":0,"state_left":"OFF","state_right":"OFF","temperature":24}'
INFO: MQTT publish: topic 'zigbee/switch/action', payload 'single_undefined'
```

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2538
Adapter firmware version: 20200211

Koenkk, 2020-10-09:
Please provide the debug logging of this.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

chernyshaw, 2020-10-10:
Same for my QBKG12LM. It's spamming 'single_undefined'. QBKG03LM at same network isn't spamming.
Zigbee2MQTT: 1.15.0 as Home Assistant add-on
Hardware: CC2538
```
Zigbee2MQTT:debug 2020-10-10 12:48:53: Received Zigbee message from 'switch_toilet', type 'attributeReport', cluster 'genBasic', data '{"65281":{"100":0,"101":0,"149":0.07830091565847397,"152":0,"154":0,"3":27,"5":6,"7":"0x0000000000000000","8":4385,"9":1537}}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-10 12:48:53: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"consumption":0.08,"linkquality":92,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:debug 2020-10-10 12:49:25: Received Zigbee message from 'switch_toilet', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 2 with groupID 0
Zigbee2MQTT:info  2020-10-10 12:49:25: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"action":"single_undefined","consumption":0.08,"linkquality":92,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:49:25: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"action":"","consumption":0.08,"linkquality":92,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:49:25: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"click":"","consumption":0.08,"linkquality":92,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:49:25: MQTT publish: topic 'zigbee2mqtt/switch_toilet/action', payload 'single_undefined'

Zigbee2MQTT:debug 2020-10-10 12:52:24: Received Zigbee message from 'switch_toilet', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
Zigbee2MQTT:info  2020-10-10 12:52:24: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"action":"single_undefined","consumption":0.08,"linkquality":89,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:52:24: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"action":"","consumption":0.08,"linkquality":89,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:52:24: MQTT publish: topic 'zigbee2mqtt/switch_toilet', payload '{"click":"","consumption":0.08,"linkquality":89,"operation_mode_left":"control_left_relay","power":0,"state_left":"OFF","state_right":"OFF","temperature":27}'
Zigbee2MQTT:info  2020-10-10 12:52:24: MQTT publish: topic 'zigbee2mqtt/switch_toilet/action', payload 'single_undefined'
```

definitio, 2020-10-10:
@Koenkk [log](https://hastebin.com/oniqiduwif.swift)

Koenkk, 2020-10-10:
Should be fixed now.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

definitio, 2020-10-10:
It's still spamming in the latest dev version.

chernyshaw, 2020-10-10:
Thanks! No more spam for me.

closed 2020-10-11 by definitio