Koenkk/zigbee2mqtt #4618: Error on level control for Busch-Jaeger 6715 (RM01)

dreamsinmotion, 2020-10-09:
I updated to the version 1.15.0 and the level control for my Busch Jaeger 6715 (RM01) is not working any more. I have returned to the previous version 1.13.0 and everything works fine again. In version 1.15.0 I get following error log:

```
Zigbee2MQTT:debug 2020-10-09 11:44:20: Received MQTT message on 'zigbee2mqtt/livingroom_buschjaeger_switch/set' with data '{"brightness":120}'
Zigbee2MQTT:debug 2020-10-09 11:44:20: Publishing 'set' 'brightness' to 'livingroom_buschjaeger_switch'
Zigbee2MQTT:error 2020-10-09 11:44:20: Publish 'set' 'brightness' to 'livingroom_buschjaeger_switch' failed: 'TypeError: Cannot read property 'convertSet' of undefined'
Zigbee2MQTT:debug 2020-10-09 11:44:20: TypeError: Cannot read property 'convertSet' of undefined
    at Object.convertSet (/app/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:2453:58)
    at EntityPublish.onMQTTMessage (/app/lib/extension/publish.js:225:52)
    at Controller.callExtensionMethod (/app/lib/controller.js:372:44)
    at Controller.onMQTTMessage (/app/lib/controller.js:269:14)
    at MQTT.emit (events.js:315:20)
    at MQTT.onMessage (/app/lib/mqtt.js:94:14)
    at WebSocket.<anonymous> (/app/lib/extension/frontend.js:86:27)
    at WebSocket.emit (events.js:315:20)
    at Receiver.receiverOnMessage (/app/node_modules/ws/lib/websocket.js:797:20)
    at Receiver.emit (events.js:315:20)
Zigbee2MQTT:info  2020-10-09 11:44:20: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"message":"Publish 'set' 'brightness' to 'livingroom_buschjaeger_switch' failed: 'TypeError: Cannot read property 'convertSet' of undefined'","meta":{"friendly_name":"livingroom_buschjaeger_switch"},"type":"zigbee_publish_error"}'
```

Koenkk, 2020-10-09:
make sure you have no changes to devices.js (for hassio zigbee_sheperd_devices: true)

dreamsinmotion, 2020-10-09:
I'm running zigbee2mqtt in docker and using the default devices.js

Koenkk, 2020-10-09:
Are you sure there is no Docker volume mapped overriding zigbee-herdsman-converters? It looks that your zigbee-herdsman-converters is not compatible with your Zigbee2MQTT verison.

dreamsinmotion, 2020-10-09:
Yes, I'm only mapping `/app/data` to a folder. My other devices are working as expected and the set state on the Busch Jaeger 6715 (RM01) is working as well. Only the level control doesn't work.

Koenkk, 2020-10-09:
I've tried to (virtually, since I don't have the device myself) reproduce this issue but failed to do so.

@prutseltje @WinSCaP @docbobo   is this working for you in Zigbee2MQTT 1.15.0?

prutseltje, 2020-10-10:
I have the Busch-Jaeger 6711 U, this is only a relay. So I'm unable to test the dimmer function. 

I am running Zigbee2MQTT 1.15.0 and do not experience any issues.

docbobo, 2020-10-10:
Don't have a dimmer either... But 1.15.0 has been running without issues so far for me...

Tom-Black, 2020-10-11:
I have the same issue with the level control of the Busch Jaeger 6715 dimmer. 
I also looked into the code and the problem seems to be that the converter RM01_light_brightness in toZigbee.js uses the converter light_brightness, but this converter has been removed in a recent [commit](https://github.com/Koenkk/zigbee-herdsman-converters/commit/354648b4ada1b28cef292e57b283485fdf3c9b28) 
```
    RM01_light_brightness: {
        key: ['brightness', 'brightness_percent'],
        convertSet: async (entity, key, value, meta) => {
            if (utils.hasEndpoints(meta.device, [0x12])) {
                const endpoint = meta.device.getEndpoint(0x12);
                return await converters.light_brightness.convertSet(endpoint, key, value, meta);
            } else {
                throw new Error('LevelControl not supported on this RM01 device.');
            }
        },
```

Koenkk, 2020-10-15:
Tanks to @Tom-Black this got fixed. Changes will be available in the latest dev branch tomorrow (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-10-15 by Koenkk