Koenkk/zigbee2mqtt #4621: Uselink power strip repor failed

shirou93, 2020-10-09:
## What happened
This power strip no update state when z2m restart and change state via button
https://www.zigbee2mqtt.io/devices/TS0115.html

## What did you expect to happen

Report state like ikea bulb and blitzwolf shp13

## How to reproduce it (minimal and precise)
pair strip stop z2m change state via button start z2m
## Debug info
Zigbee2MQTT version: 1.15.0 
Adapter hardware: CC26X2RB

cordinator:
"revision": 20200925,

log

 Failed to setup reporting for '0x842e14fffefd97ad' - Error: ConfigureReporting 0x842e14fffefd97ad/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 16644 - 1 - 3 - 6 - 7 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)

Koenkk, 2020-10-09:
According to your error, it looks that this device doesn't support reporting. @nukusinji  does it work for you?

shirou93, 2020-10-09:
> According to your error, it looks that this device doesn't support reporting. @nukusinji does it work for you?

Strange because when i change state via button when z2m is working state in home assistant will change.

Koenkk, 2020-10-09:
Sorry, I read your post wrong. If I understand correctly you have `report: true` set in `configuration.yaml` which causes the error:

```
Failed to setup reporting for '0x842e14fffefd97ad' - Error: ConfigureReporting 0x842e14fffefd97ad/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 16644 - 1 - 3 - 6 - 7 after 10000ms)
at Timeout._onTimeout (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
at listOnTimeout (internal/timers.js:549:17)
at processTimers (internal/timers.js:492:7)
```

This is your problem right?

shirou93, 2020-10-09:
Right. I thought the states would be checked and even if the device state change were if z2m was turned off. I checked shp13 and the states only change when z2m is on, same as power strip.

Only IKEA bulbs report status at intervals.

Only the power strip has a log error.

Koenkk, 2020-10-09:
Error is fixed in latest dev branch. But I don't know what you mean with this yet:

> Right. I thought the states would be checked and even if the device state change were if z2m was turned off. I checked shp13 and the states only change when z2m is on, same as power strip.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Pigotka, 2020-10-13:
I have similar problem for my Illuminance sensor:
`error 2020-10-13 09:39:45: Failed to configure 'IlluminanceSensor1', attempt 1 (Error: Bind 0x04cf8cdf3c786ead/1 genPowerCfg from '0x00124b001cd48695/1' failed (AREQ - ZDO - bindRsp after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.3/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))`
Can it be related?

Koenkk, 2020-10-13:
@Pigotka for battery powered device, try to force configure via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure and make sure to wakeup the device right before doing this (by clicking a button on it).

Pigotka, 2020-10-13:
I was maybe not clear. I don't have troubles with the device it works. I have lines like this in loge even logged before the error:
`info  2020-10-13 14:34:34: MQTT publish: topic 'zigbee2mqtt/IlluminanceSensor1', payload '{"battery":100,"voltage":3100,"linkquality":63,"illuminance":33661,"illuminance_lux":2322}'`
But still I sometimes see this error when I restart Zigbee2MQTT.

shirou93, 2020-11-16:
@Koenkk hi, this power strip have power on behavior option im tuya app. Many sockets have this option in app. How change this options on zigbee2mqtt?

closed 2020-12-17 by shirou93

SPEC1AL1ST, 2021-01-17:
Same problem