Koenkk/zigbee2mqtt #4622: QBKG12LM unable to group both relays, endpoint 2 getting error

Arkadi85, 2020-10-09:
## What happened
Trying to add endpoint 1 and 2 to a group, end point 2 getting an error.

## How to reproduce it (minimal and precise)
Try to group both relays on the switch.

## Debug info
MQTT publish: topic 'zigbee2mqtt/bridge/response/group/members/add', payload '{"data":{"device":"0x00158d0002cb6d0b/2","group":"light"},"error":"Failed to add from group (Command 0x00158d0002cb6d0b/2 genGroups.add({\"groupid\":1,\"groupname\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableRecovery\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Timeout - 61241 - 2 - 228 - 4 - 0 after 10000ms))","status":"error"}'
ERROR: Failed to add from group (Command 0x00158d0002cb6d0b/2 genGroups.add({"groupid":1,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 61241 - 2 - 228 - 4 - 0 after 10000ms))

Koenkk, 2020-10-09:
- Can you try to wakeup the device right before doing so? 
- Can you share the data/databse.db entry of this device?

Arkadi85, 2020-10-09:
1. I have 7 switches, and all of them are awake, I'm able to create groups with endpoint 1 with different switches, but endpoint 2 getting an error.
2. {"id":27,"type":"Router","ieeeAddr":"0x00158d0002cb6d0b","nwkAddr":61241,"manufId":4447,"manufName":"LUMI","powerSource":"Mains (single phase)","modelId":"lumi.ctrl_ln2.aq1","epList":[1,2,3,4,5,6,7],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,3,6,16,5,10,1,2],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"65281":{"3":27,"5":92,"7":"0x0000000000000000","8":4385,"9":3844,"100":0,"101":0,"149":35.636375427246094,"152":0,"154":0},"manufacturerName":"LUMI","powerSource":1,"zclVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"06-25-2018"}},"genOnOff":{"attributes":{"61440":117440715,"onOff":0}},"msIlluminanceMeasurement":{"attributes":{"measuredValue":0}},"msOccupancySensing":{"attributes":{"occupancy":1}}},"binds":[],"meta":{}},"2":{"profId":260,"epId":2,"devId":81,"inClusterList":[6,16],"outClusterList":[],"clusters":{"genOnOff":{"attributes":{"61440":117440608,"onOff":0}}},"binds":[],"meta":{}},"3":{"profId":260,"epId":3,"devId":9,"inClusterList":[12],"outClusterList":[12,4],"clusters":{"genAnalogInput":{"attributes":{"presentValue":0}}},"binds":[],"meta":{}},"4":{"profId":260,"epId":4,"devId":83,"inClusterList":[12],"outClusterList":[12],"clusters":{},"binds":[],"meta":{}},"5":{"profId":260,"epId":5,"devId":0,"inClusterList":[16,18],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}},"6":{"profId":260,"epId":6,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}},"7":{"profId":260,"epId":7,"devId":0,"inClusterList":[18,16],"outClusterList":[],"clusters":{"genMultistateInput":{"attributes":{"presentValue":1}}},"binds":[],"meta":{}}},"appVersion":1,"stackVersion":2,"hwVersion":18,"dateCode":"06-25-2018","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1602275251888}

Koenkk, 2020-10-09:
As can be seen in your database entry, the group cluster is only supported by endpoint 1 (group cluster number = 4, you see it in the `inClusterList`). So it seems the second endpoint does not support groups.

Arkadi85, 2020-10-09:
Ok thank you, if it possible to fix, or the device itself does not support it?

Koenkk, 2020-10-09:
I'm actually not sure if it works with a QBKG03LM, in this context it's more meant as an example of a multiple endpoint switch. Also a QBKG03LM is not identical to a QBKG12LM.

Koenkk, 2020-10-09:
The device doesn't support it, so cannot be fixed for Zigbee2MQTT.

closed 2020-10-10 by Arkadi85