Koenkk/zigbee2mqtt #4626: After renaming a device it starts to report as undefined.

danielo515, 2020-10-09:
## What happened
After renaming certain devices they stopped working. 
They are all aqara devices. I renamed them using hassio integration. After renaming if I check the logs I can see errors about undefined devices popping everytime I activate one of the renamed devices. 
Xiaomi button and a ikea bulb renames without problem, but the rest of aqara devices suffer from this problem. 

## What did you expect to happen
To rename them and still work

## How to reproduce it (minimal and precise)

## Debug info
Will post later this info 
Zigbee2MQTT version:
Adapter hardware: CC2531, CC2530, CC26X2R1, CC1352P-2, Conbee II
Adapter firmware version:

Haxiboy, 2020-10-10:
I was looking for the opposite a few days ago. Renaming the device changes the name of the entities (this is my expected behavior). There should be a dialog like in home assistant renaming system if we want to rename the entities too.

danielo515, 2020-10-10:
> Renaming the device changes the name of the entities
Isn't that exactly what is expected? For some sensors it works (home assistant recognizes the new name) but for some others it just doesn't work because the explained behavior

Koenkk, 2020-10-10:
The new api allows to specify wether to also update Home Assistant entity IDs, see https://github.com/Koenkk/zigbee2mqtt.io/blob/new_api/docs/information/mqtt_topics_and_message_structure.md#device `zigbee2mqtt/bridge/request/device/rename` , to enable the new api see https://github.com/Koenkk/zigbee2mqtt/issues/3281

danielo515, 2020-10-12:
Thanks @Koenkk, for now I repaired the decides and they picked the new name (in hassio they also have the new name, but they use the Mac as id). I'll check the new api if I get a chance

closed 2020-11-19 by github-actions[bot]