Koenkk/zigbee2mqtt #4629: UNSUPPORTED_ATTRIBUTE for 'Magnetic door & window contact sensor' (SZ-DWS04) by Sercomm

alexball811, 2020-10-10:
## What happened

Have 5 of these devices and all exhibit the same error message in logs;

Zigbee2MQTT:error 2020-10-10 17:32:05: Failed to configure '0x000d6f000ed8ac12', attempt 2 (Error: ConfigureReporting 0x000d6f000ed8ac12/1 genPowerCfg([{"attribute":"batteryPercentageRemaining","minimumReportInterval":3600,"maximumReportInterval":62000,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Status 'UNSUPPORTED_ATTRIBUTE')
    at Endpoint.checkStatus (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:173:23)
    at Endpoint.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman/dist/controller/model/endpoint.js:384:26)
    at Generator.next (<anonymous>)

## What did you expect to happen

No error in logs and hopefully provide battery level updates for the Sercomm device.

## How to reproduce it (minimal and precise)

Pair device to zigbee2mqtt and wait for errors to appear in zigbee2mqtt logs (as above).

## Debug info
Zigbee2MQTT version: 1.15.0
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608 (also tried CC2531_SOURCE_ROUTING_20190619)

Koenkk, 2020-10-10:
Try changing the definition in `devices.js` to

```js
{
    zigbeeModel: ['SZ-DWS04', 'SZ-DWS04N_SF'],
    model: 'SZ-DWS04',
    vendor: 'Sercomm',
    description: 'Magnetic door & window contact sensor',
    supports: 'contact',
    fromZigbee: [fz.ias_contact_alarm_1, fz.temperature, fz.battery],
    toZigbee: [],
    meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
    configure: async (device, coordinatorEndpoint) => {
        const endpoint = device.getEndpoint(1);
        await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
        await configureReporting.temperature(endpoint);
        try {
            await configureReporting.batteryPercentageRemaining(endpoint);
        } catch (e) {
            // batteryPercentageRemaining not supported by all versions
            // https://github.com/Koenkk/zigbee2mqtt/issues/4629
            await configureReporting.batteryVoltage(endpoint);
        }
    },
    exposes: [
        exposes.boolean('contact'), exposes.boolean('battery_low'), exposes.boolean('tamper'),
        exposes.numeric('temperature').withUnit('°C'), exposes.numeric('battery').withUnit('%'),
    ],
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

alexball811, 2020-10-11:
Hi Koenkk,

Thanks for all your asistance and a fantastic product, I have done what you have instructed but I now get the error msg;

```
/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/devices.js:10361
            exposes.boolean('contact'), exposes.boolean('battery_low'), exposes.boolean('tamper'),
            ^
ReferenceError: exposes is not defined
    at Object.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/devices.js:10361:13)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
    at Module.load (internal/modules/cjs/loader.js:985:32)
    at Function.Module._load (internal/modules/cjs/loader.js:878:14)
    at Module.require (internal/modules/cjs/loader.js:1025:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/zigbee2mqtt-1.15.0/node_modules/zigbee-herdsman-converters/index.js:3:17)
    at Module._compile (internal/modules/cjs/loader.js:1137:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1157:10)
```
Obviously if I change it to
```
    {
        zigbeeModel: ['SZ-DWS04', 'SZ-DWS04N_SF'],
        model: 'SZ-DWS04',
        vendor: 'Sercomm',
        description: 'Magnetic door & window contact sensor',
        supports: 'contact',
        fromZigbee: [fz.ias_contact_alarm_1, fz.temperature, fz.battery],
        toZigbee: [],
        meta: {configureKey: 1, battery: {voltageToPercentage: '3V_2100'}},
        configure: async (device, coordinatorEndpoint) => {
            const endpoint = device.getEndpoint(1);
            await bind(endpoint, coordinatorEndpoint, ['msTemperatureMeasurement', 'genPowerCfg']);
            await configureReporting.temperature(endpoint);
            await configureReporting.batteryVoltage(endpoint);
        },
    },
```
I get no errors;

```
Zigbee2MQTT:info  2020-10-11 11:09:36: Successfully configured '0x000d6f000ed8ac12'
Zigbee2MQTT:info  2020-10-11 11:09:36: Configuring '0x000d6f000ed86533'
Zigbee2MQTT:info  2020-10-11 11:09:39: Successfully configured '0x000d6f000ed86533'
Zigbee2MQTT:info  2020-10-11 11:09:39: Configuring '0x000d6f000c2f2ec7'
Zigbee2MQTT:info  2020-10-11 11:09:41: Successfully configured '0x000d6f000c2f2ec7'
Zigbee2MQTT:info  2020-10-11 11:09:41: Configuring '0x000d6f000c2e7716'
Zigbee2MQTT:info  2020-10-11 11:09:45: Successfully configured '0x000d6f000c2e7716'
Zigbee2MQTT:info  2020-10-11 11:09:45: Configuring '0x000d6f000c2d8a74'
Zigbee2MQTT:info  2020-10-11 11:09:49: Successfully configured '0x000d6f000c2d8a74'
```
But no battery 'batteryPercentageRemaining' but it sounds like I may have never had it in the first place, which I am fine with.

Thanks again and look forward to your feedback, Alex.

Koenkk, 2020-10-11:
Looks good, don't worry about the `exposes` part. Do you get a `battery` value now for this device?

alexball811, 2020-10-12:
Hi again Koenkk,

That has now removed all the errors in the logs but the battery status hasn't updated after a 24 hour period, I think this device may not report the battery status. 

Thanks again for your assistance, Alex.

Koenkk, 2020-10-12:
It seems so, otherwise it should have already been reported.

Koenkk, 2020-10-31:
Implemented fix.

closed 2020-10-31 by Koenkk